cariboulabs/cariboulite #5: [Question] Why these exclusions?

MikeGlennFWC, 2021-06-17:
Why are these 2 ranges excluded?

`(excluding 2398.5-2400 MHz and 2483.5-2485 MHz)`

meexmachina, 2021-06-17:
Hi @MikeGlennFWC, Thanks for asking.
The answer is that the AT86RF215 high freq (2.4GHz) channel is qualified for 2400-2483.5 MHz. Which gives us 83.5 MHz of frequency tunable window. On the other hand, the mixer RFFC5072 has an internal synthesizer that performs well starting from 85 MHz minimal LO frequency. So in order to tune for 2399MHz for instance, we will need to upconvert this signal (because it is lower than the window of the receiver). But then we will need a minimal LO to still reside within the said window. 2399+85 = 2484MHz > 2483.5 MHz. The same for the higher frequencies that will need to be downconverted by the mixer. 

In practice, a test shows that AT86RF215 may be able to perform well enough even outside the said window (2400-2483.5 MHz) at least a few MHz to each side, and thus these restrictions may not appear in reality (which means, the system may work fluently all along this spectrum). But still, according to the specifications of the component, its operation outside the specs is not guaranteed... So that's the reason, hope it answers.

MikeGlennFWC, 2021-06-17:
Thanks for the detailed answer.

That must be what this note is referring to:

`Note: The gaps are defined by the design constraints of the system and may not exist in real-life hardware. Actual modem synthesizer outputs test show wider margins at room temperature than those written in the datatsheet, but, as noted by Microchip, performance may suffer.`

I thought the gaps were the white gaps. Not the orange. Sorry about that.

closed 2021-06-18 by meexmachina