cariboulabs/cariboulite #8: Ballpark pricing?

ElRoberto538, 2021-06-21:
I've always wanted an SDR, but they still seem pretty pricey. Any idea what the target/ballpark pricing will be?

obones, 2021-06-21:
Just so you know, RTL-SDR costs around 40€ with an antenna kit. But it has limitations and cannot transmit, so like you I'm eagerly waiting for the pricing of this solution

meexmachina, 2021-06-21:
Hi guys, Thanks a lot for the interest. Reading all the great comments is our main driving force.
Actually, I'm currently optimizing the pricing and I'm going to work on it together with the super professional and experienced CrowdSupply staff. So yes - I'm waiting for the final pricing as well :) 
@ElRoberto538 - our target is definitely to make it accessible, and to do so we are optimizing the BOM to the bones without compromising the quality, testability, etc.
So I hope I'll soon have much better answers...

closed 2021-06-28 by meexmachina