cariboulabs/cariboulite #9: Gerber files: Outline missing with some manufacturers

kenokenobingo, 2021-06-26:
[OSH Park](https://oshpark.com/) and [Aisler](https://aisler.net/) can’t process the Gerber files propers because they report a missing outline. JLC PCB doesn’t have any problems with the files, though.

[image]

[image]

GordonFR24, 2021-06-28:
PCBWay have similar issues

meexmachina, 2021-06-28:
Hi @kenokenobingo and @GordonFR24 Thanks a lot,
I uploaded a new file with "cariboulite_r2.Outline" hope it helps...
please let me know it still doesn't work.

kenokenobingo, 2021-06-29:
Thanks, @meexmachina! It’s working now with AISLER, OSH Park, and PCBWay.

meexmachina, 2021-06-29:
Thanks for the feedback!

closed 2021-06-29 by meexmachina