cariboulabs/cariboulite #10: kicad project

eljuligallego, 2021-06-28:
Hi,

would like to know if you plan to share the project pcb file (in kicad would be great).

I want modify the design to make a more simple and cheapest project to evaluate the AT86RF215 with the raspberry pi and most of the RF mixer part is not needed.

P.D.

From your bom the MAAM-011206-TR1000 RF amplifier is obsolete and the replace part MAAM-011206-TR3000 cost 11 euros in mouser and you use 2!! and total bom is almost 100€ on mouser(the best place i found they have all components).

Thanks,
your project is very promising, I hope you can make it to buy one.

meexmachina, 2021-06-28:
Hey, @eljuligallego Thanks for the issue!
first the LNA:
see [https://www.mouser.co.il/ProductDetail/MACOM/MAAM-011229-TR1000?qs=%2Fha2pyFaduggI6reMC9QyZgOOE6I6Wz4uNUo%252BSLcreYC7Q9KPkXxpuV93K2DneDS]

The component p/n is "MAAM-011229-TR1000" - not "MAAM-011206-TR1000" which is outrageously expensive (though a really amazing component indeed and worth the money - see datasheet).
So yes, if it was your suggested component I would be crazy to use it for a low-cost SDR project...
To be honest I think there is something wrong in google's search - I got to the very same component you shared as well once searching for the datasheet pdf file for the LNA. And I didn't notice the p/n was different. You can imagine the feeling of... "What do you mean $13?! yesterday it was $2.5!!" And of course, after a small heart attack, I noticed the P/N is wrong, prepared a calm-down coffee, and continued coding :)

As for the suggestion to discard the MIXER and the complex RF path to minimize cost and complexity - it is something that I thought about a lot! And it will lower the cost dramatically indeed. The point here though is that the two 3PDT (Skywork) on the board are quite low-cost components (around $0.6 each). and I thought - If I will even need a board without the MIXER and all the LNA / PA / Filtering and DPDT etc., I will just NOT assemble them. Then I will only have two 3PDT assembled to conform to the current design and you will get what you need - a 2.4 GHz SDR without a widely tunable range. It is simpler (I think) than redesigning the board and layout.

I'm using Altium's Circuit Studio as my primary CAD. I'm just used to it and moving to KiCAD is not currently planned due to lack of time. Anyway, circuit studio is really nice and gives high-end features that are needed in complex designs. But I suppose I can convert the file into KiCAD - I'll check how to do it.
In any case, I'm planning to put the complete design filed as soon as I finalize them and have them fully tested.

Your thoughts?

eljuligallego, 2021-06-28:
Hi again,

Thanks a lot for taking your time in answer this issue. i will reply in order of answered.

First sorry but didn't realize the part was wrong, just take the bom from github and loaded in mouser bom calculator to have and early approach of cost with RF mixer and without and this part was loaded by mouser, my fault.

The option of having same pcb for 2 versions of boards is a good option,if you plan to sell the 2 choices ;)

You have design a complete new SDR platform, its a very ambitious project with lot of possible problems. Just the RF chain could be a nightmare, so for this reason a early version with just the transceiver could be a good starting point. 

For the kicad project don't need to start from scratch. in newer version of kicad they say there is a option to import Altium projects:

https://www.kicad.org/blog/2020/04/Development-Highlight-Altium-Pcb-Importer/

If you like when you share the altium project (.PcbDoc file) i can give a try and share the experience.

Thanks,
regards,
julian.

meexmachina, 2021-06-29:
@eljuligallego  thanks I'll give it a try and update.

mgrennan, 2021-08-31:
I check the "Nightly Build" and didn't find the Altium import yet.

closed 2023-02-09 by meexmachina