cariboulabs/cariboulite #12: stack several cariboulites

eljuligallego, 2021-06-28:
Hi again, just have another doubt:

I have seen in first version of pcb you have pictures with 2 stacked boards, is this possible?. 

It would be amazing to stack several cariboues for MIMO or radar. something like the kerberos with the rtl-sdr chip but with better price per chip and best performance and quality. 

I imagine some clock management and sync data interleaved between boards has to be done and the throughput increase almost to the limit of SMI but there is nothing similar in market at low price.

Other bottleneck could be how the raspberry manage the amount of data for processing.

Maybe I could help if you want with this since have experience with fpgas and raspberry since I'm very interested in this option.

meexmachina, 2021-06-29:
Hi @eljuligallego! Indeed I agree with you.
It is possible but I'm still working on this feature. This in not so simple though so I think this will be available only in the next revision.
So that's how it goes:
a. Each CaribouLite has a 4-bit configuration resistor set (backside of the print).
b. the SMI is a multi-slave interface so, you can control the slave address using the address bits (currently 4-bit populated).
c. The problems are:
   1. SPI3 (multiplexed between the FPGA - programming and control), modem, and mixer.
   2. control bits - mainly resets, irqs, FPGA programming.

d. To mitigate these problems, all those lines need to be buffered by a tristate multi-channel buffer. then the address 4-bits shall enable/disable the currently used subsystem.

Still to be determined: 
1. is SMI bandwidth enough for 2 or more units.
2. timing constraints.
3. clock synchronization so as to get a Kerberos type coherent txrx.

your thoughts?
any better idea?

eljuligallego, 2021-07-06:
Hi,sorry for late answer. 

My thoughts are:

maybe the control bits can be routed via the fpga instead use the raspberry pi bus directly so each board can control it owns signals via a control config for each fpga (SMI as control bus?).

the spi multiplex could be done with a spi pass trough inside the first fpga to the other devices?.

maybe the pmod can be used for this purpose?.

meexmachina, 2021-07-08:
Hi @eljuligallego,
This is also possible indeed and a great idea. Furthermore, SMI has a feature that allows a "direct more" control. Say, you are currently utilizing the bus as usual over DMA and receiving (or sending) large chunks of information. Then you got to push a small sized message in between - apparently, this is possible through a "on demand" / direct mode parallel link (with completely different set of registers). I didn't test it yet though - its in my backlog.
To be honest, the first idea was exactly that and then I realized that path is too risky for the first hardware revision until I get enough confidence in the FPGA capabilities, SMI features and so on. So I decided to go with the fullest control of those components that can be controlled through SPI to minimize the software development risks and RF testing.
Now I'm so glad you came up with this thought - that a multi unit solution can be attractive and from my current perspective it looks easily feasible with more than one engineering solution...
The use case is also interesting (in my opinion) as the stacked up solution allows a very neat expansions, direction finding, coherent transmitter / receiver? maybe a phased array beam forming possibility? smart wireless repeaters? a faster spectrum analyzer. Wow!!! and all that over a raspberry pi.

Yes, @eljuligallego this option is definitely going to be in the next version.
I'd be happy for ideas - not only on the engineering how - but also use cases that you think are feasible and usable for people like us...

Thanks, David

eljuligallego, 2021-07-10:
Hi, 
I have another idea as a workaround.

What about instead stack the caribous with the problems associated use several cariboulites with each one have it own raspberry connected and use the  pmod connector for trigger signal for synchronized captured samples?.

Just have to share the clock between devices to have a phase/freq coherent mimo system.

With this workaround the staking option could be solved without any change in hardware, just the fpga code.

And then of course you have to share the data between raspberry or maybe have one server where process all the captured data in a more powerful computer than a pi zero.

What do you think?

meexmachina, 2021-10-20:
@eljuligallego it is possible. The only problem - sharing the reference clock in the mixer level will give a coherent clock signal to the mixer synthesizers (which will result in partial phase coherence of the LO)
the problem to address further is - what about the modem - Caribou does dual conversion to get to IF.
I need to figure it out.

David

eljuligallego, 2021-10-20:
Hi, good to hear form you since all this time.

My idea is to use this setup just with the modem directly and ignore the mixer and work with just the frequencies allowed by the modem.

jks7592, 2021-11-05:
I just want to say thank you both for working on this. Ever since I heard of software defined radio I thought it would be cool to do software defined radar. I'm no expert, but I can't wait to see someone build a new version of the MIT cantenna radar with software defined radios and get impressive radar images. Since judging by the picture y'all are building to the raspberry pi zero, a pair of these might even be light enough to put on a drone as an imaging radar.

closed 2023-02-09 by meexmachina