cariboulabs/cariboulite #13: Adding connectors to bypass the RF switch?

rfamp, 2021-07-26:
Hi, 
I stumbled across your impressive project. Do you plan on adding something like U.FL sockets on J1/J3 of the RF switch (U11) to bypass it in cases where a user wants a dedicated rx and tx connector for bands higher than 1GHz? Using a 0 ohm resistor one could choose between the different options?
Regards

meexmachina, 2021-07-26:
Hi @rfamp, thanks for the issue.
So you suggest giving the possibility to separate the Tx and Rx for the 6ghz channel without passing through u11? 
It actually is a good idea for testing purposes but also to have a separate antennas for Tx and Rx. Say if you want an active antenna for Rx...
I didn’t originally plan to do what you suggest, but I’ll consider in the production revision. I’ll consult you once I open the revision to make sure I got you correctly.
Thanks a lot, if additional requests don’t hesitate to ask... 
David

rfamp, 2021-07-26:
Hi,
thank you for the answer.
> So you suggest giving the possibility to separate the Tx and Rx for the 6ghz channel without passing through u11?
Yes exactly, as you said, this would open the possibilities of separate rx/tx antennas in the 6 GHz path.

Best Regards!

closed 2023-05-05 by meexmachina