cariboulabs/cariboulite #14: HF Support

batica81, 2021-08-05:
Any chance RX/TX will work in 1-30 Mhz range? Hams would love it! :)

meexmachina, 2021-08-10:
Hey @batica81  Thanks for you message.
I will have an answer once I test the boards for these frequencies. I'm not very optimistic on the performance levels at those frequencies though, and that is why I exclude them from the official bands.

mgrennan, 2021-08-31:
This doesn't need to be HF for HAMs to love it.  (W5TSU).  Nor is TX needed.  I run Several Internet SDR receivers.  W5TSU.net.
I'd love to get this integrated into the OpenWebRX project.  
I'm unsure of the receive bandwidth.  You said the "Analog Bandwith" is <4MHz.   This is TX bandwith? 
And the "Sample Rate" of the AD is 4MSPS.  What is the IO bandwith to the PI?  IS PIO speeds?

meexmachina, 2021-08-31:
Hello @mgrennan, thanks for the message.
The analog receive bandwidth is selectable while the maximal possible analog RX bandwidth is 4 MHz as for the I/Q receiver with 4 MSPS sampling rate.
The IO digital bandwidth of the Pi is currently tested to be > 200 mbits/sec using the SMI interface. I'm currently working on pushing it towards >250 mbits/sec. 
The SMI interface is actually a great thing that is very poorly documented. I'm going to investigate it not only for this project but also for the following other projects in the community.
The OpenWebRX is a great idea for the project to be integrated in. The RPI computer can be a dual receiver node in a larger network that broadcasts information over the internet. Which is my goal. Do you know whether that is a plan to integrate TX in the OpenWebRX?

David

mgrennan, 2021-08-31:
David,  We are "In Tune" I'm hoping to create a receiver registry with API.  Beside lookup receivers, receiver could register their location, high / low freq, bandwith, protocol, (RTL, Raw IQ...).  Receivers (OpenWebRX) would read the API for a menu of receivers.  
I promoted the TDoA project for example (https://www.rtl-sdr.com/kiwisdr-tdoa-direction-finding-now-freely-available-for-public-use/).  I see a day with a receiver grid where you could watch / lission to anything you want anywhere in the world.  Millions of "Local" receivers you tune into with a WebApp.

blacktigersoftware, 2021-09-01:
I'm only interested in this project for both RX and TX support in ISM bands   I want to test different modulation techniques and their effects, so I need TX support.    If you just want to put tuners across the globe, they are many cheap and available SDR's already on the market..

mgrennan, 2021-09-01:
Will, there are kind of cheap SDR but few in this spectrum.  HackRF for example. 
I'm interested... So what part of ISM bands is broad. What modulation techniques?

blacktigersoftware, 2021-09-01:
well, although I think I'm getting off-topic, I will take the opportunity to present a use case.   For the record, I'm very inexperienced in radio, but work FPGA/Firmware professionally.  I have done the SDR-dongle thing and know a bit about I/Q.  For my hobby of flying quadcopters, basically called "FPV", we use 900 Mhz, 2.4 Ghz and 5.8 Ghz.  Some new tech came out for ELRS (https://github.com/ExpressLRS/ExpressLRS).  I want to analyze/capture signals used for RC Links.  I also want to play with trying different modulations and do a QAM based signal.   And I'm very anxiously awaiting crowdfunding on this.

mgrennan, 2021-09-01:
Wow again.  Beside being a HAM I'm also the president of the local RC Club (torks.org) and a contributor to Ardu-Copter code.  I also follow the OpenTX and MultiProtocol projects.  I didn't know anything about ExpressLRS. Reading now.  If you'd like to continue this thread email me at w5tsu @ grennan.com

meexmachina, 2021-10-20:
Hi guys, I tested a few days ago the performance down to 1 MHz and it seems to have a pretty decent performance also in the range of 1-30 MHz.
Of course, it's way out of spec, but I didn't see a very catastrophic attenuation/distortion.

andreasr81, 2022-02-18:
maybe this where a good qo100 rx/tx station, do the cheaper version cover the 2,4ghz tx range of the qo100?

closed 2023-05-05 by meexmachina