cariboulabs/cariboulite #20: Jetson support

uecken, 2022-01-31:
Hi, Im looking forward to use this compact TRX SDR.
By the way, caribou lite can be controlled by Jetson?
If so , Jetson use GPGPU can process much IQ signals faster.

meexmachina, 2022-03-19:
Hi @uecken, thanks for the question and I apologize for the long long time it took me to answer.
The Caribou is not adapted to Jetson as Jetson doesn't have the SMI interface. It can still be controlled by Jetson in the SPI level to use its components - the modem, mixer, etc. and control the modem's internal baseband. but in general it was created for RPI.
Using GPU processing on the RPI4 may be possible as well and there are some open source projects that allow it.

fungaldore, 2022-09-26:
Hi I'm also curious about compatibility with other boards since Raspberry Pis are difficult to get your hands on right now.  Is it possible to use a BeagleBoard computer?  I can't tell if they have a SMI interface (I assume you're using it for precise timing), but they have PRU chips.