cariboulabs/cariboulite #26: ADS-B with CaribouLite ISM?

Vinz87, 2023-01-03:
Hi, based on specifications ADS-B (1090 MHz) should be slightly outside CaribouLite ISM frequency range (1020 MHz max).
But I also read that frequency ranges are somewhat conservative.
Is the maximum tunable frequency of 1020 MHz a hardware limit in the sense that the modem can't be tuned at 1090 MHz? Or it will be some kind of software-based limit?
So overall the question is if I could be able to receive ADS-B (1090 MHz) signals with CaribouLite ISM version, thank you.

Batbx, 2023-02-01:
Great question, also waiting for the answer.

Vinz87, 2023-02-15:
Actually I received answer to this question by email:

> Tried it 😊
> Unfortunately, the maximal frequency of the ISM is around ~1050 and above this level, the synthesizer stability (inside the modem) loses stability (pll doesn't lock).
> It may work if you cool the board however 😉 to -60deg... but it would be cheaper to go with either the full version or another SDR - RTLSDR in this case...
> 
> Cheers,
> David

closed 2023-02-15 by Vinz87