cariboulabs/cariboulite #29: aarch 22.04 testing and feedback

alphafox02, 2023-01-29:
Working w/ a Pi4 and DragonOS Pi64 (22.04). So far having installed the libs and iir I'm getting the following results 

```
SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
01-28 19:03:12.399 11346 11346 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:535 driver initializing
01-28 19:03:12.400 11346 11346 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:544 Initializing signals                  
01-28 19:03:12.400 11346 11346 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:678 setting up signal handler            
01-28 19:03:12.400 11346 11346 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:178 Setting up board I/Os                            
01-28 19:03:12.400 11346 11346 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio                                                  
01-28 19:03:12.400 11346 11346 V initCheckPermitted@pigpio.c:7396                                                                                
+---------------------------------------------------------+                                                                                      
|Sorry, you don't have permission to run this program.    |                                                                                      
|Try running as root, e.g. precede the command with sudo. |                                                                                      
+---------------------------------------------------------+                                                                                      
                                                                                                                                                 
                                                                                                                                                 
01-28 19:03:12.400 11346 11346 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed                                           
01-28 19:03:12.400 11346 11346 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:181 Error setting up io_utils                        
01-28 19:03:12.400 11346 11346 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:313 initializing submodules                   
01-28 19:03:12.400 11346 11346 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:317 INIT FPGA SMI communication               
01-28 19:03:12.400 11346 11346 I CARIBOU_SMI_Main caribou_smi_init@caribou_smi.c:141 initializing caribou_smi                                    
01-28 19:03:12.400 11346 11346 E CARIBOU_SMI_Main caribou_smi_init@caribou_smi.c:146 can't open smi driver file '/dev/smi'                       
01-28 19:03:12.400 11346 11346 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:321 Error setting up smi submodule            
01-28 19:03:12.400 11346 11346 E FPGA caribou_fpga_close@caribou_fpga.c:144 caribou_fpga_close: dev not initialized                              
01-28 19:03:12.401 11346 11346 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:221 Releasing board I/Os - closing SPI             
01-28 19:03:12.401 11346 11346 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:326 closing uninitialized device                                 
01-28 19:03:12.401 11346 11346 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:224 Releasing board I/Os - io_utils_cleanup        
[ERROR] cariboulite_init_driver() failed                                                                                                         
01-28 19:03:12.401 11346 11346 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:678 setting up signal handler
Printing 'findCariboulite' Request:                                                                                                              
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3                                                                          
Found device 0
  channel = S1G
  device_id = 0
  driver = Cariboulite
  label = CaribouLite S1G[ca381142]
  name = CaribouLite RPI Hat
  serial = ca381142
  uuid = c522fca7-00fb-4669-a0c5-f56a417af315
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 1
  channel = HiF
  device_id = 1
  driver = Cariboulite
  label = CaribouLite HiF[ca381143]
  name = CaribouLite RPI Hat
  serial = ca381143
  uuid = c522fca7-00fb-4669-a0c5-f56a417af315
  vendor = CaribouLabs LTD
  version = 0x0001
```

I can run w/ Sudo and get the following,
```
sudo SoapySDRUtil --find                                             
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
01-28 19:03:42.792 12081 12081 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:535 driver initializing
01-28 19:03:42.792 12081 12081 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:544 Initializing signals                  
01-28 19:03:42.792 12081 12081 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:678 setting up signal handler            
[... 71 lines cut ...]
01-28 19:03:45.530 12081 12081 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:221 Releasing board I/Os - closing SPI             
01-28 19:03:45.530 12081 12081 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:326 closing uninitialized device                                 
01-28 19:03:45.530 12081 12081 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:224 Releasing board I/Os - io_utils_cleanup        
01-28 19:03:45.530 12081 12081 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:698 driver released
```

Attempting to access with GQRX (non root)
```
soapy_sighandler caught SIGSEGV
[INFO] soapy_sighandler killing soapy_cariboulite (cariboulite_release_driver)
01-28 19:07:05.729 18473 18473 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:690 driver being released
01-28 19:07:05.729 18473 18473 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:698 driver released                            
CaribouLite: Signal [11] received from pid=[0]                                                                                                   
Signal [11] caught, with the following information:                                                                                              
   signal errno = 0                                                                                                                              
   signal process pid = 0                                                                                                                        
   signal process uid = 0                                                                                                                        
   signal status = 0                                                                                                                             
   signal errno / SIGSEGV / the process access invalid region of memory                                                                          
SIGSEGV: memory access violation                                                                                                                 
01-28 19:07:05.729 18473 18473 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:690 driver being released                      
01-28 19:07:05.729 18473 18473 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:698 driver released                            
01-28 19:07:05.820 18473 18473 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:690 driver being released                      
01-28 19:07:05.820 18473 18473 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:698 driver released
```

alphafox02, 2023-01-29:
Well, looks like on aarch64 22.04 that maybe the /dev/smi is not available even after turning on dtoverlay=smi and dtoverlay=smi-dev. I'll compare on a regular pi build.

EelkeVisser, 2023-01-29:
I also tried that on a Raspberry Pi 4B, it didn't work out. Note that there are specific overlay files *.dtbo at:
https://github.com/cariboulabs/cariboulite/tree/main/software/devicetrees
I dropped them in /boot/overlays and turned all 3 on in config.txt. It still reports smi problems.

alphafox02, 2023-01-29:
So I did get it working on a regular Pi OS build after turning on dtoverlay=smi and dtoverlay=smi-dev. I hadn’t noticed or used the dtbo files you’re pointing to. I’ll see what that changes, but I was successful in getting a connection to the Pi OS which I had running soapy remote server with sudo. Connection was made with SigDigger and I was able to demodulate an listen to an FM station.

alphafox02, 2023-01-29:
I don’t think those files are meant to go on the Pi, I think all you need to do is turn on the dtoverlay options and then you’ll find there’s a /dev/smi becomes available.

alphafox02, 2023-01-29:
Setting the following in the config.txt has exposed the /dev/smi 
Not sure what if anything else I did caused this to happen.
#dtparam=i2c_arm=on
#dtparam=spi=on
dtparam=i2c_vc=on
#dtoverlay=smi
#dtoverlay=smi-dev

closed 2023-01-29 by alphafox02