cariboulabs/cariboulite #30: HiF missing: only S1G is found

ixoo, 2023-01-30:
`SoapySDRUtil --probe` only returns S1G.

I also get a "wrong kernel driver" error during the SMI initialization:
01-29 20:12:05.420  1155  1155 I CARIBOULITE_CONFIG cariboulite_config_print_board_info@cariboulite_config.c:137 # Board Info - Product Type: CaribouLite FULL
01-29 20:12:05.420  1155  1155 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:313 initializing submodules
01-29 20:12:05.420  1155  1155 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:317 INIT FPGA SMI communication
01-29 20:12:05.421  1155  1155 I CARIBOU_SMI_Main caribou_smi_init@caribou_smi.c:141 initializing caribou_smi
01-29 20:12:05.421  1155  1155 E CARIBOU_SMI_Main caribou_smi_init@caribou_smi.c:185 failed reading native batch length, setting the default - this error is not fatal but we have wrong kernel drivers
01-29 20:12:05.421  1155  1155 I CARIBOU_SMI_Main caribou_smi_init@caribou_smi.c:190 Finished interogating 'smi' driver. Native batch length (bytes) = 524288

Any hint at what to look at/investigate would be great ;)

EelkeVisser, 2023-01-30:
Try `sudo SoapySDRUtil --find` It should list all devices and should show the second CaribouLite which is the other channel.

ixoo, 2023-01-30:
Indeed I can see it with --find, sorry for the confusion.

I am getting the following error message:
```
[...] E CARIBOU_SMI_Main caribou_smi_init@caribou_smi.c:185 failed reading native batch length, setting the default - this error is not fatal but we have wrong kernel drivers
```
Is this expected?

Closing the issue as I indeed see HiF now ;)

closed 2023-01-30 by ixoo