cariboulabs/cariboulite #31: zeromq required

edk4971, 2023-01-30:
clean install of raspberry pi OS, running install.sh on the R1 branch:

`lcd.c:9:10: fatal error: zmq.h: No such file or directory
    9 | #include <zmq.h>
      |          ^~~~~~~
compilation terminated.
make[2]: *** [src/production_utils/CMakeFiles/production_utils.dir/build.make:108: src/production_utils/CMakeFiles/production_utils.dir/lcd.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1090: src/production_utils/CMakeFiles/production_utils.dir/all] Error 2
make: *** [Makefile:149: all] Error 2`

Need to add [libzmq3-dev](https://packages.debian.org/bullseye/arm64/libzmq3-dev/filelist) to the list of packages pulled in from apt

alphafox02, 2023-02-02:
Even after having libzmq3-dev in place on a stock 64bit Pi OS or aarch64 22.04 build,

This is still happening with the R1 branch (for me) as of the most current pull. Are you finding that the same is not happening during the process of building cariboulite?

`Scanning dependencies of target cariboulite_prod
[ 88%] Building C object CMakeFiles/cariboulite_prod.dir/src/cariboulite_production.c.o
[ 90%] Linking CXX executable cariboulite_prod
/usr/bin/ld: src/production_utils/libproduction_utils.a(lcd.c.o): in function `lcd_close':
lcd.c:(.text+0x110): undefined reference to `zmq_send'
/usr/bin/ld: lcd.c:(.text+0x12c): undefined reference to `zmq_recv'
/usr/bin/ld: lcd.c:(.text+0x144): undefined reference to `zmq_close'
/usr/bin/ld: lcd.c:(.text+0x14c): undefined reference to `zmq_ctx_destroy'
/usr/bin/ld: src/production_utils/libproduction_utils.a(lcd.c.o): in function `lcd_init':
lcd.c:(.text+0x280): undefined reference to `zmq_ctx_new'
/usr/bin/ld: lcd.c:(.text+0x28c): undefined reference to `zmq_socket'
/usr/bin/ld: lcd.c:(.text+0x29c): undefined reference to `zmq_connect'
/usr/bin/ld: lcd.c:(.text+0x2ec): undefined reference to `zmq_close'
/usr/bin/ld: lcd.c:(.text+0x2f4): undefined reference to `zmq_ctx_destroy'
/usr/bin/ld: src/production_utils/libproduction_utils.a(lcd.c.o): in function `lcd_clear_screan':
lcd.c:(.text+0x3b4): undefined reference to `zmq_send'
/usr/bin/ld: lcd.c:(.text+0x3d0): undefined reference to `zmq_recv'
/usr/bin/ld: src/production_utils/libproduction_utils.a(lcd.c.o): in function `lcd_write':
lcd.c:(.text+0x430): undefined reference to `zmq_send'
/usr/bin/ld: lcd.c:(.text+0x444): undefined reference to `zmq_recv'
/usr/bin/ld: src/production_utils/libproduction_utils.a(lcd.c.o): in function `lcd_writeln':
lcd.c:(.text+0x4b4): undefined reference to `zmq_send'
/usr/bin/ld: lcd.c:(.text+0x4d0): undefined reference to `zmq_recv'
/usr/bin/ld: lcd.c:(.text+0x500): undefined reference to `zmq_send'
/usr/bin/ld: lcd.c:(.text+0x51c): undefined reference to `zmq_recv'
/usr/bin/ld: src/production_utils/libproduction_utils.a(lcd.c.o): in function `lcd_get_keys':
lcd.c:(.text+0x570): undefined reference to `zmq_send'
/usr/bin/ld: lcd.c:(.text+0x584): undefined reference to `zmq_recv'
/usr/bin/ld: src/production_utils/libproduction_utils.a(lcd.c.o): in function `lcd_set_params':
lcd.c:(.text+0x674): undefined reference to `zmq_send'
/usr/bin/ld: lcd.c:(.text+0x690): undefined reference to `zmq_recv'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cariboulite_prod.dir/build.make:116: cariboulite_prod] Error 1
make[1]: *** [CMakeFiles/Makefile2:422: CMakeFiles/cariboulite_prod.dir/all] Error 2
make: *** [Makefile:149: all] Error 2`

RobertSwirsky, 2023-02-02:
I had the same issue. I went back to "main" even though the developer says to use the R1 branch.

MartinHerren, 2023-02-02:
Got it to compile  on develop_R1 with the following micro-patch to solve the libzmq linking error (already mentioned in issue #28):
```
diff --git a/software/libcariboulite/src/production_utils/CMakeLists.txt b/software/libcariboulite/src/production_utils/CMakeLists.txt
index 1d64b3a..0342705 100644
--- a/software/libcariboulite/src/production_utils/CMakeLists.txt
+++ b/software/libcariboulite/src/production_utils/CMakeLists.txt
@@ -22,4 +22,5 @@ target_link_libraries(test_power_mon rt m pthread zmq ${SUPER_DIR}/zf_log/build/
 
 #Generate the static library from the sources
 add_library(production_utils STATIC ${SOURCES_LIB})
+target_link_libraries(production_utils zmq)
 target_include_directories(production_utils PUBLIC ${CMAKE_CURRENT_SOURCE_DIR})
```

alphafox02, 2023-02-03:
I’ll go give it a shot, thanks for the reminder. Have you noticed any difference between it and main, in terms of actually running applications and using the hat? Have type unloaded the stock kernel module and loaded the one built during the install process?

MartinHerren, 2023-02-04:
> It does compile, so thank you for that, but for me the end result in terms of using the hardware is the same. At least on aarch64 with or without using the custom kernel module recompiled for this target. I’ll switch back to the Pi OS stock tomorrow, but so far it’s been the same results there too.

No i haven't managed to get as far as using anything. 

On `develop_R1` branch it builds a `cariboulite_app` that seems to start and probe the device correctly. `setup.sh` does also a lot of other setup (kernel module, overlay ?). `SoapyCariboulite` doesn't get built on this branch.

On `master` branch it builds a `cariboulite_app` that doesn't work, but builds and installs `SoapyCariboulite`  that correctly detects the devices. Managed to start a SoapySDRServer on it but when i connect to any device it immediately segfaults.

All tests done so far on a Pi Zero W so 32 bit OS. Will try again on a Pi Zero 2W with a 64 bit OS as it seems the focus of the developers now. As soon as i have a little time.

edegraaff, 2023-02-07:
> No i haven't managed to get as far as using anything.
>
> On `develop_R1` branch it builds a `cariboulite_app` that seems to start and probe the device correctly. `setup.sh` does also a lot of other [...]

I am also using pi zero and 32 bit os, if you have a procedure that is working please feel free to share so i can try to get the whol working.

meexmachina, 2023-02-09:
zeroMQ requirement was removed in the cmake file. it was intended for production software, which is part of the repo.
libzmq3-dev can be installed but it doesn't resolve the dependencies issue for some reason. we built it from source for the production phase but this package is currently redundant in the mainline software. thus was removed from building.

closed 2023-02-10 by meexmachina