cariboulabs/cariboulite #32: generate_binary_blob built for ELF 32

edk4971, 2023-02-01:
Trying to compile on Raspberry Pi OS (64 bit) results in the following error
`make[2]: ../../../../../utils/generate_bin_blob: No such file or directory
make[2]: *** [CMakeFiles/smi_stream_dev.dir/build.make:88: smi_stream_dev] Error 127
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/smi_stream_dev.dir/all] Error 2
make: *** [Makefile:103: all] Error 2`

Because generate_bin_blob was built against arch armhf
`~/projects/cariboulite/software/utils $ file generate_bin_blob
generate_bin_blob: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=573ee2b3695f08006e45fa327db5f261bc70f094, not stripped`

The solution if you have this problem is to install the armhf libc
`sudo dpkg --add-architecture armhf && sudo apt update && sudo apt-get install libc6:armhf`

MartinHerren, 2023-02-05:
The correct fix would be not to add binaries to git but to build them directly as the source for these binaries is part of this repo.

You could generate them yourself for the correct arch by building them first in the utils folder.

meexmachina, 2023-02-09:
the new install shell script resolves this issue as it recompiles the needed dependencies before building the mainline software.

There still is an issue with the kernel module compilation: The smi kernel module is an out of tree module that is built during installation. For it to be built, kernel headers are required (which are apt-gotten during installations). kernel modules unlike regular programs need to match the users kernel version. otherwise, they won't load. Until the revised SMI streaming module will be a part of the kernel tree (for RPI), this compilation phase may require "re-installation / recompilation" for each new kernel version.