cariboulabs/cariboulite #35: List of dependencies not found

blacktigersoftware, 2023-02-02:
As I'm doing a workaround for the missing setup.sh, the instructions seem to require building some stuff.   From a Stock Raspbian 64 Full Install, I seem to be missing some build tools.  Most importantly, is cmake, which was not installed.

I'm hesitant from going further, because it always leads down a hole of installing stuff, over and over.  I'm sure someone has it all figured out, by now. 

Will the "setup.sh" install these?   If not, could we have a list of build dependencies to install starting from a stock raspbian install?

alphafox02, 2023-02-02:
Switch to the dev r2 branch with git checkout. That currently has the script and newer code

blacktigersoftware, 2023-02-03:
dev r2?????   I see an install.sh in develop_r1, but it uses cmake without checking if it first exists........

alphafox02, 2023-02-03:
Sorry, I was sorta distracted when I replied. You’re correct, the install script partially works but sadly not completely. I’ve had to use a combination of the main branch and develop_r1 but even then I’ve only managed to connect and use the caribou with sudo soapy server running on the Pi and then connecting to it from SigDigger on another laptop on the network.

alphafox02, 2023-02-03:
Check the pull requests, there’s some work someone’s done to clean up and make the install script a little better. I just used the zmq pull request to get the libcaribou software to build without errors.

edegraaff, 2023-02-06:
All, suggestion since there is a repo with installation documentation and so on. If the software is under development please make a note there so others do not spend time in documentation that is outdated. Since the pi-zero is not that fast some good pereparation is needed :-) i asumed the documentation in the readme.md is or complete or make some notes where to look for more info.