cariboulabs/cariboulite #36: Self test returns not AT86RF215 modem

justinh24, 2023-02-02:
Running self test usually fails due to a mismatched modem.  The exception is if I perform a receive I/Q test and then run the self test it passes.  Running the self-test a second time I get the error message again:

02-02 16:42:25.355   692   692 I CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:457 Testing modem communication and versions
02-02 16:42:25.355   692   692 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:463 The assembled modem is not AT86RF215 / IQ variant (product number: 0x0d)
02-02 16:42:25.355   692   692 I CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:472 Testing mixer communication and versions
02-02 16:42:25.356   692   692 I CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:484 Testing smi communication
02-02 16:42:25.356   692   692 E CARIBOULITE Setup cariboulite_self_test@cariboulite_setup.c:493 Self-test process finished with errors
	MODEM Version: not AT86RF215 IQ capable modem (product number: 0x0d, versions 03)

meexmachina, 2023-02-02:
checkout the develop_R1 branch

meexmachina, 2023-02-02:
When the program starts it self test everything and shows:
```
02-02 19:45:37.155  2206  2206 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:288 INIT MODEM - AT86RF215
02-02 19:45:37.155  2206  2206 I AT86RF215_Main at86rf215_init@at86rf215.c:147 configuring reset and irq pins
02-02 19:45:37.155  2206  2206 I AT86RF215_Main at86rf215_init@at86rf215.c:155 Adding chip definition to io_utils_spi
02-02 19:45:37.156  2206  2206 I AT86RF215_Main at86rf215_init@at86rf215.c:184 Modem identity: Version: 03, Product: 35
02-02 19:45:37.156  2206  2207 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO09: Transceiver ready
02-02 19:45:37.156  2206  2213 I CARIBOU_SMI_Utils smi_utils_set_realtime_priority@utils.c:30 Trying to set thread realtime prio = 97

```

02-02 19:45:37.156  2206  2206 I AT86RF215_Main at86rf215_init@at86rf215.c:184 Modem identity: Version: 03, Product: 35

And indeed when the selftest is running from the CLI it shows your result. A software problem. I will check shortly

meexmachina, 2023-02-02:
found the problem. It returns the register address 0x0d (PN) rather than the data in this case. thanks a lot!

meexmachina, 2023-02-02:
a workaround pushed into develop_r1

closed 2023-02-02 by meexmachina