cariboulabs/cariboulite #40: [Suggestion] rpisdr fork

zbyrek, 2023-02-04:
Hello,
I got issues and many other users also to make it work, I know it is still in dev phase so my idea is that maybe you will provide fork of rpisdr image (or raspbian with sdr software like sdr++) with all drivers and libs configured which will allow quick start for users to check how it works.

blacktigersoftware, 2023-02-08:
lol.  I got everything to compile without errors, but no luck in getting SoapySDRUtil to find anything......

blacktigersoftware, 2023-02-08:
honestly, at this point, all I really *need* at this point is for a remote endpoint that works with Windows SDR software like SDRSharp.   If someone could be so kind to create an image that can be installed via Etcher or Raspberry PI Image Writer, it would be SOOOO appreciated.

edegraaff, 2023-02-08:
an route could be the use of rtl_tcp but in that case the rtl_tcp need to be able to recognise the HAT. I hope the dev team will come with some updates soon to show is it was a good investment.

edegraaff, 2023-02-24:
> honestly, at this point, all I really _need_ at this point is for a remote endpoint that works with Windows SDR software like SDRSharp. If someone could be so kind to create an image that can be installed via Etcher or Raspberry PI Image Writer, it would be SOOOO appreciated.

There is a more or less working solution https://cariboulitesdr.freeforums.net/thread/2/headless-setup