cariboulabs/cariboulite #41: Misleading Instructions

G0MJW, 2023-02-04:
Please can we change the readme as it is unhelpful to beginners. I understand it isn't ready yet, it might be best to say that at the top and ask non-developers to wait., we waited a year for the hardware shortage to resolve so a few more weeks for software isn't going to be an issue.. Also it needs to state the requirements, which OS to install on, what packages need to be installed (PIOS lite doesn't even come with cmake or git) etc. need to be made clearer.

zhangtemplar, 2023-02-04:
I agree, at least mention use branch `develop_R1`

edegraaff, 2023-02-06:
> I agree, at least mention use branch `develop_R1`

I received the hardware today, and i am preparing my pi zero. I expected to folow the installation instructions. Do we have an issue here and should i use another installation set?

cpm5280, 2023-02-13:
It's a little bit baffling to me that these wrong instructions remain up. Makes me wonder if anyone's in charge or paying attention?

When you ship a product to users, you should be ready to help them use that product. Even if it's an experimental thing, as this clearly is, ensuring that there is a clear path to *correct* instructions is sort of the bare minimum.

G0MJW, 2023-02-17:
They have now been updated.

closed 2023-02-17 by G0MJW