cariboulabs/cariboulite #43: Which Raspberry PI OS to install ?

hb9dug, 2023-02-07:
What is the recommended OS for a successful installation of CaribouLite?

I'm using PiSDR Version 6.1 (64-bits) and today's install script (develop_R1) went through without error...

How do I check the correct installation of CaribouLite?

Thanks in advance for your support.
Michel

RobertSwirsky, 2023-02-08:
I'm having better luck with 64-bit, though I haven't gotten it to run stabily yet (core dump when changing frequencies while receiveing)

Also, I discovered that GnuRadio 3.9 and 3.10 aren't building on 32-bit RaspberryPi OS so if you want to use GnuRadio, that's a factor.

alphafox02, 2023-02-09:
Using DragonOS Pi64. If you can follow and do this with your PiSDR, you're probably about as far as anyone else.
[
](https://youtu.be/YVl0995zDS4)

rhammock1, 2023-02-12:
I used PiSDR 64-bit on a raspberry pi zero 2, and everything worked correctly. Switched to 64-bit Raspberry Pi OS on an RPI 4 and everything compiled once the proper dependencies were installed. 

I checked the installation was successful by running the `cariboulite_app` app that was created after compiling and going through the different menus. I have an rtl-sdr I had connected to a separate laptop that could receive signals sent by the Cariboulite from the `cariboulite_app`. (Menu option 8, then setting a high or low frequency, then setting CW high or low to actually transmit the signal)

hb9dug, 2023-02-14:
I successfully installed Cariboulite software on a RPI 4 this morning using the following procedure:

OS :  2022-09-22-raspios-bullseye-arm64.img

sudo apt update
sudo apt upgrade
sudo reboot

mkdir ~/projects
cd ~/projects
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite

git checkout develop_R1

sudo ./install.sh

addded 'dtparam=i2c_vc=on' to the '/boot/config.txt' file

sudo reboot

cd ~/projects/cariboulite/build
sudo ./cariboulite_test_app

sudo SoapySDRUtil --probe="driver=Cariboulite,device_id=0"
sudo SoapySDRUtil --probe="driver=Cariboulite,device_id=1"

Bonne bidouille !

pocki, 2023-02-14:
I can confirm @hb9dug install procedure also with Raspberry Pi OS Lite (arm64).
Extra step at beginning: sudo apt install git

hb9dug, 2023-02-15:
I successfully installed Cariboulite software on a RPI 4 this morning following the procedure on the Main branch on a PiSDR OS (2022-01-07-PiSDR-vanilla.img).

I run into problem trying executing GQRX as root to bypass the issue described in 3. sudoing , I get "Pulseaudio error: Connection refused" ...  :-(

With SDR++, at some point of the boot process of SDR++, I get a warning "ConfigManager locked, waiting...  :-(

Will have to wait that the "io_utils" sub-module is re-designed to access the hardware through the "gpiomem" and "spidev" modules.  

Thanks to the developer, we are getting closer to the goal!

fransla, 2023-02-16:
To run SDR++ you need to start pulseaudio in a separate window as root first.

Create a config folder if it does not exist for root

```
sudo su
cd
mkdir .config
exit
```

Fix permissions:

```
sudo adduser root pulse-access
```

Start pulseaudio in another window:

```
sudo pulseaudio --system=true
```

Now you can run SDR++ as root without hanging at startup:

```
sudo sdrpp
```

hb9dug, 2023-02-17:
@fransla  - Thank you for the tips !  
Now SDR++ works but unfortunately I have no sound...
Certainly related to the error message when I start pulseaudio

pi@pisdr:~ $ sudo pulseaudio --system=true
W: [pulseaudio] main.c: Running in system mode, but --disallow-exit not set.
W: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set.
N: [pulseaudio] main.c: Running in system mode, forcibly disabling SHM mode.
N: [pulseaudio] main.c: Running in system mode, forcibly disabling exit idle time.
W: [pulseaudio] main.c: Home directory of user 'pulse' is not '/var/run/pulse', ignoring.
W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please make sure that you actually do want to do that.
W: [pulseaudio] main.c: Please read http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/ for an explanation why system mode is usually a bad idea.
E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="2" name="platform-fef05700.hdmi" card_name="alsa_card.platform-fef05700.hdmi" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
W: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.config/pulse/cookie': No such file or directory
W: [pulseaudio] authkey.c: Failed to load authentication key '/var/run/pulse/.config/pulse/cookie': No such file or directory
W: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.pulse-cookie': No such file or directory
W: [pulseaudio] authkey.c: Failed to load authentication key '/var/run/pulse/.pulse-cookie': No such file or directory

Thank you in advance for your help