cariboulabs/cariboulite #45: Hardware Issue Question

siffland, 2023-02-08:
I just got my Caribou Lite (6GHZ R2.8) from mouser.  I am having issues getting it working.  I just want to verify that there should be no components soldered in the "MXR IN" box on the top of the unit and nothing soldered in on the MIXER part of the unit on the bottom (All picture I see have these fully populated).  I know it went through a lot of hardware revisions.

I am probably just not setting up the software correctly but want to make sure I am not chasing my tail and just didn't get an unfinished piece of hardware.

Thanks,
Sean

sorvis, 2023-02-09:
@siffland that sounds like what my raspberry pi 4 was doing as well. It was an older image that I had just ran updates on. I tried a pre-made image for SDR and that appears to at least boot.

https://github.com/luigifcruz/pisdr-image

Still the same as everyone else though with not having a working setup, but nice having the SDR++ pre-compiled though after the driver starts working.

siffland, 2023-02-09:
OK, on a fresh card with PiSDR imaged i was able to get get the software compiled with the following, after the final reboot the smi_stream_dev module loads (can verify with dmesg).  I2C can actually see the device.  Have not had a chance to test actually receiving with the actual hardware yet.  

`
sudo apt update
sudo apt dist-upgrade

echo "dtparam=i2c_vc=on" | sudo tee -a /boot/config.txt

reboot

mkdir ~/projects
cd ~/projects
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite
git checkout remotes/origin/develop_R1

sed -i 's/add_subdirectory(src\/iir EXCLUDE_FROM_ALL)/add_subdirectory(src\/iir)/g' software/libcariboulite/CMakeLists.txt

sudo ./install.sh

sudo cp /home/pi/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build/smi_stream_dev.ko /lib/modules/5.15.84-v8+/kernel/drivers/misc/

sudo depmod

reboot
`

Back to my original question are the MXR IN and MIXER area on the bottom and top of the board supposed to be blank with no components?

RobertSwirsky, 2023-02-09:
> Back to my original question are the MXR IN and MIXER area on the bottom and top of the board supposed to be blank with no components?

There are two versions of the board, the "full" one, and the Sub-GHz one. You may have the "S1G" version of it.

RobertSwirsky, 2023-02-09:
I'm having no luck. I tried following @siffland 's instructions and it build with no errors, but I got this:

```
pi@raspberrypi:~ $ dmesg | grep smi
[    5.162270] smi-bcm2835 fe600000.smi: initialised
[    5.873641] smi-dev-bcm2835 soc:smi_dev: initialised
[    5.921170] smi_stream_dev: loading out-of-tree module taints kernel.
pi@raspberrypi:~ $ sudo SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found!
pi@raspberrypi:~ $
```

(Your sed didn't work, so I edited the "EXCLUDE_FROM_ALL" out of that file by hand.)

siffland, 2023-02-09:
I used the PiSDR image, it already has SoapySDR installed.  I fixed the sed command, it did work, I failed to use the code block and it displayed incorrectly.  Fixing it now.

Well that is unfortunate that I appear to have the S1G since i paid for the full one.

Even after getting it compiled i am still having issues

```
 sudo ./cariboulite_app
02-09 15:11:10.449  2380  2380 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-09 15:11:10.450  2380  2380 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-09 15:11:10.450  2380  2380 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-09 15:11:10.450  2380  2380 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-09 15:11:10.450  2380  2380 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-09 15:11:10.599  2380  2380 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
02-09 15:11:10.599  2380  2380 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
02-09 15:11:10.599  2380  2380 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
02-09 15:11:10.599  2380  2380 I FPGA caribou_fpga_init@caribou_fpga.c:115 configuring reset and irq pins
02-09 15:11:10.599  2380  2380 I FPGA caribou_fpga_init@caribou_fpga.c:123 Initializing io_utils_spi
02-09 15:11:10.599  2380  2380 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
02-09 15:11:10.599  2380  2380 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
02-09 15:11:10.600  2380  2380 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
        System Version: 01
        Manu. ID: 01
        Sys. Ctrl Version: 01
        IO Ctrl Version: 01
        SMI Ctrl Version: 01
02-09 15:11:10.600  2380  2380 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:187 FPGA already operational - not programming (use 'force_prog=true' to force update)
02-09 15:11:10.600  2380  2380 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [0,1,1,1]
02-09 15:11:10.600  2380  2380 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
02-09 15:11:10.601  2380  2380 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
02-09 15:11:10.601  2380  2380 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
02-09 15:11:10.601  2380  2380 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0002, Numeric: 2
02-09 15:11:10.601  2380  2380 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
02-09 15:11:10.601  2380  2380 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: abdd597c-ea54-447a-b3a0-8b780483831b, Numeric serial: 0xF2299A9E
02-09 15:11:10.601  2380  2380 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
02-09 15:11:10.601  2380  2380 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:754 # Board Info - Product Type: CaribouLite ISM
02-09 15:11:10.601  2380  2380 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
02-09 15:11:10.601  2380  2380 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
02-09 15:11:10.601  2380  2380 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:410 initializing caribou_smi
02-09 15:11:10.602  2380  2380 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
02-09 15:11:10.630  2380  2380 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:428 couldn't open smi driver file '/dev/smi'
02-09 15:11:10.630  2380  2380 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
02-09 15:11:10.630  2380  2380 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
02-09 15:11:10.630  2380  2380 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-09 15:11:10.630  2380  2380 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-09 15:11:10.631  2380  2380 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-09 15:11:10.631  2380  2380 E FPGA caribou_fpga_close@caribou_fpga.c:224 caribou_fpga_close: dev not initialized
02-09 15:11:10.631  2380  2380 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-09 15:11:10.631  2380  2380 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
02-09 15:11:10.647  2380  2380 E CARIBOULITE Main main@cariboulite_app.c:68 driver init failed, terminating...
```

sorvis, 2023-02-09:
Is that empty spot to the left what would indicate getting the S1G version? I should have gotten the full version as well.

[image]

siffland, 2023-02-09:
So this is interesting, after compiling and trying to run the cariboulite_app, i get the above error.  However if i execute

```
i2cdetect -y 2
```
then 

```
sudo build/cariboulite_app
```
The cariboulite_app runs fine.  Soapy still cannot find device.  I must be missing something in my config.txt to initialize the device.

RobertSwirsky, 2023-02-09:
> Is that empty spot to the left what would indicate getting the S1G version? I should have gotten the full version as well.

Those empty 12 pads all the way on the left? That's correct. Those are additional connecttions for the FPGA in case you want some custom I/O for it. Your board should have come with a header you can solder for those pads. 

Your board looks exactly like mine. (Not that i've been able to get mine to do anything!)

siffland, 2023-02-09:
No I mean the entire MXR IN section, and the MIXER on the bottom

[image]

RobertSwirsky, 2023-02-09:
No, that's not right @siffland! Look at the scheatics. It's either an S1G only board, or a production error!

blacktigersoftware, 2023-02-10:
I have two of the 6 Ghz version and both are the same as yours.   I've gotten pretty much as far as you have, cariboulite_app works, but nothing else.

blacktigersoftware, 2023-02-10:
This is what I get, for now:

root@caribou-pi:~# uname -a
Linux caribou-pi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
root@caribou-pi:~# lsmod | grep smi
bcm2835_smi_dev        16384  0
bcm2835_smi            20480  1 bcm2835_smi_dev
root@caribou-pi:~#

#### changed dirs.  Notice I copyied "smi_stream_dev.ko" here, myself.

root@caribou-pi:/home/kend/sdr/cariboulite/build# pwd
/home/kend/sdr/cariboulite/build
root@caribou-pi:/home/kend/sdr/cariboulite/build# ls -al
total 2344
drwxr-xr-x  5 kend kend   4096 Feb  9 01:52 .
drwxr-xr-x 11 kend kend   4096 Feb  9 01:31 ..
-rwxr-xr-x  1 kend kend 859760 Feb  9 01:50 cariboulite_app
-rwxr-xr-x  1 kend kend 909232 Feb  9 01:50 cariboulite_prod
-rw-r--r--  1 kend kend  17955 Feb  9 01:50 CMakeCache.txt
drwxr-xr-x  9 kend kend   4096 Feb  9 01:59 CMakeFiles
-rw-r--r--  1 kend kend   2299 Feb  9 01:50 cmake_install.cmake
-rw-r--r--  1 root root     31 Feb  9 01:51 install_manifest.txt
-rw-r--r--  1 kend kend 103546 Feb  9 01:50 libcariboulite.a
-rw-r--r--  1 kend kend  32987 Feb  9 01:50 Makefile
-rw-r--r--  1 kend kend 436320 Feb  9 01:52 smi_stream_dev.ko
drwxr-xr-x 14 kend kend   4096 Feb  9 01:50 src
drwxr-xr-x  2 kend kend   4096 Feb  9 01:50 test

root@caribou-pi:/home/kend/sdr/cariboulite/build# ./cariboulite_app
02-09 19:35:38.141  2907  2907 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-09 19:35:38.142  2907  2907 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-09 19:35:38.143  2907  2907 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-09 19:35:38.144  2907  2907 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-09 19:35:38.144  2907  2907 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-09 19:35:38.274  2907  2907 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
02-09 19:35:38.274  2907  2907 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
02-09 19:35:38.275  2907  2907 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
02-09 19:35:38.275  2907  2907 I FPGA caribou_fpga_init@caribou_fpga.c:115 configuring reset and irq pins
02-09 19:35:38.276  2907  2907 I FPGA caribou_fpga_init@caribou_fpga.c:123 Initializing io_utils_spi
02-09 19:35:38.276  2907  2907 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
02-09 19:35:38.276  2907  2907 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
        System Version: FF
        Manu. ID: FF
        Sys. Ctrl Version: FF
        IO Ctrl Version: FF
        SMI Ctrl Version: FF
02-09 19:35:38.278  2907  2907 I CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
02-09 19:35:38.279  2907  2907 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:259 Sending bitstream of size 32220
02-09 19:35:39.633  2907  2907 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:281 bitstream sent 32220 bytes
02-09 19:35:39.634  2907  2907 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
02-09 19:35:39.634  2907  2907 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:203 0 dummy clocks sent
02-09 19:35:39.634  2907  2907 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:291 FPGA programming - Success!

FPGA Versions:
        System Version: 01
        Manu. ID: 01
        Sys. Ctrl Version: 01
        IO Ctrl Version: 01
        SMI Ctrl Version: 01
02-09 19:35:39.735  2907  2907 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
02-09 19:35:39.735  2907  2907 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
02-09 19:35:39.736  2907  2907 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
02-09 19:35:39.736  2907  2907 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
02-09 19:35:39.736  2907  2907 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
02-09 19:35:39.736  2907  2907 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
02-09 19:35:39.736  2907  2907 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: 67340ff2-d1b8-4f6d-bd1a-54fbf5cca155, Numeric serial: 0x0B961406
02-09 19:35:39.737  2907  2907 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
02-09 19:35:39.737  2907  2907 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
[... 181 lines cut ...]
02-09 19:37:28.786  3517  3551 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:960 SMI reading operation returned timeout
Rx N=0 SC=0: [0, 0] [0, 0] [0, 0] [0, 0] . . . [0, 0] [0, 0] [4098, 8] [0, 0]
02-09 19:37:28.786  3517  3551 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:825 Activating channel 1, dir = RX, activate = 1
02-09 19:37:28.786  3517  3551 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
[... cut ...]
02-09 19:37:28.789  3517  3551 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (2)
02-09 19:37:28.789  3517  3551 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:956 SMI reading operation failed
reader thread failed to read SMI!

root@caribou-pi:/home/kend/sdr/cariboulite/build# ^C
root@caribou-pi:/home/kend/sdr/cariboulite/build# lsmod | grep smi
smi_stream_dev         20480  0
bcm2835_smi            20480  1 smi_stream_dev

siffland, 2023-02-10:
Seems to work, however the first time i choose 9 then 2 i got this error:

```
 Choice: 2
        Parameters:
        [1] Ch1 (899.99987 MHz) RX Not Active
        [2] Ch2 (2399.99974 MHz) RX Active
        [3] Push Debug Not Active
        [4] Pull Debug Not Active
        [5] LFSR Debug Not Active
        [99] Return to main menu
        Choice: 02-09 20:34:19.393  1210  1219 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:830 Activating channel 1, dir = RX, activate = 1
02-09 20:34:19.394  1210  1211 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO24: Transceiver ready
02-09 20:34:19.394  1210  1219 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:839 Setup Modem state tx_prep
02-09 20:34:19.395  1210  1219 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:865 Setup Modem state cmd_rx
02-09 20:34:19.422  1210  1219 E CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:962 SMI reading operation failed
reader thread failed to read SMI!
Rx N=131072 SC=18: [256, -896] [-2816, -896] [2304, -3968] [3328, -896] . . . [0, 0] [0, 0] [0, 0] [0, 0] 
Rx N=131072 SC=3: [-2816, -2817] [2304, -2944] [1280, 128] [1280, -896] . . . [2431, -1793] [-641, 3327] [3328, 2303] [3328, 3200] 
Rx N=131072 SC=3: [3328, -3968] [3328, 2176] [256, -769] [-1792, -2817] . . . [2431, -1793] [-641, 3327] [3328, 2303] [3328, 3200] 
Rx N=131072 SC=4: [1280, -3841] [-2816, -2817] [-1792, 1279] [1407, -769] . . . [2431, -1793] [-641, 3327] [3328, 2303] [3328, 3200] 
Rx N=131072 SC=3: [-2816, -1920] [2304, -2944] [-3840, -2944] [2431, -1920] . . . [1407, 2303] [-641, 3327] [3328, 2303] [3328, 3200] 
```

second time the "SMI reading operation failed" was gone.

```
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Diginal I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [99]  Quit
    Choice:   9

=====================================
02-09 20:30:52.328 12568 12568 D CARIBOULITE Radio cariboulite_radio_set_frequency@cariboulite_radio.c:799 Frequency setting CH: 0, Wanted: 900000000.00 Hz, Set: 0.00 Hz (MOD: 899999872.00, MIX: 0.00)
02-09 20:30:52.328 12568 12568 D CARIBOULITE Radio cariboulite_radio_set_frequency@cariboulite_radio.c:799 Frequency setting CH: 1, Wanted: 2400000000.00 Hz, Set: 0.00 Hz (MOD: 2399999744.00, MIX: 0.00)
Entering sampling thread
02-09 20:30:52.328 12568 12568 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:830 Activating channel 0, dir = RX, activate = 0
02-09 20:30:52.329 12568 12568 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:839 Setup Modem state tx_prep
02-09 20:30:52.329 12568 12568 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:850 Setup Modem state trx_off
02-09 20:30:52.329 12568 12568 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:830 Activating channel 1, dir = RX, activate = 0
02-09 20:30:52.330 12568 12569 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO09: Transceiver ready
02-09 20:30:52.330 12568 12569 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO24: Transceiver ready
02-09 20:30:52.331 12568 12568 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:839 Setup Modem state tx_prep
02-09 20:30:52.331 12568 12568 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:850 Setup Modem state trx_off
        Parameters:
        [1] Ch1 (899.99987 MHz) RX Not Active
        [2] Ch2 (2399.99974 MHz) RX Not Active
        [3] Push Debug Not Active
        [4] Pull Debug Not Active
        [5] LFSR Debug Not Active
        [99] Return to main menu
        Choice: 2
        Parameters:
        [1] Ch1 (899.99987 MHz) RX Not Active
        [2] Ch2 (2399.99974 MHz) RX Active
        [3] Push Debug Not Active
        [4] Pull Debug Not Active
        [5] LFSR Debug Not Active
        [99] Return to main menu
        Choice: 02-09 20:30:55.585 12568 12577 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:830 Activating channel 1, dir = RX, activate = 1
02-09 20:30:55.586 12568 12569 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO24: Transceiver ready
02-09 20:30:55.586 12568 12577 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:839 Setup Modem state tx_prep
02-09 20:30:55.586 12568 12577 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:865 Setup Modem state cmd_rx
Rx N=131072 SC=8: [2304, 3200] [-768, 1152] [1280, 2176] [-3713, 128] . . . [0, 0] [0, 0] [0, 0] [0, 0] 
Rx N=131072 SC=8: [2304, 2303] [-2816, -1793] [-3840, 255] [256, 255] . . . [0, 0] [0, 0] [0, 0] [0, 0] 
Rx N=131072 SC=3: [2176, 1407] [-896, 3455] [255, -641] [-2817, -3713] . . . [-896, -2689] [0, 0] [0, 0] [0, 0] 
Rx N=131072 SC=3: [-1920, 3328] [-2817, -2816] [-3841, 3328] [-1793, 1407] . . . [-896, -2689] [0, 0] [0, 0] [0, 0] 
Rx N=131072 SC=3: [-1792, -2817] [2304, 3327] [-3840, 3327] [3328, -2817] . . . [-896, -2689] [0, 0] [0, 0] [0, 0] 
Rx N=131072 SC=3: [-768, 255] [2304, 3200] [1280, 3200] [-768, 128] . . . [1280, -3968] [0, 0] [0, 0] [0, 0] 
[... cut ...]
Rx N=131072 SC=3: [-1920, 2431] [-3968, 3455] [1152, -1665] [-2944, 3328] . . . [1279, -1665] [-3841, 383] [-2816, -1920] [-3840, 1152] 
Rx N=131072 SC=3: [128, -3840] [3200, 1280] [1152, -2816] [-2944, -1792] . . . [1279, -1665] [-3841, 383] [-2816, -1920] [-3840, 1152] 
Rx N=131072 SC=4: [2176, -3713] [128, -3713] [1152, -2689] [255, -2689] . . . [1279, -1665] [-3841, 383] [-2816, -1920] [-3840, 1152] 
Rx N=131072 SC=3: [-2816, 255] [-1792, 3327] [2304, 1152] [256, 3200] . . . [1279, -1665] [-3841, 383] [-2816, -1920] [-3840, 1152] 
```

blacktigersoftware, 2023-02-10:
@siffland Just tried it again.  Multiple attempts, you've pulled ahead in getting things to work. 👍  
How did you stop the read?   I just "ctl-c", but it drops me out of the app, as well.  
Also tried 'soft reset'.   Tried multiple times of Ch2 and Ch1, no change.

meexmachina, 2023-02-10:
@siffland hello and sorry for the late reply. indeed - this is the ISM board. not 6G. I suppose you ordered the 6G and if so, you are not alone. A few additional supporters received the wrong board. If you still have the package, I would love you to make its snapshot and send it to me here so that I will be able to check where the mistake happened (production facility? packing? mouser fulfillment?)

Thank you and I aplogize.

blacktigersoftware, 2023-02-10:
@meexmachina - I assume this goes for everybody?  I ordered two 6 Ghz boards and it looks just like @siffland .

meexmachina, 2023-02-10:
I'm waiting for cs response on that. It is a weird situation indeed

meexmachina, 2023-02-10:
Can you send me your order number? I will pass to cs

siffland, 2023-02-10:
I e-mailed these pics but will post here as well.  The static bag does say 2.4GHz, the Mouser bag 6G.  I only read the Mouser bag and then like a kid at Christmas i opened the static bag and began playing.

[image]
[image]

blacktigersoftware, 2023-02-10:
Both my static bags say 6 Ghz.

[image]

slesto, 2023-02-10:
@siffland - Thanks for all your effort so far. I can now get the app running and it seems to work thanks to you. Stinks that it looks like Mouser messed up your order but hopefully it gets corrected soon.  I will watch this thread closely as I am eager to get this recognized by Soapy

wrju333, 2023-02-15:
I don'tn know what I'm doing, but if I do a `sudo killall SoapySDRServer,` then `sudo SDRUtil --find` it finds the both radios, then I can do a `sudo SoapySDRServer --bind `and it will let SDR++ see the soapy devices, the terminal from soapysrd shows that it can modify the frequency, and choose each radio, but I get no waterfall, sound, or any other indication that its working.  Don't know why I have to run with sudo, except that the new raspian installs don't use the user pi anymore.  I also copied the soapycariboulite.so to the soapysdr modules folder before it would work.  Don't know if any of this helps or if I'm a mile behind you all.

mishafarms, 2023-02-20:
It appears that I also got a Caribou ISM (2.4GHz) in a 6G outer bag.
Order 149116 was for a 6GHz.

Please let me know how this is going to be addressed.
Michael

mishafarms, 2023-02-20:
@wrju333
 > I don't know what I'm doing, but if I do a `sudo killall SoapySDRServer,` then `sudo SDRUtil --find` it finds the both radios, then I can do a `sudo SoapySDRServer --bind `and it will let SDR++ see the soapy devices, the terminal from soapysrd shows that it can modify the frequency, and choose each radio, but I get no waterfall, sound, or any other indication that its working. Don't know why I have to run with sudo, except that the new raspian installs don't use the user pi anymore. I also copied the soapycariboulite.so to the soapysdr modules folder before it would work. Don't know if any of this helps or if I'm a mile behind you all.

In the SDR++ display, go to the "sink" tab and select None. Until I did that I didn't see a waterfall either. After that I did.
Michael

rhammock1, 2023-03-13:
Has anyone who received the wrong board heard anything about a resolution?

siffland, 2023-03-13:
I received the wrong board and have heard nothing since they confirmed in this issue that wrong boards went out.

mishafarms, 2023-03-14:
I contacted Crowd supply directly through a web support inquiry on 2/23. 
It appears that my replacement will be delivered
tomorrow.

Michael

great9, 2023-03-17:
Running sudo SoapySDRUtil -probe gives me this, the important part is that it says: `FPGA programming - Success!` in caribou_prog and then `FPGA programming - Failed` in caribou_fpga
It all results in `[ERROR] cariboulite_init_driver() failed`

```
[INFO] SoapyCaribouliteSession, sessionCount: 0
03-17 11:10:21.844  3851  3851 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
03-17 11:10:21.845  3851  3851 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
03-17 11:10:21.845  3851  3851 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
03-17 11:10:21.845  3851  3851 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
03-17 11:10:21.845  3851  3851 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
03-17 11:10:21.960  3851  3851 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
03-17 11:10:21.960  3851  3851 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
03-17 11:10:21.960  3851  3851 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
03-17 11:10:21.960  3851  3851 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
03-17 11:10:21.960  3851  3851 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
03-17 11:10:21.961  3851  3851 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
03-17 11:10:21.961  3851  3851 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
03-17 11:10:21.961  3851  3851 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
        System Version: 00
        Manu. ID: 00
        Sys. Ctrl Version: 00
        IO Ctrl Version: 00
        SMI Ctrl Version: 00
03-17 11:10:21.961  3851  3851 I CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
03-17 11:10:21.963  3851  3851 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:259 Sending bitstream of size 32220
03-17 11:10:23.466  3851  3851 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:281 bitstream sent 32220 bytes
03-17 11:10:23.466  3851  3851 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
03-17 11:10:23.466  3851  3851 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:203 0 dummy clocks sent
03-17 11:10:23.466  3851  3851 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:291 FPGA programming - Success!

FPGA Versions:
        System Version: 00
        Manu. ID: 00
        Sys. Ctrl Version: 00
        IO Ctrl Version: 00
        SMI Ctrl Version: 00
03-17 11:10:23.567  3851  3851 E FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:182 Programming failed
03-17 11:10:23.567  3851  3851 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:599 FPGA programming failed
03-17 11:10:23.567  3851  3851 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
03-17 11:10:23.567  3851  3851 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
03-17 11:10:23.567  3851  3851 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
03-17 11:10:23.567  3851  3851 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
03-17 11:10:23.567  3851  3851 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
[ERROR] cariboulite_init_driver() failed
03-17 11:10:23.582  3851  3851 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
Printing 'findCariboulite' Request:
Found device 0
  channel = S1G
  device_id = 0
  driver = Cariboulite
  label = CaribouLite S1G[9e02dae0]
  name = CaribouLite RPI Hat
  serial = ##########
  uuid = #########
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 1
  channel = HiF
  device_id = 1
  driver = Cariboulite
  label = CaribouLite HiF[9e02dae1]
  name = CaribouLite RPI Hat
  serial = ##########
  uuid = #######
  vendor = CaribouLabs LTD
  version = 0x0001

03-17 11:10:23.584  3851  3851 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
03-17 11:10:23.585  3851  3851 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
```

K7MDL2, 2023-03-17:
> @siffland Just tried it again. Multiple attempts, you've pulled ahead in getting things to work. 👍 How did you stop the read? I just "ctl-c", but it drops me out of the app, as well. Also tried 'soft reset'. Tried multiple times of Ch2 and Ch1, no change.

Type 99, hit enter, the sampling will stop and drop back to the menu.

great9, 2023-03-23:
Just an update, I got it working:

I made sure to remove all commented fields from /boot/config.txt and /boot/firmware/usercfg.txt on my rpi3b. and then added:
```
dtparam=audio=on
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=i2c_vc=on
dtoverlays=smi
dtoverlays=smi-dev
dtoverlay=dwc2,dr_mode=peripheral
```