cariboulabs/cariboulite #46: SoapySDRUtil --find commands fails

sorvis, 2023-02-09:
Using the develop_R1 branch and cherry-picked in the PRs from @MartinHerren 

Has anyone worked past this issue?

```
sudo SoapySDRUtil --find 
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] SoapySDR::loadModule(/usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8/libSoapyCariboulite.so)
  dlopen() failed: libiir.so.1: cannot open shared object file: No such file or directory
No devices found! 

```

MartinHerren, 2023-02-09:
Hi,

i fixed this issue with my PR #39 
Another solution would be to statically link libiir to the module.

sorvis, 2023-02-09:
Thank you @MartinHerren ...perhaps I'm missing a step here or need to uninstall the module first or something? I tried applying the change and recompiling and not seeing a difference.

``` bash
pi@pisdr:~/projects/cariboulite/software/libcariboulite/build $ git diff
diff --git a/software/libcariboulite/CMakeLists.txt b/software/libcariboulite/CMakeLists.txt
index ac7da1e..208874d 100644
--- a/software/libcariboulite/CMakeLists.txt
+++ b/software/libcariboulite/CMakeLists.txt
@@ -24,7 +24,7 @@ add_subdirectory(src/rffc507x EXCLUDE_FROM_ALL)
 add_subdirectory(src/hat EXCLUDE_FROM_ALL)
 add_subdirectory(src/production_utils EXCLUDE_FROM_ALL)
 add_subdirectory(src/zf_log EXCLUDE_FROM_ALL)
-add_subdirectory(src/iir EXCLUDE_FROM_ALL)
+add_subdirectory(src/iir)
 
 # Create the library LibCaribouLite
 add_library(cariboulite STATIC ${SOURCES_LIB})
pi@pisdr:~/projects/cariboulite/software/libcariboulite/build $ cmake ../ | tail -n 1
-- Build files have been written to: /home/pi/projects/cariboulite/software/libcariboulite/build
pi@pisdr:~/projects/cariboulite/software/libcariboulite/build $ make | tail -n 1[100%] Built target iirdemo
pi@pisdr:~/projects/cariboulite/software/libcariboulite/build $ sudo make install | tail -n 1
-- Up-to-date: /usr/local/lib/libcariboulite.a
pi@pisdr:~/projects/cariboulite/software/libcariboulite/build $ sudo SoapySDRUtil --find 
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] SoapySDR::loadModule(/usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8/libSoapyCariboulite.so)
  dlopen() failed: libiir.so.1: cannot open shared object file: No such file or directory
No devices found! 

```

sorvis, 2023-02-10:
Reading this page:
https://github.com/cariboulabs/cariboulite/tree/63fb30606a76bd764143720e1be23bf6d35dae19/software/libcariboulite/src/caribou_smi

I noticed that only bcm2835_smi  is returning from lsmod

Under this dir:
software/libcariboulite/src/caribou_smi/kernel/build

I did the 
```
cmake ../
make
```

and now have smi_stream_dev.ko

Then manually load it with

```
sudo insmod smi_stream_dev.ko
sudo depmod -a
```

Now it shows up:
```
lsmod | grep smi
smi_stream_dev         20480  0
bcm2835_smi            20480  1 smi_stream_dev
```

Running the carriboulite app fails though:
```
sudo ./cariboulite_app 

02-10 06:39:40.684  5747  5747 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
02-10 06:39:40.684  5747  5747 E CARIBOU_SMI_MODULES caribou_smi_insert_smi_modules@caribou_smi_modules.c:71 Module insertion 'smi_stream_dev' failed
02-10 06:39:40.684  5747  5747 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:419 Problem reloading SMI kernel modules
02-10 06:39:40.684  5747  5747 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
02-10 06:39:40.684  5747  5747 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
02-10 06:39:40.684  5747  5747 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-10 06:39:40.684  5747  5747 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-10 06:39:40.684  5747  5747 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-10 06:39:40.684  5747  5747 E FPGA caribou_fpga_close@caribou_fpga.c:224 caribou_fpga_close: dev not initialized
02-10 06:39:40.684  5747  5747 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-10 06:39:40.684  5747  5747 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
```

That seems to have unloaded the module
```
lsmod | grep smi
bcm2835_smi            20480  0
```

Back to the [readme](https://github.com/cariboulabs/cariboulite/tree/63fb30606a76bd764143720e1be23bf6d35dae19/software/libcariboulite/src/caribou_smi)...sounds like I should not have to manually load the module. So that makes me wonder if my firmware needs updated with the latest changed, but looks like the directions to do that are in the README
https://github.com/cariboulabs/cariboulite/blob/63fb30606a76bd764143720e1be23bf6d35dae19/docs/flashing/README.md

siffland, 2023-02-11:
if you get to 

```
[ERROR] SoapySDR::loadModule(/usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8/libSoapyCariboulite.so)
  dlopen() failed: libiir.so.1: cannot open shared object file: No such file or directory
```

Try a 
```
sudo ldconfig
```

then reboot.  See if you get the same result.  Even with cherry picking merging the PR's, i cannot get the libSoapyCariboulite.so module to build.  Not sure what i am doing wrong.  I am more than sure i could work through the library issue if i could just get that to build.

rhammock1, 2023-02-12:
I also haven't been able to get `SoapySDRUtil` to find the Caribou device. I implemented the change in @MartinHerren's PR and rebuilt the `cariboulite_app`, but no change. I might need to attempt rebuilding `SoapySDRUtil`, though.  I have yet to be able to test if any of the changes from yesterday have affected this.

MartinHerren, 2023-02-12:
Did you run again a `sudo make install` after doing the changes ?

rhammock1, 2023-02-12:
I'm 95% sure I did, but I'll be able to test and do it all again in a few hours

sorvis, 2023-02-13:
Ah, @MartinHerren  your right I needed to do 
sudo make install

thank you!...on to the next problem to be solved.

closed 2023-02-13 by sorvis