cariboulabs/cariboulite #47: CMake Error at gr-cariboulite

uecken, 2023-02-09:
Doen anyone resolve this issue?

```
pi@pisdr:~/projects/cariboulite/software/gr-cariboulite/build $ cmake ../
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at CMakeLists.txt:15 (find_package):
  By not providing "FindCaribouLite.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "CaribouLite", but CMake did not find one.

  Could not find a package configuration file provided by "CaribouLite"
  (requested version 1.0) with any of the following names:

    CaribouLiteConfig.cmake
    cariboulite-config.cmake

  Add the installation prefix of "CaribouLite" to CMAKE_PREFIX_PATH or set
  "CaribouLite_DIR" to a directory containing one of the above files.  If
  "CaribouLite" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at CMakeLists.txt:20 (include):
  include could not find load file:

    GrComponent

CMake Error at CMakeLists.txt:21 (GR_REGISTER_COMPONENT):
  Unknown CMake command "GR_REGISTER_COMPONENT".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.18)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/pi/projects/cariboulite/software/gr-cariboulite/build/CMakeFiles/CMakeOutput.log".
```

siffland, 2023-02-10:
This is how i got the cariboulite_app to work, even after the last reboot i have to run the cariboulite_app several times (I think it is still initializing and i am just not patient enough).

```
###Fresh SDCard with PiSDR###

sudo apt update
sudo apt dist-upgrade

sudo apt install libczmq-dev

echo "dtparam=i2c_vc=on" | sudo tee -a /boot/config.txt
echo "dtoverlays=smi" | sudo tee -a /boot/config.txt
echo "dtoverlays=smi-dev" | sudo tee -a /boot/config.txt

reboot

mkdir ~/projects
cd ~/projects
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite
git checkout remotes/origin/develop_R1
git submodule update --init --recursive

sed -i 's/add_subdirectory(src\/iir EXCLUDE_FROM_ALL)/add_subdirectory(src\/iir)/g' software/libcariboulite/CMakeLists.txt

sudo ./install.sh

sudo cp /home/pi/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build/smi_stream_dev.ko /lib/modules/5.15.84-v8+/kernel/drivers/misc/

sudo depmod

reboot

i2cdetect -y 2

cd projects/cariboulite
sudo ./build/cariboulite_app
```

G0MJW, 2023-02-10:
Almost worked with my PI Zero 2. Everything built OK but

sudo ./build/cariboulite_app
02-10 19:24:41.804  1220  1220 I CARIBOULITE Setup cariboulite_init_driver_minim                             al@cariboulite_setup.c:560 driver initializing
02-10 19:24:41.805  1220  1220 I CARIBOULITE Setup cariboulite_init_driver_minim                             al@cariboulite_setup.c:569 Initializing signals
02-10 19:24:41.805  1220  1220 I CARIBOULITE Setup cariboulite_setup_signal_hand                             ler@cariboulite_setup.c:687 setting up signal handler
02-10 19:24:41.805  1220  1220 I CARIBOULITE Setup cariboulite_setup_io@cariboul                             ite_setup.c:202 Setting up board I/Os
02-10 19:24:41.805  1220  1220 I IO_UTILS_Main io_utils_setup@io_utils.c:26 init                             ializing pigpio
02-10 19:24:41.986  1220  1220 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigp                             io version 79
02-10 19:24:41.986  1220  1220 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:3                             21 configuring gpio setups
02-10 19:24:41.986  1220  1220 D CARIBOULITE Setup cariboulite_init_driver_minim                             al@cariboulite_setup.c:583 Initializing FPGA
02-10 19:24:41.986  1220  1220 I FPGA caribou_fpga_init@caribou_fpga.c:116 confi                             guring reset and irq pins
02-10 19:24:41.986  1220  1220 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initi                             alizing io_utils_spi
02-10 19:24:41.986  1220  1220 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:8                             7 device init completed
02-10 19:24:41.986  1220  1220 D CARIBOULITE Setup cariboulite_init_driver_minim                             al@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
        System Version: FF
        Manu. ID: FF
        Sys. Ctrl Version: FF
        IO Ctrl Version: FF
        SMI Ctrl Version: FF
02-10 19:24:41.987  1220  1220 I CARIBOU_PROG caribou_prog_configure_prepare@car                             ibou_prog.c:148 RESET low, Waiting for CDONE low
02-10 19:24:41.989  1220  1220 I CARIBOU_PROG caribou_prog_configure_from_buffer                             @caribou_prog.c:259 Sending bitstream of size 32220
02-10 19:24:43.741  1220  1220 I CARIBOU_PROG caribou_prog_configure_from_buffer                             @caribou_prog.c:281 bitstream sent 32220 bytes
02-10 19:24:43.742  1220  1220 I CARIBOU_PROG caribou_prog_configure_finish@cari                             bou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
02-10 19:24:43.742  1220  1220 I CARIBOU_PROG caribou_prog_configure_finish@cari                             bou_prog.c:203 0 dummy clocks sent
02-10 19:24:43.742  1220  1220 I CARIBOU_PROG caribou_prog_configure_from_buffer                             @caribou_prog.c:291 FPGA programming - Success!

FPGA Versions:
        System Version: 01
        Manu. ID: 01
        Sys. Ctrl Version: 01
        IO Ctrl Version: 01
        SMI Ctrl Version: 01
02-10 19:24:43.843  1220  1220 D CARIBOULITE Setup cariboulite_init_driver_minim                             al@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0.                             .3]: [1,1,1,1]
02-10 19:24:43.843  1220  1220 I CARIBOULITE Setup cariboulite_init_driver_minim                             al@cariboulite_setup.c:622 Detecting Board Information
02-10 19:24:43.844  1220  1220 I HAT hat_print_board_info@hat.c:703 # Board Info                              - Category name: hat
02-10 19:24:43.844  1220  1220 I HAT hat_print_board_info@hat.c:704 # Board Info                              - Product name: CaribouLite RPI Hat
02-10 19:24:43.844  1220  1220 I HAT hat_print_board_info@hat.c:705 # Board Info                              - Product ID: 0x0001, Numeric: 1
02-10 19:24:43.844  1220  1220 I HAT hat_print_board_info@hat.c:706 # Board Info                              - Product Version: 0x0001, Numeric: 1
02-10 19:24:43.844  1220  1220 I HAT hat_print_board_info@hat.c:707 # Board Info                              - Product UUID: 0882a4a2-4d2f-4b43-94fb-969e8640dfc5, Numeric serial: 0xD156B66                             4
02-10 19:24:43.844  1220  1220 I HAT hat_print_board_info@hat.c:708 # Board Info                              - Vendor: CaribouLabs LTD
02-10 19:24:43.844  1220  1220 I CARIBOULITE Setup cariboulite_print_board_info@                             cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
02-10 19:24:43.844  1220  1220 I CARIBOULITE Setup cariboulite_init_submodules@c                             ariboulite_setup.c:272 initializing submodules
02-10 19:24:43.844  1220  1220 D CARIBOULITE Setup cariboulite_init_submodules@c                             ariboulite_setup.c:276 INIT FPGA SMI communication
02-10 19:24:43.844  1220  1220 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:413                              initializing caribou_smi
02-10 19:24:43.877  1220  1220 D CARIBOU_SMI_MODULES caribou_smi_check_modules@c                             aribou_smi_modules.c:98 Unloading smi-stream module
02-10 19:24:43.905  1220  1220 D CARIBOU_SMI_MODULES caribou_smi_check_modules@c                             aribou_smi_modules.c:111 Loading smi-stream module
02-10 19:24:44.092  1220  1220 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:431                              couldn't open smi driver file '/dev/smi'
[... cut ...]
02-10 19:24:44.092  1220  1220 E FPGA caribou_fpga_close@caribou_fpga.c:225 cari                             bou_fpga_close: dev not initialized
02-10 19:24:44.092  1220  1220 I CARIBOULITE Setup cariboulite_release_io@caribo                             ulite_setup.c:240 Releasing board I/Os - closing SPI
02-10 19:24:44.093  1220  1220 I CARIBOULITE Setup cariboulite_release_io@caribo                             ulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
02-10 19:24:44.238  1220  1220 E CARIBOULITE Main main@cariboulite_app.c:68 driv                             er init failed, terminating...

siffland, 2023-02-10:
Try to run sudo ./build/cariboulite_app a few more times, also i always have to run i2cdetect -y 2 after the reboot and before the cariboulite_app.  Mine is running on the zero 2 as well.  It seems inconsistent, and i cannot get the soapysdr module to even build.  I even tried with this last build over lunch to manually apply all the PR's for the github.

G0MJW, 2023-02-10:
I tried. I have run it many times. The i2C detect app only works before this - it may be crashing the driver perhaps. I2C detect detects nothing after a reboot whereas it detected things before..

Hang on, on the 10th run it ran... 

 MODEM Version: AT86RF215IQ (without basebands), version: 03
=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Diginal I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [99]  Quit
    Choice:   2

=====================================
Board Information (HAT)
        Category name: hat
        Product name: CaribouLite RPI Hat
        Product ID: 0x0001, Numeric: 1
        Product Version: 0x0001, Numeric: 1
        Product UUID: 0882a4a2-4d2f-4b43-94fb-969e8640dfc5, Numeric serial: 0xD156B664
        Vendor: CaribouLabs LTD
        Product Type: CaribouLite FULLFPGA Versions:
        System Version: 01
        Manu. ID: 01
        Sys. Ctrl Version: 01
        IO Ctrl Version: 01
        SMI Ctrl Version: 01
        MODEM Version: AT86RF215IQ (without basebands), version: 03
Library Versions:
        (Major, Minor, Rev): (1, 0, 1)

=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Diginal I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [99]  Quit
    Choice:

sorvis, 2023-02-11:
Good instructions @siffland I was able to make the caribou app turn on the radio for both channels at 467612500 which is channel 10 on my walki-talki. Couldn't get the CW mode to work, but it would turn the radio on and off and could hear that click on my walki-talki. 

I can't get SoapySDRUtil to find anything still though.

Note about the frequency...
The parameters section seems to say its in MHz but when you type it in it says Hz...

G0MJW, 2023-02-11:
I also failed to get Soapy to work. I seem to have broken the installation as well. I think we are going to have to wait for a working main branch or someone to produce a working distribution like PISDR or DragonOS. I hate to think how many frustrated people might be out there right now attempting to follow the instructions as Mouser recently delivered several hundred early bird units to backers.

edegraaff, 2023-02-12:
> /home/pi/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build/

With your description i get some progress, is the smi_stream_dev.ko  the driver that should enable carbibou lite board? 

I am searchin for a way to enable the board as an sdr... also for the near future, the main idea was add the board to my pi4, and when i run OpenWebRX i have more or less an embedded WebSDR that way my current OpenWebRX is upgraded.. it was promissed to work, so i am trying to read all comments .. but first get it running on the Pi Zero 2W, your build guide 

Here it seems i walk in the dark : 
root@pisdr:/home/pi# i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@pisdr:/home/pi# SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found!
root@pisdr:/home/pi#

edegraaff, 2023-02-12:
I am also here: 
=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Digital I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [99]  Quit
    Choice:   2

=====================================
Board Information (HAT)
        Category name: hat
        Product name: CaribouLite RPI Hat
        Product ID: 0x0001, Numeric: 1
        Product Version: 0x0001, Numeric: 1
        Product UUID: 457dca56-0d60-41c2-9d9a-83b72f69af2d, Numeric serial: 0xD5                                                                                                                                                             870BD0
        Vendor: CaribouLabs LTD
        Product Type: CaribouLite FULLFPGA Versions:
        System Version: 01
        Manu. ID: 01
        Sys. Ctrl Version: 01
        IO Ctrl Version: 01
        SMI Ctrl Version: 01
        MODEM Version: AT86RF215IQ (without basebands), version: 03
Library Versions:
        (Major, Minor, Rev): (1, 0, 1)

=====================================
 Select a function:
 [0]  Hard reset FPGA
 [1]  Soft reset FPGA
 [2]  Print board info and versions
 [3]  Program FPGA
 [4]  Perform a Self-Test
 [5]  FPGA Digital I/O
 [6]  FPGA RFFE control
 [7]  FPGA SMI fifo status
 [8]  Modem transmit CW signal
 [9]  Modem receive I/Q stream
 [99]  Quit

uecken, 2023-02-19:
This issue is for develop_R1. I will close it. 
Now main branch is up to date now.

closed 2023-02-19 by uecken