cariboulabs/cariboulite #54: FPGA programming failed

fransla, 2023-02-13:
Using develop_R1 build, I get the following error when running cariboulite_test_app:

```
sudo ./cariboulite_test_app 
02-13 18:17:07.649  2166  2166 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-13 18:17:07.649  2166  2166 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-13 18:17:07.650  2166  2166 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-13 18:17:07.650  2166  2166 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-13 18:17:07.650  2166  2166 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-13 18:17:07.759  2166  2166 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
02-13 18:17:07.759  2166  2166 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
02-13 18:17:07.759  2166  2166 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
02-13 18:17:07.759  2166  2166 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
02-13 18:17:07.759  2166  2166 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
02-13 18:17:07.760  2166  2166 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
02-13 18:17:07.760  2166  2166 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
02-13 18:17:07.760  2166  2166 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
	System Version: FF
	Manu. ID: FF
	Sys. Ctrl Version: FF
	IO Ctrl Version: FF
	SMI Ctrl Version: FF
02-13 18:17:07.761  2166  2166 I CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
02-13 18:17:07.763  2166  2166 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:259 Sending bitstream of size 32220
02-13 18:17:08.898  2166  2166 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:281 bitstream sent 32220 bytes
02-13 18:17:08.898  2166  2166 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
02-13 18:17:08.898  2166  2166 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:203 0 dummy clocks sent
02-13 18:17:08.898  2166  2166 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:291 FPGA programming - Success!

FPGA Versions:
	System Version: FF
	Manu. ID: FF
	Sys. Ctrl Version: FF
	IO Ctrl Version: FF
	SMI Ctrl Version: FF
02-13 18:17:08.999  2166  2166 E FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:182 Programming failed
02-13 18:17:08.999  2166  2166 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:599 FPGA programming failed
02-13 18:17:08.999  2166  2166 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-13 18:17:08.999  2166  2166 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-13 18:17:08.999  2166  2166 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-13 18:17:08.999  2166  2166 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-13 18:17:08.999  2166  2166 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
02-13 18:17:09.005  2166  2166 E CARIBOULITE Test main@cariboulite_test_app.c:68 driver init failed, terminating...
pi@raspberrypi:~/projects/cariboulite/build $ 

```

rhammock1, 2023-02-13:
What kind of Pi are you using, and what OS?

fransla, 2023-02-13:
Pi M400

Raspberry Pi OS with desktop
Release date: September 22nd 2022
System: 64-bit
Kernel version: 5.15
Debian version: 11 (bullseye)

fransla, 2023-02-14:
I tried again using PiSDR on a Pi 4 and it worked.

cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Model		: Raspberry Pi 4 Model B Rev 1.1

Probably something strange with the Pi 400 GPIO pins or my adapter.

closed 2023-02-14 by fransla

mvmn, 2023-12-03:
I have the same issue. How did you fix it?