cariboulabs/cariboulite #56: Undefined symbol

zbyrek, 2023-02-16:
Hello,
I built latest commit of main branch it works better than before (at least less errors during SoapySdrUtil and built in test binary works), however still got one issue that prohibits sdr plus plus usage:

`SoapySDRUtil: symbol lookup error: /usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8/libSoapyCariboulite.so: undefined symbol: _ZN3Iir7CascadeC2Ev
`
Is this known issue?

pocki, 2023-02-16:
I have the same issue

EelkeVisser, 2023-02-17:
I have the same issue when running sdrpp as root.
```
sdrpp: symbol lookup error: /usr/lib/arm-linux-gnueabihf/SoapySDR/modules0.7/libSoapyCariboulite.so: undefined symbol: _ZN3Iir7CascadeC2Ev
```

EelkeVisser, 2023-02-17:
I solved it. It is caused by old libs that are still installed by the previous installation attempts. I cleaned out all libiir and CaribouLite libraries and reinstalled. See this forum post:
https://community.libre.space/t/cariboulite-need-help/10068/19

zbyrek, 2023-02-17:
I tried replicate your steps but didn't help.

EelkeVisser, 2023-02-17:
No, I am not using a clean image. I am using my Satnogs station to receive satellite data, and installed the CaribouLite on it. I think it once started as Raspbian, and then I installed Satnogs and other things. And now I am adding the CaribouLite.

I guess a clean Raspbian image should work. O, and I am using a Pi4.

closed 2023-02-25 by zbyrek