cariboulabs/cariboulite #58: update on compiling and getting things work

edegraaff, 2023-02-16:
Today i did some rework, and installed as suggested the PIsdr image and did a full update and upgrade.
Then i installed the modules sudo apt-get install soapysdr-module-all libsoapysdr-dev soapysdr-tools
And used the main branch. 
The compile proces had no issues .. it looked all fine.
When i started SDR ++ it still looked fine and then... 
starting up is saw these : Today i did some rework, and installed as suggested the PIsdr image and did a full update and upgrade.
Then i installed the modules sudo apt-get install soapysdr-module-all libsoapysdr-dev soapysdr-tools
And used the main branch. 
The compile proces had no issues .. it looked all fine.
When i started SDR ++ it still looked fine and then... 
starting up is saw these : 
[image]

Thats the first issue any clue? 
Then it all looks ok, soapy sdr, and even cariboulite is shown .. well this is again a step forward... 
[image]

But then i press start and this is the result : 
[image]

The situation:
PiZero 2 with PiSDR 

snippet config.txt 
[all]
#dtparam=i2c_vc=on
dtoverlays=smi
#dtoverlays=smi-dev

any clues where i have made an errror?

justinh24, 2023-02-16:
I think that was happening to me when the SoapyRemote service was running.
# will tell you if it's running
sudo systemctl status soapyremote-server.service
# will stop it
sudo systemctl stop soapyremote-server.service

slesto, 2023-02-17:
I get the same thing and the SoapyRemote service is not running

alphafox02, 2023-02-17:
The one thing I’ve been able to get running on the pi itself is retrogram soapy edition. Passing the variables needed I was able to use the complete SDR to receive so long as I didn’t change the frequency, however it’s like almost a terminal looking fft. This was on DragonOS Pi64.

alphafox02, 2023-02-17:
I can now run the main branch installed on DragonOS Pi64 w/ two small changes. Then I can run sudo Soapy Server on the Pi and then on my desktop version of DragonOS I can connect to the CaribouLite with SigDigger and can even change the freq now! Plus demod audio. Not sure why the spectrum looks somewhat mirrored now, but it’s working. 
[image]

slesto, 2023-02-17:
I finally got it working... after about 30 minutes of playing around the fm band I noticed the pi and cariboulite started to get a bit hot.  Next step is to get a case with a fan to help keep this thing cool

edegraaff, 2023-02-18:
> I finally got it working... after about 30 minutes of playing around the fm band I noticed the pi and cariboulite started to get a bit hot. Next step is to get a case with a fan to help keep this thing cool

What is the receipt to get it running? I will give it a try to on the pi zero 2 and see if it will work.