cariboulabs/cariboulite #60: Driver fail with SoapySDRServer

daviddawe1982, 2023-02-17:
I have finally got to the point with PiSDR where I can run cariboulite_test_app without problems and also run SoapySDRServer --bind.

The SDR serve is seen as CaribouLite S1G[bd0e0b9a] and CaribouLite HiF[bd0e0b9b] in CubicSDR but as soon as I try to run I get the following.

`pi@pisdr:~/projects/cariboulite/build $ sudo SoapySDRServer --bind
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.6.0-gc09b2f10
Server UUID: ce191aba-8ba0-1744-8567-1308007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[INFO] Avahi version:  avahi 0.8
[INFO] Avahi hostname: pisdr
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     pisdr.local
[INFO] avahi_entry_group_add_service(pisdr._soapy._tcp)
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.1.7]:55410)
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.1.7]:55411)
SoapyServerListener::accept([::ffff:192.168.1.7]:55412)
02-17 18:21:36.427  4872  4887 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-17 18:21:36.427  4872  4887 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-17 18:21:36.427  4872  4887 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-17 18:21:36.427  4872  4887 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-17 18:21:36.427  4872  4887 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-17 18:21:36.586  4872  4887 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
02-17 18:21:36.586  4872  4887 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
02-17 18:21:36.586  4872  4887 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
02-17 18:21:36.586  4872  4887 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
02-17 18:21:36.586  4872  4887 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
02-17 18:21:36.586  4872  4887 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
02-17 18:21:36.586  4872  4887 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
02-17 18:21:36.586  4872  4887 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
	System Version: 01
	Manu. ID: 01
	Sys. Ctrl Version: 01
	IO Ctrl Version: 01
	SMI Ctrl Version: 01
02-17 18:21:36.587  4872  4887 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:188 FPGA already operational - not programming (use 'force_prog=true' to force update)
02-17 18:21:36.587  4872  4887 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
02-17 18:21:36.587  4872  4887 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
02-17 18:21:36.588  4872  4887 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
02-17 18:21:36.588  4872  4887 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
02-17 18:21:36.588  4872  4887 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
02-17 18:21:36.588  4872  4887 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
02-17 18:21:36.588  4872  4887 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: fa16cc83-a1cf-4706-855e-da09ba493401, Numeric serial: 0xDE8705CD
02-17 18:21:36.588  4872  4887 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
02-17 18:21:36.588  4872  4887 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
02-17 18:21:36.588  4872  4887 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
02-17 18:21:36.588  4872  4887 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
02-17 18:21:36.588  4872  4887 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:425 initializing caribou_smi
02-17 18:21:36.590  4872  4887 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:98 Unloading smi-stream module
02-17 18:21:36.620  4872  4887 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
SMI SETTINGS:
    width: 0
    pack: Y
    read setup: 0, strobe: 2, hold: 0, pace: 0
    write setup: 1, strobe: 4, hold: 1, pace: 0
    dma enable: Y, passthru enable: Y
    dma threshold read: 1, write: 63
    dma panic threshold read: 32, write: 32
02-17 18:21:36.634  4872  4887 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:286 INIT MODEM - AT86RF215
02-17 18:21:36.634  4872  4887 I AT86RF215_Main at86rf215_init@at86rf215.c:160 configuring reset and irq pins
02-17 18:21:36.634  4872  4887 I AT86RF215_Main at86rf215_init@at86rf215.c:168 Adding chip definition to io_utils_spi
[... 208 lines cut ...]
02-17 18:21:46.210  4872  4921 D CARIBOULITE Radio cariboulite_radio_ext_ref@cariboulite_radio.c:82 Setting ext_ref = 32MHz
02-17 18:21:46.210  4872  4921 D RFFC5072 rffc507x_disable@rffc507x.c:252 rfc5071_disable
02-17 18:21:46.211  4872  4921 D RFFC5072 rffc507x_enable@rffc507x.c:260 rfc5071_enable
02-17 18:21:46.212  4872  4921 I RFFC5072 rffc507x_print_stat@rffc507x.c:421 RFFC507X STAT: 0x1C7C PLL_LOCK: 0, CT_CAL: 28, KV_CAL: 31, CT_CAL_FAIL: 0
02-17 18:21:46.212  4872  4921 I RFFC5072 rffc507x_print_stat@rffc507x.c:421 RFFC507X STAT: 0xA17C PLL_LOCK: 1, CT_CAL: 33, KV_CAL: 31, CT_CAL_FAIL: 0
02-17 18:21:46.212  4872  4921 D CARIBOULITE Radio cariboulite_radio_set_frequency@cariboulite_radio.c:782 Frequency setting CH: 1, Wanted: 100000000.00 Hz, Set: 99999999.36 Hz (MOD: 2494999808.00, MIX: 2594999807.36)
02-17 18:21:46.213  4872  4889 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO24: Transceiver ready
02-17 18:21:46.214  4872  4921 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:805 Activating channel 1, dir = RX, activate = 1
02-17 18:21:46.214  4872  4921 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:853 Setup Modem state cmd_rx
02-17 18:21:46.323  4872  4921 D CARIBOULITE Radio cariboulite_radio_ext_ref@cariboulite_radio.c:82 Setting ext_ref = 32MHz
02-17 18:21:46.323  4872  4921 D RFFC5072 rffc507x_disable@rffc507x.c:252 rfc5071_disable
02-17 18:21:46.323  4872  4921 D RFFC5072 rffc507x_enable@rffc507x.c:260 rfc5071_enable
02-17 18:21:46.325  4872  4921 I RFFC5072 rffc507x_print_stat@rffc507x.c:421 RFFC507X STAT: 0x217C PLL_LOCK: 0, CT_CAL: 33, KV_CAL: 31, CT_CAL_FAIL: 0
02-17 18:21:46.325  4872  4921 I RFFC5072 rffc507x_print_stat@rffc507x.c:421 RFFC507X STAT: 0x8E7C PLL_LOCK: 1, CT_CAL: 14, KV_CAL: 31, CT_CAL_FAIL: 0
02-17 18:21:46.325  4872  4921 D CARIBOULITE Radio cariboulite_radio_set_frequency@cariboulite_radio.c:782 Frequency setting CH: 1, Wanted: 100000000.00 Hz, Set: 99999999.36 Hz (MOD: 2494999808.00, MIX: 2594999807.36)
caribou_smi_read -> Timeout
02-17 18:21:46.584  4872  4923 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:971 SMI reading operation returned timeout
caribou_smi_read -> Timeout
02-17 18:21:46.847  4872  4923 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:971 SMI reading operation returned timeout
caribou_smi_read -> Timeout
02-17 18:21:47.109  4872  4923 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:971 SMI reading operation returned timeout
caribou_smi_read -> Timeout
02-17 18:21:47.371  4872  4923 D CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:971 SMI reading operation returned timeout
caribou_smi_read -> Timeout
`

daviddawe1982, 2023-02-17:
Update it now is working with CubicSDR on the S1G but not HF and when I say working it is running so slow that it can not be used

G0MJW, 2023-02-17:
I still get driver init failures.

02-17 14:50:15.810  3179  3179 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
02-17 14:50:15.811  3179  3179 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
02-17 14:50:15.812  3179  3179 E CARIBOU_SMI_MODULES caribou_smi_insert_smi_modules@caribou_smi_modules.c:71 Module insertion 'smi_stream_dev' failed
02-17 14:50:15.812  3179  3179 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:451 Problem reloading SMI kernel modules
02-17 14:50:15.813  3179  3179 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
02-17 14:50:15.813  3179  3179 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
02-17 14:50:15.813  3179  3179 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-17 14:50:15.813  3179  3179 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-17 14:50:15.813  3179  3179 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-17 14:50:15.813  3179  3179 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
02-17 14:50:15.813  3179  3179 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-17 14:50:15.813  3179  3179 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
**[ERROR] cariboulite_init_driver() failed**

G0MJW, 2023-02-17:
OK - I figured it out - it can't install the kernel module. The question is why.

pi@pisdr:~/projects/cariboulite/build $ sudo ./cariboulite_test_app 
02-17 17:59:59.573  1485  1485 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-17 17:59:59.573  1485  1485 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-17 17:59:59.573  1485  1485 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-17 17:59:59.574  1485  1485 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-17 17:59:59.574  1485  1485 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-17 17:59:59.765  1485  1485 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
02-17 17:59:59.765  1485  1485 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
02-17 17:59:59.765  1485  1485 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
02-17 17:59:59.765  1485  1485 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
02-17 17:59:59.765  1485  1485 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
02-17 17:59:59.765  1485  1485 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
02-17 17:59:59.766  1485  1485 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
02-17 17:59:59.766  1485  1485 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
	System Version: 01
	Manu. ID: 01
	Sys. Ctrl Version: 01
	IO Ctrl Version: 01
	SMI Ctrl Version: 01
02-17 17:59:59.767  1485  1485 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:188 FPGA already operational - not programming (use 'force_prog=true' to force update)
02-17 17:59:59.767  1485  1485 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
02-17 17:59:59.767  1485  1485 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
02-17 17:59:59.768  1485  1485 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
02-17 17:59:59.768  1485  1485 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
02-17 17:59:59.768  1485  1485 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
02-17 17:59:59.768  1485  1485 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
02-17 17:59:59.768  1485  1485 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: 0882a4a2-4d2f-4b43-94fb-969e8640dfc5, Numeric serial: 0xD156B664
02-17 17:59:59.768  1485  1485 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
02-17 17:59:59.769  1485  1485 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
02-17 17:59:59.769  1485  1485 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
02-17 17:59:59.769  1485  1485 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
02-17 17:59:59.769  1485  1485 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
02-17 17:59:59.771  1485  1485 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:98 Unloading smi-stream module
02-17 17:59:59.792  1485  1485 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
02-17 17:59:59.810  1485  1485 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:460 couldn't open smi driver file '/dev/smi'
02-17 17:59:59.810  1485  1485 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
02-17 17:59:59.810  1485  1485 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
02-17 17:59:59.810  1485  1485 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-17 17:59:59.810  1485  1485 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-17 17:59:59.810  1485  1485 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-17 17:59:59.810  1485  1485 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
02-17 17:59:59.810  1485  1485 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-17 17:59:59.810  1485  1485 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
02-17 17:59:59.832  1485  1485 E CARIBOULITE Test main@cariboulite_test_app.c:68 driver init failed, terminating...

G0MJW, 2023-02-17:
and dmesg reports - could it be the PI Zero 2W does not have sufficient memory

[ 2228.628645] smi-stream-dev soc:smi_dev: smi-stream-dev: smi-stream dev removed
[ 2228.704659] smi-stream-dev: smi_stream_dev_probe
[ 2228.705451] smi-stream-dev: creating a device and registering it with sysfs
[ 2228.705903] smi-stream-dev soc:smi_dev: initialised
[ 2228.707228] smi-stream-dev soc:smi_dev: SMI device opened.
[ 2228.717032] cariboulite_tes: page allocation failure: order:10, mode:0x40cc0(GFP_KERNEL|__GFP_COMP), nodemask=(null),cpuset=/,mems_allowed=0
[ 2228.717140] CPU: 0 PID: 1810 Comm: cariboulite_tes Tainted: G         C O      5.15.84-v8+ #1613
[ 2228.717156] Hardware name: Raspberry Pi Zero 2 W Rev 1.0 (DT)

G0MJW, 2023-02-17:
Progress. Take the same SD card, put it in a PI4 - all works. So, it would appear, at the moment at least, the SDR is not compatible with the Pi Zero 2 it was designed to be used with. I don't know if this is fixable. I hope so because you can't get a PI4 these days and the form factor and low power consumption was attractive.

daviddawe1982, 2023-02-18:
Just tried running as headless and still no good.
But I did notice that we are maxing out the CPU, wondering if that gives any hints as to what's going on.
The image is CubicSDR running on Windows and SSH into RPI running SoapySDRServer and HTOP 
[image]

daviddawe1982, 2023-02-19:
This was due to me using  R1 branch on Zero 2w, switched to main and it's running now..

closed 2023-02-19 by daviddawe1982