cariboulabs/cariboulite #61: SoapySDR CaribouLite driver with SDR++

hb9dug, 2023-02-17:
Following the recommendation here: https://github.com/cariboulabs/cariboulite/issues/43#issuecomment-1432511102

SDR++ works but the spectrum is not displayed correctly. The spectrum is mirrored.

The spectrum is displayed correctly when using an RTL-SDR key with its SoapySDR driver.

EelkeVisser, 2023-02-17:
I have the same problem. See:
[image]

alphafox02, 2023-02-19:
Same happens now in SigDigger, GQRX, CubicSDR, etc. It wasn’t that way before, but at least now the frequencies can be changed and it can be started and stopped. Just pointing this out as it’s not specific to SDR++.

G0MJW, 2023-02-19:
If mirrored, just swap I and Q, but your example doesn't look like an inverted spectrum, it looks more like some other issue with the sampling.

daviddawe1982, 2023-02-19:
I'm also getting a mirrored signal with RPI Zero 2W running SoapyRemote.

konimaru, 2023-02-27:
Mirroring appears to be sorted https://github.com/cariboulabs/cariboulite/issues/70#issuecomment-1446268724