cariboulabs/cariboulite #62: Getting cariboulite working with OpenWebRX

edegraaff, 2023-02-18:
When the project strated and i signed in for this board one of my questions was : 
Will the board run with OpenWebRX and there was stated Yes it will! 
This was for me one of the reasons to support the board since that could be a great SDR solution, the Cariboulite in combination with a great websdr.

Does someone have any clue you i should get this working since the board is not in the list of supported SDR's and i could imagine some work needs to be done to get it gooing but would be great ... 

nl13974

edegraaff, 2023-02-19:
I made some progress, this is what i have managed to reach: 
Building SoapyRemote from zero failed on the pi zero.
pi os light 64 bit 

increase swap \
sudo dphys-swapfile swapoff
 sudo nano /etc/dphys-swapfile
CONF_SWAPSIZE=2048
sudo dphys-swapfile setup
sudo dphys-swapfile swapon
sudo reboot

sudo apt-get install soapysdr-module-all libsoapysdr-dev soapysdr-tools build-essential git
add : dtparam=i2c_arm=on

sudo apt-get install  soapyremote-server

sudo SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
02-19 18:35:24.698   897   897 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-19 18:35:24.698   897   897 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-19 18:35:24.698   897   897 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-19 18:35:24.698   897   897 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-19 18:35:24.699   897   897 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-19 18:35:24.699   897   897 V initInitialise@pigpio.c:8306 Can't lock /var/run/pigpio.pid
02-19 18:35:24.699   897   897 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed
02-19 18:35:24.699   897   897 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:205 Error setting up io_utils
[ERROR] cariboulite_init_driver() failed
02-19 18:35:24.699   897   897 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
Printing 'findCariboulite' Request:
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
Found device 0
  channel = S1G
  device_id = 0
  driver = Cariboulite
  label = CaribouLite S1G[ab0e17a0]
  name = CaribouLite RPI Hat
  serial = ab0e17a0
  uuid = 457dca56-0d60-41c2-9d9a-83b72f69af2d
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 1
  channel = HiF
  device_id = 1
  driver = Cariboulite
  label = CaribouLite HiF[ab0e17a1]
  name = CaribouLite RPI Hat
  serial = ab0e17a1
  uuid = 457dca56-0d60-41c2-9d9a-83b72f69af2d
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 2
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = PulseAudio

Found device 3
  channel = S1G
  device_id = 0
  driver = remote
  label = CaribouLite S1G[ab0e17a0]
  name = CaribouLite RPI Hat
  remote = tcp://192.168.1.161:55132
  remote:driver = Cariboulite
  serial = ab0e17a0
  uuid = 457dca56-0d60-41c2-9d9a-83b72f69af2d
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 4
  channel = HiF
  device_id = 1
  driver = remote
  label = CaribouLite HiF[ab0e17a1]
  name = CaribouLite RPI Hat
  remote = tcp://192.168.1.161:55132
  remote:driver = Cariboulite
  serial = ab0e17a1
  uuid = 457dca56-0d60-41c2-9d9a-83b72f69af2d
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 5
  default_input = True
  default_output = True
  device_id = 0
  driver = remote
  label = PulseAudio
  remote = tcp://192.168.1.161:55132
  remote:driver = audio

I am able to connect from the openwebrx to the SoapySDRserver but signal is bad 
if i make a signal with my porto i see it in the waterfall, but  the freq should be 434.700 and it is projected at 434.925 mhz
[image]

Also the sample rate can not be higher than 1 since otherwise the stream is getting hickups. 

Eelco

ImDroided, 2023-03-01:
Can you give me some insite on what you did to get openwebrx to connect... I have openwebrx installed and soapy stuff enabled but I keep getting a error saying bad sample rate. I have tried multiple sample rates and its all the same output. what sample rate are you using and how do you have the device settings/profile page in openwebrx? This is my output....

user@caribou:~ $ sudo docker run --tmpfs=/tmp/openwebrx --device /dev/bus/usb -p 8073:8073 -v openwebrx-settings:/var/lib/openwebrx jketterl/openwebrx-full
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2023-03-01 07:21:41,718 - owrx.__main__ - INFO - OpenWebRX version v1.3.0-dev starting up...
2023-03-01 07:21:43,464 - owrx.source.rtlsdr - INFO - Started sdr source: rtl_connector -g 29 -s 2400000 -f 438800000 -p 49475 -c 53285
2023-03-01 07:21:43,478 - owrx.source.rtlsdr - INFO - STDOUT: setting up control socket...
2023-03-01 07:21:43,479 - owrx.source.rtlsdr - INFO - STDOUT: control socket started on 53285
2023-03-01 07:21:43,481 - owrx.source.rtlsdr - INFO - STDOUT: socket setup complete, waiting for connections
2023-03-01 07:21:43,494 - owrx.source.rtlsdr - WARNING - STDERR: No supported devices found.
2023-03-01 07:21:43,494 - owrx.source.rtlsdr - WARNING - STDERR: no device found.
2023-03-01 07:21:43,496 - owrx.source.rtlsdr - WARNING - STDERR: Connector::open() failed
2023-03-01 07:21:43,569 - owrx.source.rtlsdr - ERROR - Exception during postStart()
Traceback (most recent call last):
  File "/opt/openwebrx/owrx/source/__init__.py", line 366, in start
    self.postStart()
  File "/opt/openwebrx/owrx/source/connector.py", line 57, in postStart
    self.controlSocket.connect(("localhost", self.controlPort))
ConnectionRefusedError: [Errno 111] Connection refused
2023-03-01 07:21:43,573 - owrx.connection - WARNING - SDR device "RTL-SDR USB Stick" has failed, selecting new device
2023-03-01 07:21:43,598 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - INFO - Started sdr source: soapy_connector -d 1,driver=remote,remote=192.168.1.31:55132 -s 24000 -f 99500000 -p 56047 -c 60283
2023-03-01 07:21:43,608 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - INFO - STDOUT: setting up control socket...
2023-03-01 07:21:43,609 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - INFO - STDOUT: control socket started on 60283
2023-03-01 07:21:43,610 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - INFO - STDOUT: socket setup complete, waiting for connections
2023-03-01 07:21:43,707 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - INFO - STDOUT: control connection established
2023-03-01 07:21:43,709 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - INFO - STDOUT: client connection established
2023-03-01 07:21:43,820 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32, scaleFactor=65536, mtu=1500, window=44040192)
2023-03-01 07:21:43,821 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Client side stream bound to 172.17.0.2:49236
2023-03-01 07:21:43,823 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Client side status bound to 172.17.0.2:34068
2023-03-01 07:21:43,823 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Using format CS16.
2023-03-01 07:21:43,824 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Server side stream bound to [::ffff:192.168.1.31]:50285
2023-03-01 07:21:43,825 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Server side stream connected to [::ffff:172.17.0.2]:49236
2023-03-01 07:21:43,826 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Server side status connected to [::ffff:172.17.0.2]:34068
2023-03-01 07:21:43,827 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 176 KiB
2023-03-01 07:21:43,828 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=176 KiB
2023-03-01 07:21:43,829 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Client side stream connected to 192.168.1.31:50285
2023-03-01 07:21:43,830 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 176 KiB
2023-03-01 07:21:43,831 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: [INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=176 KiB
2023-03-01 07:21:43,832 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR: terminate called after throwing an instance of 'std::runtime_error'
2023-03-01 07:21:43,833 - owrx.source.82f30d08-e57d-4394-8b44-c53d9b036fda - WARNING - STDERR:   what():  RemoteError: RtAudio init error 'RtApiPulse::probeDeviceOpen: unsupported sample rate.
2023-03-01 07:21:44,083 - owrx.connection - WARNING - SDR device "Caribou Lite 0" has failed, selecting new device
[... cut ...]
2023-03-01 07:21:45,719 - owrx.connection - WARNING - not answering client request since handshake is not complete
2023-03-01 07:21:45,721 - owrx.connection - WARNING - not answering client request since handshake is not complete
2023-03-01 07:21:45,722 - owrx.connection - WARNING - not answering client request since handshake is not complete
2023-03-01 07:21:46,681 - owrx.connection - WARNING - not answering client request since handshake is not complete
2023-03-01 07:21:46,682 - owrx.connection - WARNING - not answering client request since handshake is not complete

ImDroided, 2023-03-19:
I tried that still failed. Could you please post screenshots of how you have the device set up in OpenWebRX. Im not even sure I have that part correct.

edegraaff, 2023-03-19:
With the last install on my pi4 i failed to get it working with openwebrx... will give it a new try after clean install. It is still not good enough and imho the board and software in a development stage. There is some progress, but i still mis the standard "build it like this and it will work"  approach. Some reports show a working solution but in my environment it is still experimental. 

Eelco

K7MDL2, 2023-04-02:
This was likely solved in antoher issue/time, but I see the line in the log presented:

2023-03-01 07:21:43,464 - owrx.source.rtlsdr - INFO - Started sdr source: rtl_connector -g 29 **_-s 2400000_** -f 438800000 -p 49475 -c 53285

The -s 2400000  above should be -s 4000000, or 4MHz in the config sample rate setup.