cariboulabs/cariboulite #63: No receiving any signal.

daviddawe1982, 2023-02-19:
Has anyone actually managed to tune in to any frequency yet?

I finally have it running on my RPI Zero 2W with SoapyRemote but I cannot even pick my local radio station...

G0MJW, 2023-02-19:
Congratulations, I had to give up on the Pi zero. When I did get it working on my PI4 I found it to be incredibly insensitive, which implies it was not configured correctly. You might want to try a strong signal from your signal generator, not too strong though, you don't want to damage the front end!

daviddawe1982, 2023-02-19:
Your right!! I grabbed an 433mhz remote, centred the frequency on 433 and yes with the remote close I could definitely see a signal. 

I guess this is now a setting issue as you suggest.

If your still interested in the zero my build process is as follows..

Start with RPI light 64 image (headless)

Increase swap to 2gb (needed for Soapy install)
Install SoapySDR from github.
Then install Cariboulite MAIN branch.
Then install SoapyRemote

I have then setup usb as ethernet gadget and shut down wifi.

All is working with 40% Max cpu and little ram usage. 

Let me know if you want the step by step...

G0MJW, 2023-02-19:
Thanks - I will give that a go later. I have been using a PI4 borrowed from another project but it is quite a lot of effort to use as at the moment you need root access and that doesn't work with the audio without another process starting first. It might be worth stating which exact distro you used. I tried pisdr after failing with DragonOS but both are really too much for the PI Zero 2W. 

What is the USB ethernet gadget? Sounds useful.

As for deafness, I am hoping it is settings as it isn't usable at the current level of sensitivity. I would suspect the gain is currently set to the minimum. 

I am also hoping to work with SDR Angel to transmit. So far I have only managed to use the test programme to generate a CW signal.

Mike

G0MJW, 2023-02-19:
I have it working, but it doesn't work with SoapyRemote so I am missing something. The server is running OK, could it be permissions?

wrju333, 2023-02-19:
@G0MJW  Not sure if this applies anymore, but at one point in my testing I found that the caribouSoapyremote.so module needed to be copied manually into the Soapy modules library folder.  also after startup I use sudo killall SoapyServer, then SoapySDRUtil --find, then SoapySDRServer --bind and it seems to start up for me.  this may have changed over the last few commits as I have been away for a few days.

edegraaff, 2023-02-19:
> Your right!! I grabbed an 433mhz remote, centred the frequency on 433 and yes with the remote close I could definitely see a signal.
>
> I guess this is now a setting issue as you suggest.
>
> If your [...]

I am interested in the build proces for the pi zero w 2 , i have tried DragonOS but is demands to much mem boot fails. The ideal model for me would be:  headless pi, pi zero with software and from there use consile like soapy or even better connect it to the OpenWebRX > RTL-SDR (via SoapySDR) or  Device connected to a SoapyRemote server .. and then it could be a working chain ...   i will give your approack a try today ... 
thanks
Eelco

G0MJW, 2023-02-19:
I can transmit but still no luck with SoapyRemote.

[image]

daviddawe1982, 2023-02-19:
I will post my "script " after work.. 

Did you try sudo SoapySDRserver --bind

G0MJW, 2023-02-19:
Yes but it didn't work. It seems you have to install soapyremote manually and not from the repository.

git clone https://github.com/pothosware/SoapyRemote.git
cd SoapyRemote
mkdir build
cd build
cmake ..
make
sudo make install

Mike

daviddawe1982, 2023-02-19:
Yes that's what I did.. hoes it looking now

edegraaff, 2023-02-19:
> Yes but it didn't work. It seems you have to install soapyremote manually and not from the repository.
> 
> git clone https://github.com/pothosware/SoapyRemote.git cd SoapyRemote mkdir build cd build cmake .. make sudo make install
> 
> Mike
Did you have no buils errors ? I did not succeed with building 

    /usr/share/cmake/SoapySDR/SoapySDRConfig.cmake, version: 0.7.2-2

-- AVAHI_INCLUDE_DIRS=/usr/include
-- AVAHI_LIBRARIES=/usr/lib/aarch64-linux-gnu/libavahi-common.so;/usr/lib/aarch64-linux-gnu/libavahi-client.so
-- CMAKE_THREAD_LIBS_INIT: -lpthread
CMake Error at client/CMakeLists.txt:4 (SOAPY_SDR_MODULE_UTIL):
  Unknown CMake command "SOAPY_SDR_MODULE_UTIL".

G0MJW, 2023-02-19:
No build errors with Soapy-Server, it works now when run as root but it isn't working properly. The sensitivity is very low and so was the power except for the sub-1GHz range. I tried receiving there but it was still insensitive. I am hampered by not having any windows apps compatible with the Soapy server to properly test. I thought SDR Angel was an option, it also transmits, but it doesn't appear to have soapy server support implemented in Windows binaries. Cubic SDR works but seems contrary in the interface and it crashes quite a lot. However enough testing was possible with that to realise it is not receiving anything much except really strong signals. I will try SDR++.

edegraaff, 2023-02-19:
> No build errors with Soapy-Server, it works now when run as root but it isn't working properly. The sensitivity is very low and so was the power except for the sub-1GHz range. I tried receiving [...]

https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.4
There is a windows build here... i have installed and connected, i get noise but ... i am not able to tune, this is mentioned also in other issues.

daviddawe1982, 2023-02-20:
I have put instructions up at [https://cariboulitesdr.freeforums.net/](url) as this is not a great place for forum-like discussion.

edegraaff, 2023-02-20:
> https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.4
found it! :-)

edegraaff, 2023-02-20:
> Has anyone actually managed to tune in to any frequency yet?
> 
> I finally have it running on my RPI Zero 2W with SoapyRemote but I cannot even pick my local radio station...

I have a working chain, i get a waterfall, and when i use my porto to send i see the signal, but there are some issue to solve there. It is in another threat.  Tuning semms not yet to work as expected and also the right freq, but the progress over the weeks is we can build, and get remote signal.

G0MJW, 2023-02-25:
OK - I have found out why it is deaf. Hardware damaged in shipping. Looking in the plastic bag it came in I found a small core along with the header. I can see where it is supposed to go.

[image]

This must have been knocked off in shipping. Mouser packed it well so I assume it was sent to them in the anti-static bag and with the header being loose things moved around. Anyone else with deafness, this might be it. Now how to fix it? I can do SMD but I don't have a replacement part.

ImDroided, 2023-03-20:
> OK - I have found out why it is deaf. Hardware damaged in shipping. Looking in the plastic bag it came in I found a small core along with the header. I can see where it is supposed to go.
>
> [image] [...]

If you have not found a solution just get some proper wire and rewind what they have there. then look at the schematics and see what needs to be soldered to where.

mvmn, 2023-12-05:
I have a different issue - my Sub 1GHz receiver works, but it's quite useless because of how limited the ranges are. It picks up my PMR 446MHz radio, but can't hear any FM radio because it doesn't go below 390MHz. 

The HiFrequency channel, however, seems to be broken or something. Whatever frequency I pick - it shows a bit of a lump in the middle, but nothing else.

I don't see any damage on the outside (though packaging it better for shipment would be very good idea, but still).