cariboulabs/cariboulite #64: SDR is center frequency problems

daviddawe1982, 2023-02-20:
With my unit up and running I have been able to confirm that it does tune to stations BUT...

It is hard to explain, when moving the center frequency causing the SDR to change frequency I can slightly move left and right of where known stations are and I will get nothing but then I will see "pops" and if I manage to stop exactly when the "pops" happen then i can listen to a station or two. But if I move the center 1Hz then I'm back to the left right dance to get it to stick again..

Is any one else experiencing this behavior?

yoramrotbach, 2023-02-24:
Yes
I have the same "POPS" 

Yoram

K7MDL2, 2023-03-19:
I get audible and visual "popping" when AGC in SDR++ is enabled.  I disable AGC and manually adjust AGC gain to the point where it begins to pop or just below.   LIkely unrelated, I do have image mirroring on the data from the CaribouLite source for some reason, like the I and Q is not sent or received, just I + I maybe.  If I choose a RTLSDR device on the same SDR++ Server PI connection, I and Q is correct.   Auto AGC works on the RTLSDR source (no pops).   Suggests to me the data from the CaribouLite is maybe I and Q.  Runing on main branch on a Pi4B.