cariboulabs/cariboulite #65: couldn't open smi driver file '/dev/smi'

bcshort, 2023-02-20:
Hi All.

Hardware: Raspi Zero W v 1.3
Software: RaspberryPi OS bullseye 32bit

I've been having trouble with the kernel module appears to fail (log below) with couldn't open smi driver file '/dev/smi' This error seems to occur AFTER rebooting the Zero when it had no issue. Rebooting does not resolve the issue, and I seem to have patchy success in cloning the main repo and recompiling. Log below.

The one time I have managed to get SoapySDRServer to see anything, connecting to it just saw Gqrx or CubicSDR lock up, or operate VERY slowly. 

I'm not particularly code-savvy so any advice appreciated :)

==========================
02-20 20:42:28.245  1186  1186 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
02-20 20:42:28.245  1186  1186 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
02-20 20:42:28.255  1186  1186 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
02-20 20:42:28.828  1186  1186 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:460 couldn't open smi driver file '/dev/smi'
02-20 20:42:28.831  1186  1186 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
02-20 20:42:28.843  1186  1186 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
02-20 20:42:28.850  1186  1186 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-20 20:42:28.859  1186  1186 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-20 20:42:28.863  1186  1186 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-20 20:42:28.864  1186  1186 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
02-20 20:42:28.868  1186  1186 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-20 20:42:28.871  1186  1186 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
02-20 20:42:28.940  1186  1186 E CARIBOULITE Test main@cariboulite_test_app.c:68 driver init failed, terminating...
==========================

daviddawe1982, 2023-02-20:
Are you using deveop r1 branch?
If so try switching to the main.

bcshort, 2023-02-20:
Hi David, 
Can confirm I am using the 'main' branch :)

bcshort, 2023-02-23:
Fresh eyes found PEBKAC Error.
at some point while troubleshooting and using someone else's notes, I had added dtoverlays=smi to /boot/config.txt which caused a conflict. Commented out the line and rebooting resolved this issue and I can now get to the menu again with caribou_test_app.

I am however still getting errors after loading SoapySDRServer and attempting to connect:
` CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:963 SMI reading operation failed`

I'll follow that one up in a seperate issue if one is not already open

closed 2023-02-23 by bcshort