cariboulabs/cariboulite #66: install.sh failed with "initializing pigpio failed"

zhangtemplar, 2023-02-20:
when I run install.sh, I got the following errors (all other steps are good). I am using the latest code as 02/19/2023
```
3. Main software...
mkdir: cannot create directory ‘build’: File exists
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-20 00:23:09.386  7756  7756 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-20 00:23:09.388  7756  7756 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-20 00:23:09.390  7756  7756 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-20 00:23:09.392  7756  7756 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-20 00:23:09.398  7756  7756 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-20 00:23:09.402  7756  7756 V initCheckPermitted@pigpio.c:7396 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

02-20 00:23:09.409  7756  7756 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed
02-20 00:23:09.410  7756  7756 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:205 Error setting up io_utils
[ERROR] cariboulite_init_driver() failed
02-20 00:23:09.411  7756  7756 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-20 00:23:09.421  7756  7756 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
02-20 00:23:09.423  7756  7756 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-20 00:23:09.573  7765  7765 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-20 00:23:09.574  7765  7765 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-20 00:23:09.576  7765  7765 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-20 00:23:09.578  7765  7765 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-20 00:23:09.580  7765  7765 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-20 00:23:09.582  7765  7765 V initCheckPermitted@pigpio.c:7396 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

02-20 00:23:09.587  7765  7765 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed
02-20 00:23:09.589  7765  7765 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:205 Error setting up io_utils
[ERROR] cariboulite_init_driver() failed
02-20 00:23:09.591  7765  7765 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-20 00:23:09.598  7765  7765 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
02-20 00:23:09.600  7765  7765 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
-- Configuring done
-- Generating done
-- Build files have been written to: /home/qiang/projects/cariboulite/build
[ 12%] Built target iir
[ 20%] Built target datatypes
[ 24%] Built target ustimer
[ 27%] Built target caribou_prog
[ 29%] Built target caribou_fpga
[ 37%] Built target at86rf215
Scanning dependencies of target caribou_smi
[ 38%] Building C object src/caribou_smi/CMakeFiles/caribou_smi.dir/caribou_smi_modules.c.o
[ 40%] Linking C static library libcaribou_smi.a
[ 42%] Built target caribou_smi
[ 53%] Built target io_utils
[ 55%] Built target rffc507x
[ 59%] Built target hat
[ 66%] Built target production_utils
[ 70%] Built target zf_log
[ 75%] Built target cariboulite
[ 77%] Linking CXX executable cariboulite_util
[ 79%] Built target cariboulite_util
[ 81%] Linking CXX executable cariboulite_test_app
[ 83%] Built target cariboulite_test_app
[ 85%] Linking CXX executable test/fpgacomm
[ 87%] Built target fpgacomm
[ 88%] Linking CXX executable test/caribou_programmer
[ 90%] Built target caribou_programmer
[ 92%] Linking CXX shared module libSoapyCariboulite.so
[100%] Built target SoapyCariboulite
[ 12%] Built target iir
[ 20%] Built target datatypes
[ 24%] Built target ustimer
[ 27%] Built target caribou_prog
[ 29%] Built target caribou_fpga
[ 37%] Built target at86rf215
[ 42%] Built target caribou_smi
[ 53%] Built target io_utils
[ 55%] Built target rffc507x
[ 59%] Built target hat
[ 66%] Built target production_utils
[ 70%] Built target zf_log
[ 75%] Built target cariboulite
[ 79%] Built target cariboulite_util
[ 83%] Built target cariboulite_test_app
[ 87%] Built target fpgacomm
[ 90%] Built target caribou_programmer
[100%] Built target SoapyCariboulite
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.8-2/libSoapyCariboulite.so
-- Set runtime path of "/usr/local/lib/arm-linux-gnueabihf/SoapySDR/modules0.8-2/libSoapyCariboulite.so" to ""
-- Up-to-date: /usr/local/lib/libcariboulite.a
-- Installing: /usr/local/bin/cariboulite_util
-- Set runtime path of "/usr/local/bin/cariboulite_util" to ""
```

zhangtemplar, 2023-02-20:
When I ran `cariboulite_test_app`, I got the similar errors:
```
sudo ./cariboulite_test_app 
02-20 12:21:26.897 11788 11788 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-20 12:21:26.901 11788 11788 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-20 12:21:26.903 11788 11788 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-20 12:21:26.904 11788 11788 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-20 12:21:26.907 11788 11788 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-20 12:21:27.016 11788 11788 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
02-20 12:21:27.018 11788 11788 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
02-20 12:21:27.020 11788 11788 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
02-20 12:21:27.022 11788 11788 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
02-20 12:21:27.024 11788 11788 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
02-20 12:21:27.025 11788 11788 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
02-20 12:21:27.027 11788 11788 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
	System Version: FF
	Manu. ID: FF
	Sys. Ctrl Version: FF
	IO Ctrl Version: FF
	SMI Ctrl Version: FF
02-20 12:21:27.031 11788 11788 I CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
02-20 12:21:27.033 11788 11788 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:259 Sending bitstream of size 32220
02-20 12:21:30.223 11788 11788 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:281 bitstream sent 32220 bytes
02-20 12:21:30.225 11788 11788 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
02-20 12:21:30.226 11788 11788 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:203 0 dummy clocks sent
02-20 12:21:30.226 11788 11788 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:291 FPGA programming - Success!

FPGA Versions:
	System Version: 01
	Manu. ID: 01
	Sys. Ctrl Version: 01
	IO Ctrl Version: 01
	SMI Ctrl Version: 01
02-20 12:21:30.331 11788 11788 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
02-20 12:21:30.331 11788 11788 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
02-20 12:21:30.333 11788 11788 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
02-20 12:21:30.333 11788 11788 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
02-20 12:21:30.333 11788 11788 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
02-20 12:21:30.333 11788 11788 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
02-20 12:21:30.333 11788 11788 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: 9d2d5d98-384d-4f8b-998c-1ec5ae1e7729, Numeric serial: 0x3CECCB24
02-20 12:21:30.333 11788 11788 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
02-20 12:21:30.333 11788 11788 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
02-20 12:21:30.333 11788 11788 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
02-20 12:21:30.333 11788 11788 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
02-20 12:21:30.333 11788 11788 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
02-20 12:21:30.342 11788 11788 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
02-20 12:21:30.348 11788 11788 E CARIBOU_SMI_MODULES caribou_smi_insert_smi_modules@caribou_smi_modules.c:71 Module insertion 'smi_stream_dev' failed
02-20 12:21:30.350 11788 11788 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:451 Problem reloading SMI kernel modules
02-20 12:21:30.351 11788 11788 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
02-20 12:21:30.354 11788 11788 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
02-20 12:21:30.358 11788 11788 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-20 12:21:30.359 11788 11788 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-20 12:21:30.360 11788 11788 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-20 12:21:30.362 11788 11788 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
02-20 12:21:30.363 11788 11788 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-20 12:21:30.364 11788 11788 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
02-20 12:21:30.441 11788 11788 E CARIBOULITE Test main@cariboulite_test_app.c:68 driver init failed, terminating...
```

bcshort, 2023-02-21:
Hi @zhangtemplar 

I Had the same issue, and it turned out to be a mismatch between the running Linux kernel and the installed header files.

Have a try doing `apt update && apt upgrade && reboot` prior to cloning the git repo and running install.sh and try again :)

zhangtemplar, 2023-02-26:
thanks @bcshort , let me try

zhangtemplar, 2023-02-27:
Still got the error

```
sudo ./cariboulite_test_app 
02-26 20:02:04.583  7934  7934 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-26 20:02:04.587  7934  7934 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-26 20:02:04.588  7934  7934 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-26 20:02:04.590  7934  7934 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-26 20:02:04.592  7934  7934 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-26 20:02:04.703  7934  7934 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
02-26 20:02:04.704  7934  7934 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
02-26 20:02:04.706  7934  7934 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
02-26 20:02:04.707  7934  7934 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
02-26 20:02:04.709  7934  7934 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
02-26 20:02:04.711  7934  7934 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
02-26 20:02:04.713  7934  7934 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
	System Version: FF
	Manu. ID: FF
	Sys. Ctrl Version: FF
	IO Ctrl Version: FF
	SMI Ctrl Version: FF
02-26 20:02:04.716  7934  7934 I CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
02-26 20:02:04.719  7934  7934 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:259 Sending bitstream of size 32220
02-26 20:02:07.612  7934  7934 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:281 bitstream sent 32220 bytes
02-26 20:02:07.615  7934  7934 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
02-26 20:02:07.617  7934  7934 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:203 0 dummy clocks sent
02-26 20:02:07.619  7934  7934 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:291 FPGA programming - Success!

FPGA Versions:
	System Version: 01
	Manu. ID: 01
	Sys. Ctrl Version: 01
	IO Ctrl Version: 01
	SMI Ctrl Version: 01
02-26 20:02:07.730  7934  7934 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
02-26 20:02:07.731  7934  7934 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
02-26 20:02:07.733  7934  7934 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
02-26 20:02:07.734  7934  7934 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
02-26 20:02:07.736  7934  7934 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
02-26 20:02:07.736  7934  7934 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
02-26 20:02:07.736  7934  7934 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: 9d2d5d98-384d-4f8b-998c-1ec5ae1e7729, Numeric serial: 0x3CECCB24
02-26 20:02:07.737  7934  7934 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
02-26 20:02:07.738  7934  7934 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
02-26 20:02:07.739  7934  7934 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
02-26 20:02:07.739  7934  7934 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
02-26 20:02:07.739  7934  7934 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
02-26 20:02:07.749  7934  7934 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
02-26 20:02:07.788  7934  7934 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:460 couldn't open smi driver file '/dev/smi'
02-26 20:02:07.790  7934  7934 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
02-26 20:02:07.797  7934  7934 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
02-26 20:02:07.799  7934  7934 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
02-26 20:02:07.800  7934  7934 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
02-26 20:02:07.808  7934  7934 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
02-26 20:02:07.824  7934  7934 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
02-26 20:02:07.827  7934  7934 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
02-26 20:02:07.827  7934  7934 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
02-26 20:02:07.915  7934  7934 E CARIBOULITE Test main@cariboulite_test_app.c:68 driver init failed, terminating...
```

zhangtemplar, 2023-02-27:
Confirmed /dev/smi exists but when try to open it got this error: 
```
sudo cat /dev/smi 
cat: /dev/smi: Cannot allocate memory
```

zhangtemplar, 2023-03-04:
One thing weird is that with `lsmod | grep smi`, I got
```
smi_stream_dev         20480  0
bcm2835_smi            20480  1 smi_stream_dev
```
instead of
```
bcm2835_smi_dev     16384   0
bcm2835_smi         20480   1   bcm2835_smi_dev
```

K7MDL2, 2023-03-18:
> One thing weird is that with `lsmod | grep smi`, I got
> 
> ```
> smi_stream_dev         20480  0
> bcm2835_smi            20480  1 smi_stream_dev
> ```
> 
> instead of
> 
> ```
> bcm2835_smi_dev     16384   0
> bcm2835_smi         20480   1   bcm2835_smi_dev
> ```

I initially got the same results as you.  smi_stream_dev appears to be what is now expected as of today.  
https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/caribou_smi/README.md appears out of date.
After much problem solving today somewhere along the way I lost one of the smi entries and now only have 

bcm2835_smi         20480   0  bcm2835_smi_dev

pi@raspberrypi:~/projects/cariboulite $ cat /proc/modules
...
bcm2835_smi 20480 0 - Live 0x0000000000000000
...
The above is the only smi related line in modules.

I believe this and different kernel lib header file versions (also brought up in issue #55) are now causing the below errors (with further dependent downstream errors) and perhaps the last things between me and a working system.

03-17 20:36:47.223 12343 12343 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
03-17 20:36:47.224 12343 12343 E CARIBOU_SMI_MODULES caribou_smi_insert_smi_modules@caribou_smi_modules.c:71 Module insertion 'smi_stream_dev' failed
03-17 20:36:47.224 12343 12343 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:434 Problem reloading SMI kernel modules
03-17 20:36:47.224 12343 12343 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule

The kernel header mismatch (this is part of the cariboulite install.sh script)

pi@raspberrypi:~/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build $ sudo cmake ../
-- Kernel release: 6.1.19-v8+
-- Kernel headers: /usr/src/linux-headers-5.15.84-v8+
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build
pi@raspberrypi:~/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build $ sudo make
compiling module smi_stream_dev.ko...
make[3]: *** /usr/src/linux-headers-5.15.84-v8+: No such file or directory.  Stop.
make[2]: *** [CMakeFiles/smi_stream_dev.dir/build.make:87: smi_stream_dev] Error 2
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/smi_stream_dev.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

This error does not appear to stop the install script which finishes with "All Good, now reboot" messages.

I am on the latest bullseye OS on Pi4B.

In summary I see 4 things:
1. The kernel header files mismatch I think is still a problem.
2. The make file error for the above is not caught or corrected.
3. The https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/caribou_smi/README.md is out of date
4. I lost a module entry and do not know how to fix it (updates/upgrades and reboots have not fixed it)

Looking for how to resolve the above issues.

K7MDL2, 2023-03-18:
Some more clues from dmesg fresh from a reboot.  See the last 2 lines about versions.
Possibly sorting out the headers mismatch might also fix up the module loading?
Will continue to poke around, not my area of expertise.  This is on a Pi4B with branch develop_R1.

```
pi@raspberrypi:~/projects/cariboulite $ dmesg | grep smi
[    5.633249] smi-bcm2835 fe600000.smi: initialised
[    5.867003] smi-dev-bcm2835 soc:smi_dev: initialised
[    6.771490] rc rc2: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
[  118.257462] smi-dev-bcm2835 soc:smi_dev: SMI character dev removed - OK
[  118.287653] smi_stream_dev: disagrees about version of symbol module_layout
[  118.453603] smi_stream_dev: disagrees about version of symbol module_layout
```

K7MDL2, 2023-03-18:
Focusing on the version issues.  I do not know why CMake is coming up with 5.15.84-v8+.    Here is the relevent portion of code in /home/pi/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/CMakeLists.txt 
```
# Module info
add_definitions(-D__KERNEL__ -DMODULE)

# Find the kernel release
execute_process(
        COMMAND uname -r
        OUTPUT_VARIABLE KERNEL_RELEASE
        OUTPUT_STRIP_TRAILING_WHITESPACE
)

# Find the headers
find_path(
        KERNELHEADERS_DIR
        include/linux/user.h
        PATHS /usr/src/linux-headers-${KERNEL_RELEASE}
)

message(STATUS "Kernel release: ${KERNEL_RELEASE}")
message(STATUS "Kernel headers: ${KERNELHEADERS_DIR}")
```

It puts out both the 6.1.19 and 5.15.84 for some reason.  My OS is 6.1.19 as seen with uname -r.  I edited 
/home/pi/projects/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build/CMakeCache.txt 

```
//Path to a file.
KERNELHEADERS_DIR:PATH=/usr/src/linux-headers-6.1.19-v8+
```
to force the header file path to the current OS version then ran cariboulite/install.sh again.  This time I can run the test and SDR apps and the smi module is added without error.  I still get SMI read timed out with SoapySDRServer but at least the driver errors are gone.

I now have this result.
```
pi@raspberrypi:~ $ lsmod | grep smi
smi_stream_dev         20480  0
bcm2835_smi            20480  1 smi_stream_dev
```
Now to find out why the SMI reads are timing out.  Using the caribouLite_test_app I can see data is read from the both channels.  below is the SMI read timeout error I always have.   This error message below happens to be from SDR++ server mode using SoapySDR.  The same SMI timeout error occurs on a direct Soapy connection and SoapySDRServer.  RTLSDR works fine over SoapySDR local and remote.  
```
CARIBOULITE Radio cariboulite_radio_read_samples@cariboulite_radio.c:971 SMI reading operation returned timeout
```

K7MDL2, 2023-03-18:
Progress. I switched from develop_R1 branch to main branch, ran install.sh.  The linux header version remained correct, Not  sure if it was the old modified file or it worked correctly yet.  In any case I rebooted and can now RX on SDR++ over SDR++Server (both on the RPi) network connection.  Runing SDR++ client on the RPi4 it is at 100% CPU and the audio is choppy.   Running the client on a Windows CPU the PI is at abut 14%.   There is a large hump in both clients display but the audio is clear for FM radio band.  The HF bands such as 5MHz has pops and low signal level with the display flashing at each pop.  It is a big step forward in any case.  

Later I tried CubicSDR and some other configs and the SMI timeout returned.   Rebooting the Pi and starting up sdrpp --server made it work again, same hump in display but audio not bad, some visual flickering,  On a PC, the CaribouLite HiF RX, listening to FM stereo, SDR++ reports around 50-57Mbit/s rate.  The Pi is at 12%.  The FM stations are visible riding along the hump.  The RTLSDR shows a proper spectrum with larger signal amplitudes, same antenna.  So it is a bit fragile, but it is now within reach of work right.  

This is on the main branch.  The SMI kernel driver is significantly changed between main and develop_R1 branches according to posts.

[image]

I found the display and audio popping on the lower bands like 5Mhz AM or 7.074Mhz USB (FT-8) the audio pops and display show up but go away it seems if you turn off the AGC.   Raise the Gain high and it pops.  So something is funky with the AGC. Data rate is now around 26Mbit/s (compression on). 

Increasing the Source (REMOTE) bandwidth to 2MHz, changing to Float32, Decimation at something other than none, and adjusting the modem AGC gain down to around 35 the hump went away, sound is clear, spectrum and waterfall is pretty decent.  The FM band hump also diminishes.  I am getting mirror images however.  Uncompressed the stereo FM stations was 240MBits/s!!.  The Pi4B seemed not bothered.  Over time while writing these notes the sdr++ client eventually faded out and then completely stopped updating the spectrum and audio, the spectrum width also failed to widen according to the bandwidth settings.  Restarting the client solved those problems.  Decimation = 4-8 seemed like a good compromise.

[image]

Hopefully these notes will help make progress.

- Mike

zhangtemplar, 2023-03-18:
I found the issue. I made a mistake between `i2c_vc`, `i2c_arm` and `i2c`. Now it is working properly.

closed 2023-03-18 by zhangtemplar

K7MDL2, 2023-03-19:
The original problem with the install.sh messages requiring sudo is fixed if you place sudo in front of cmake on line 117 per below.

```
printf "${CYAN}3. Main software...${NC}\n"
cd $ROOT_DIR
mkdir build
cd build
sudo cmake $ROOT_DIR/software/libcariboulite/
make
sudo -u root make install
```

edegraaff, 2023-03-19:
> I found the issue. I made a mistake between `i2c_vc`, `i2c_arm` and `i2c`. Now it is working properly.

should you enable those in the /boot/config.txt or disable them ? the insttall scripts mention to add a line, while the main site advices to disable them ...

zhangtemplar, 2023-03-20:
you should only have `i2c_vc` enabled and the others should be disabled

Mudary, 2023-04-17:
Hi, wondering what or whose git version, cariboulite or konimaru are you running.
Thanks Mudary

K7MDL2, 2023-04-17:
konimaru fork is working for me.  I think the sudo cmake comment above still likely applies.