cariboulabs/cariboulite #68: Failed installing Soapy.

Alle1307, 2023-02-23:
Tried to install on a RPi- ZW with the latest and fresh OS. Followed the instruction as stated.
Stage 3 encountered problems:

[  3  ] Checking Soapy SDR installation (SoapySDRUtil)...
Did not find SoapySDRUtil. Do you want to clone and install? (Y/[N]): 
./install.sh: line 55: SoapySDRUtil: command not found
./install.sh: line 56: SoapySDRUtil: command not found
mkdir: missing operand
Try 'mkdir --help' for more information.

[  4  ] Checking the installed Soapy utilities...

Failed installing Soapy. Exiting...

How can I fix this?

daviddawe1982, 2023-02-23:
git clone https://github.com/pothosware/SoapySDR.git
cd SoapySDR

mkdir build
cd build
cmake ..
make -j4
sudo make install
sudo ldconfig

Install like above then run install.sh again

Alle1307, 2023-02-27:
OK, that worked so far. 
Now I got the error that python does not find the SoapySDR module.
Rerun install.sh gives the following errors and does not allow to be run under sudo.

3. Main software...
mkdir: cannot create directory 'build': File exists
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-27 22:25:12.681  2150  2150 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-27 22:25:12.687  2150  2150 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-27 22:25:12.691  2150  2150 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-27 22:25:12.695  2150  2150 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-27 22:25:12.701  2150  2150 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-27 22:25:12.705  2150  2150 V initCheckPermitted@pigpio.c:7401 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

02-27 22:25:12.711  2150  2150 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed
02-27 22:25:12.711  2150  2150 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:205 Error setting up io_utils
[ERROR] cariboulite_init_driver() failed
02-27 22:25:12.716  2150  2150 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-27 22:25:12.725  2150  2150 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
02-27 22:25:12.726  2150  2150 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[INFO] SoapyCaribouliteSession, sessionCount: 0
02-27 22:25:12.937  2159  2159 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
02-27 22:25:12.940  2159  2159 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
02-27 22:25:12.944  2159  2159 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-27 22:25:12.946  2159  2159 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
02-27 22:25:12.949  2159  2159 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
02-27 22:25:12.955  2159  2159 V initCheckPermitted@pigpio.c:7401 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

02-27 22:25:12.958  2159  2159 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed
02-27 22:25:12.960  2159  2159 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:205 Error setting up io_utils
[ERROR] cariboulite_init_driver() failed
02-27 22:25:12.963  2159  2159 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
02-27 22:25:12.970  2159  2159 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
02-27 22:25:12.974  2159  2159 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released

jfoxr, 2023-03-19:
I have the sane issue and have swig installed

K7MDL2, 2023-04-02:
A setlocale error shows up, maybe try to clear it and try again?

"bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)"

I saw a couple resolutions possible when I did a search for that same error.