cariboulabs/cariboulite #69: Proposal to get alligned.

edegraaff, 2023-02-24:
All, i would like to make an proposal.
We have a discussions tab and we have the main page.
We all try to get the cariboulite hat working on all kind of platforms and hardware.
Would it be an idea to:
- ask the founders to investigate what hardware and what os is at this moment supported and signed off as working
- ask the founders to turn the procedure 180 degrees and when the want to let one of us to do a specific test on os or hardware they announce it in the discussion forum of in the main page or here? 
- make a seperate location with build procedures that have been tested and deliverd a result or partly result where the result is also documented

For me:  i have build with succes on a pi zero 2n but tuning is still an issue, i am willing to rewrite what i have done, the depends, and so on, and the outcome. 

It could be an approach for the founders and developers to have a bette overview in progress and prioritize the issues.

We all want to have a working cariboulite, and for me, it would be great if it runs on a pizero but when we conclude it needs a pi4 it is better to make that statement than try time over time to get something working on hardware that does not suit the needs. 

regards 
Eelco

gerd0355, 2023-02-24:
All,
I like the proposal of edegraaff.
If we want to have a working cariboulite, it would be better to concentrate for a pi4 or or a pi zero.
I think the entrance for many users should be a simple and useable desription of the installation of SDR++ on an RPI 4 or an RPI zero.
The next step should be to describe methods for persons who are not experts on Linux or C++ or python. 
I say it directly: The specifications of cariboulite are very good, but the support of the founders is not so good. 

Thank you and regards
gerd0355

edegraaff, 2023-02-24:
I think the main issue is time, the project has been uder pressure, lets see if we could things step by step up and running, the specs are good, so we should find some benefits :p)

zbyrek, 2023-02-25:
> The next step should be to describe methods for persons who are not experts on Linux or C++ or python.

That is why i thought about premade image with everything installed, i spent way too much time to discover that cariboulite was working, just sdr++ got issues when run as root, because you have to run pulseaudio as root.

There is still bug in firmware or software that for 6GHz channel samples for Q are same as for I (that is why only half bandwidth is kinda usable and other half is mirrored, sdr program can't distinguish between fc+fb and fc-fb), first i thought that modem is in wrong mode (baseband), but then when reading datasheet i noticed that in baseband mode is not using same pins as IQ mode and also I got IQ version of IC that doesn't have baseband at all.

konimaru, 2023-02-26:
Just to chime in here, I'm all for this being a concerted effort. I had my share of (private) swearing wrt getting started but that doesn't get me/us anywhere. Taking on one problem at a time always worked for me ...

ImDroided, 2023-02-28:
I bought a $50 Firefly Libre ROC-RK3328-CC to try to get this working with. It is a raspberry Pi form factor, quad core ARM coretex-A53, gig ethernet, USB 3, 4gbs DDR4, and a tiny ass emmc spot that I picked up a 32gb for $20. 

I have it installed on a pi3 currently because I was running into a roadblock with the other board. For the most part everything got installed but was having issues with the smi stuff. When I get more time I will take another stab at it because the ROC-3328-CC is way better priced currently $50 on amazon and decently loaded for the price.

stevewells20, 2023-02-28:
As far as I can tell from the datasheet, the RK3328 doesn't have a Secondary Memory Interface. This interface is what allows data transfer from the Cariboulite to the Raspberry Pi. Unless I'm missing something, the RK3328 is not compatible with the Cariboulite.