cariboulabs/cariboulite #70: IQ stream is mirrored

yoramrotbach, 2023-02-24:
Hi,

Finally managed to make the CaribouLite work.
My setup: RaspiOS 64bit on Raspberry PI 3B
Latest main branch
Although there is a bug in the install.sh script which fails to install SoapySDRServer, I've installed it manually and it is working
The application I'm using is GQRX on my Mac
External VHF antenna
[image]

When connecting to the SoapySDRServer, the spectrum display is mirrored.
Is there a way to fix this?

In addition, there are those "POPS" that occur every few seconds and appear as a bright line across the spectrum

Yoram

zbyrek, 2023-02-24:
Well this is because I=Q, there must be bug in software or firmware:
[image]

zbyrek, 2023-02-25:
The mirror issue here is that samples for Q and I channels are same for 6GHz channel (for sub1Ghz channel everything is fine), there must be a bug in firmware or software.

Pops thing is probably because some samples are lost.

G0MJW, 2023-02-25:
I am also seeing this now I have fixed the broken hardware. I and Q are the same so everything is mirrored. I also lose many samples, this is on a Pi Zero 2W connected via USB, WiFi is unusable. No doubt the lost samples are due to the CPU but the mirroring is likely to be software.

Ideally we should decimate before trying to send 4 Ms x 32 bits over the USB.

Sub 1 GHz channel doesn't receive much of interest except 70cm but you are right that it is not mirrored. Hmm, let's hope it really is software.

yoramrotbach, 2023-02-25:
Same behavior also on a PI400. So I don't think it has to do with CPU. WiFi on the RPI 3B failed and it was able to push only ~3Mb/sec. When connected via ethernet this is pushing ~17Mb/sec
I agree that the sub 1GHz port is not as useful (at least for me)

konimaru, 2023-02-27:
It's not always this easy but the fix below works for me (effectively a swap operation gone wrong).
```
diff --git a/software/libcariboulite/src/caribou_smi/caribou_smi.c b/software/libcariboulite/src/caribou_smi/caribou_smi.c
index d85914d..3e08417 100644
--- a/software/libcariboulite/src/caribou_smi/caribou_smi.c
+++ b/software/libcariboulite/src/caribou_smi/caribou_smi.c
@@ -268,6 +268,7 @@ static int caribou_smi_rx_data_analyze(caribou_smi_st* dev,
        else
        {
         unsigned int i = 0;
+        int16_t temp;
         // Print buffer
         //smi_utils_dump_bin(buffer, 16);
         
@@ -294,8 +295,9 @@ static int caribou_smi_rx_data_analyze(caribou_smi_st* dev,
                 // reverse phase in the high channel
                 if (channel == caribou_smi_channel_2400)
                 {
+                    temp           = cmplx_vec[i].i;
                     cmplx_vec[i].i = cmplx_vec[i].q;
-                    cmplx_vec[i].q = cmplx_vec[i].i;
+                    cmplx_vec[i].q = temp;
                 }
             }
         }
```

zbyrek, 2023-02-27:
> It's not always this easy (to find) but the fix below works for me (effectively a swap operation gone wrong).

Thanks! That fix the issue, i think this is good candidate for pull request.

As for pops it seems that it is just rf receiver clipping (or leaking 2.4GHz), lowering gain solves the issue.

G0MJW, 2023-02-27:
Swapping I and Q?
A
Let I = Q
Let Q = I

type error, which explains things. It would be good if this could be fixed quickly in Main.

The popping in my case is not gain, it is the Pi Zero USB losing samples. If we could decimate before sending, i.e. allow sample rates less than 4Ms it would probably fix it.  I assume we are sending 32 bits at 4Ms (16 I and 16 Q) which is 128Mb/s. That's less than 480Mb/s USB2 but there is overhead. Larger buffers may also fix it. Don't try to use WiFi as it probably won't fit..

yoramrotbach, 2023-03-01:
Thank you Konimaru

This solved the issue :)

> It's not always this easy (to find) but the fix below works for me (effectively a swap operation gone wrong).
> 
> ```
> diff --git a/software/libcariboulite/src/caribou_smi/caribou_smi.c b/software/libcariboulite/src/caribou_smi/caribou_smi.c
> index d85914d..3e08417 100644
> --- a/software/libcariboulite/src/caribou_smi/caribou_smi.c
> +++ b/software/libcariboulite/src/caribou_smi/caribou_smi.c
> @@ -268,6 +268,7 @@ static int caribou_smi_rx_data_analyze(caribou_smi_st* dev,
>         else
>         {
>          unsigned int i = 0;
> +        typeof(samples_out->i) temp;
>          // Print buffer
>          //smi_utils_dump_bin(buffer, 16);
>          
> @@ -294,8 +295,9 @@ static int caribou_smi_rx_data_analyze(caribou_smi_st* dev,
>                  // reverse phase in the high channel
>                  if (channel == caribou_smi_channel_2400)
>                  {
> +                    temp           = cmplx_vec[i].i;
>                      cmplx_vec[i].i = cmplx_vec[i].q;
> -                    cmplx_vec[i].q = cmplx_vec[i].i;
> +                    cmplx_vec[i].q = temp;
>                  }
>              }
>          }
> ```

edegraaff, 2023-03-16:
> Hi,
>
> Finally managed to make the CaribouLite work. My setup: RaspiOS 64bit on Raspberry PI 3B Latest main branch Although there is a bug in the install.sh script which fails to install [...]

Ok, i will make a new clean install with gui as last resort, i am more and more puzzeling if the device itselve is not broke, i am not able to tune the soapysdrserver i use cubicsdr on windows, it connects audiostream works, but tuning is an issue.