cariboulabs/cariboulite #71: Initial setup questions

mrspirytus, 2023-02-24:
Hello,

I am about to start setting up my unit I did recived about a month ago. After reading info on main page and croudsource page, I have some general questions.

1. It sounds like DragonOS is a ready image with everything preset? Should run install.sh at least once after the first boot? Or is it intended for post-first boot using "Raspberry Pi Imager" (Any Debian image) and not DragonOS?
2. Looks like cloning and building is done on rPi itself. (No cross compiling) . How much space is required? I have a few 8G SD cards free. Or should I get 16 or larger?
3. How do you check which firmware you running and how do you upgrade in future? I do not see anything in the install.sh script related to firmware upgrade.
4. In the future, besides ```apt upgrade``` to upgrade system, do you just rerun the install.sh again after pulling the latest? or one should always reclone the latest so you rebuild with clean. How does the incremental build looks now. Any issues?
5. Related to HF Support #14 , looks like there are no additional changes/configurations to Rx in HF bands. So SDR++ or any other software should be able to access that renge. Correct?
6. Anyone did add an RFI shield? Did you notice the lower noise floor?  Can you purchase a shield anywhere or did you just make it yourself?

Thank you

yoramrotbach, 2023-02-25:
The install.sh script isn’t built for DragonOS
1. the config.txt file is located in /boot/firmware
2. the script assumes the the user is pi not user ubuntu
3. biggest problem i’ve encountered is with the kernel module not loading

so far i’ve used the git pull command to update the files and re run the install.sh script. Assuming that the SoapySDR stuff is already installed, it will skip this part the second time.
there is a bug in the script that fails to install SoapySDRServer (wrong path). this is important if you wish to run an SDR application on a different computer

alphafox02, 2023-02-25:
To run the installer on DragonOS open it and remove raspberrypi-kernel-headers from the apt install. Also just change the lines that say /boot/config.txt to /boot/firmware/config.txt as mentioned above. Then make the edits to the config.txt that it tells you to.

I check the code about a week ago and I was able to use the CaribouLite by running sudo SoapySDRServer —bind and then connect to it from another DragonOS laptop.

I will check again this week and see how it’s all progressed, goal being to include everything running by default in DragonOS Pi64.

ImDroided, 2023-02-28:
I picked up a aluminum case and drilled 2 holes for the sma connections. looks real nice and should provide shielding.

mrspirytus, 2023-03-01:
> I picked up a aluminum case and drilled 2 holes for the sma connections. looks real nice and should provide shielding.

Mind sharing the part list/link and overall impression/picture?

edegraaff, 2023-03-16:
> > I picked up a aluminum case and drilled 2 holes for the sma connections. looks real nice and should provide shielding.
> 
> Mind sharing the part list/link and overall impression/picture?

yes please for me to

ImDroided, 2023-03-19:
It's a basic aluminum enclosure that I laser burned. I have not gotten my setup working yet and I am kind of worried about heat so I have been running with the top off. I may look for a bigger case to house a fan with it. I don't remember in what post but somebody mentioned that the thing warmed up pretty good and so did the pi. 
[image]
[image]
[image]

ImDroided, 2023-03-19:
[image]

edegraaff, 2023-03-19:
Pi3 case right ?

ImDroided, 2023-03-19:
> Pi3 case right ?

Correct. These can be found on Amazon or eBay for like 10 bucks or less

ImDroided, 2023-03-19:
But again, I would like to mention that somebody noted in another thread that they noticed that their pie and their caribou we're getting kind of warm so I have not put the lid on this case for fear of overheating.