cariboulabs/cariboulite #72: Cariboulite Python examples

gerd0355, 2023-02-24:
Hi,
my setup: RPi OS 32bit on RPi 4B, latest main branch
I have installed the cariboulite software as described in README.md. 
If I run cariboulite_test_app it works very well. 
I wanted to start the examples for python. 
When I run radio_test.py, I had a problem, because Python 3.9 does not match to Numpy 1.24.2.
I tried the example "soapy_synth.py",but the RPi could not find the module SoapySDR.
Although I have installed the SoapySDR APIs for CaribouLite as described in README.md. By the way the descripiton 
is not very exactly.

hudouseko, 2023-02-25:
Hi, I have experienced the same issue - while SoapySDR was installed automatically using install.sh from cariboulite, for some reason respective python files for SoapySDR were not installed. What helped me as a workaround was to go through installation process of SoapySDR from source (https://github.com/pothosware/SoapySDR/wiki/BuildGuide). I did this on RPi ZeroW, Raspbian 11 bullseye.

gerd0355, 2023-02-25:
Hi,
Thank you for the hints. 
But I was only successful in installing the module SoapySDR. So I could start the example "soapy_synth.py". But it doesn't work. I got many errors. 
The TX was not activated. 
The problem with the different versions of Numpy and Python is not solved!

closed 2023-03-24 by gerd0355