cariboulabs/cariboulite #77: Soapy issue

stevewa2066, 2023-02-28:
Seems to hang on installation on a PI 3. 64 bit

[image]

Any suggestions?

I'll try to install Soapy 
From the GitHub site?

Steve

konimaru, 2023-02-28:
What's your memory situation at this point?
```
free -h
```
I had similar issues compiling something else but soon ran out of swap and physical memory. Increasing swap size to 2G solved this for me.

yoramrotbach, 2023-02-28:
Make sure the make command uses 3 cores or less. With 4 cores used, the PI hangs and barely responsive.
Also make sure you have 1G of swap

stevewa2066, 2023-02-28:
I will check it out.

Thanks!