cariboulabs/cariboulite #81: Progress on the pi zero build procedure progress till here...

edegraaff, 2023-03-05:
I can confirm the whole build procedure does work. 
This is my current procedure

20-02-2023
pi os light 64 bit
increase swap 
sudo dphys-swapfile swapoff
sudo nano /etc/dphys-swapfile
CONF_SWAPSIZE=2048
sudo dphys-swapfile setup
sudo dphys-swapfile swapon
sudo reboot

sudo apt install libczmq-dev git cmake g++ libpython3-dev python3-numpy swig avahi-daemon libavahi-client-dev raspberrypi-kernel-headers module-assistant pkg-config libncurses5-dev libzmq3-dev rfkill -y
sudo apt-get install python3-distutils --reinstall

cd /home/pi
SoapySDR
git clone https://github.com/pothosware/SoapySDR.git
cd SoapySDR
mkdir build
cd build
cmake ..
make -j4
sudo make install
sudo ldconfig

cd \home\pi

SoapyRemote
git clone https://github.com/pothosware/SoapyRemote.git
cd SoapyRemote
mkdir build
cd build
cmake ..
make
sudo make install

sudo ldconfig
cd \home\pi

Cariboulite
mkdir ~/projects
cd ~/prsoject
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite
./install.sh

sudo SoapySDRServer --bind

and from there i can connect with openwebrx as well as cubicsdr SoapyRemote but.. the output is impossible to understand, i tried to send with my porto, and listen, i see the waterfall, but the sound is inreconisable. 

It is working but for me it is way from a functional SDR that can be used. At least in my situation, i a curious who did ave a working SDR now and how ?

bcshort, 2023-03-06:
What version of Pi Zero? Zero W 1.3? or 2.0?
From my experience the Pi Zero W 1.3 (which is what I have), did not have the CPU resources to run SoapySDR. In fact the Pi 3 rev B is maxed out as well, though it can run it (something like 80% CPU for Soapy, and 15% for other overheads).

konimaru, 2023-03-06:
Also, how do  you get the data off chip? You'll need at least 128Mbit for CS16 (64Mbit for CS8).
For me - using a 3B with full 64bit install - I managed CS8 over the builtin ethernet or CS16 over an attached Gbit adapter. I can't confirm the **maxed out** bit though.

I never actually tried listening in before the **I==Q patch** so I can't comment on that but do you have the patch installed (#75)? Otherwise **HiF** mode will have a mirrored spectrum (and may sound odd).

daviddawe1982, 2023-03-06:
I have it working on a Zero 2w using a soapy remote with the #75 patch installed currently monitoring ADS-B data and uploading it to FlightAware and flightradar24 using 20% CPU and 115 MB ram. 

My main pc is currently displaying the data and proving the server for uploading via sdrangel.

Heat is an issue as without a fan the frequency will wander. My solution is to use a 90-degree pin header adapter to give the CPU some breathing room.

edegraaff, 2023-03-06:
> What version of Pi Zero? Zero W 1.3? or 2.0? From my experience the Pi Zero W 1.3 (which is what I have), did not have the CPU resources to run SoapySDR. In fact the Pi 3 rev B is maxed out as well, though it can run it (something like 80% CPU for Soapy, and 15% for other overheads).

Pi Zero 2w

edegraaff, 2023-03-06:
> I have it working on a Zero 2w using a soapy remote with the #75 patch installed currently monitoring ADS-B data and uploading it to FlightAware and flightradar24 using 20% CPU, 115 MB ram and CF64 via ethernet adapter.
> 
> My main pc is currently displaying the data and proving the server for uploading via sdrangel.
> 
> Heat is an issue as without a fan the frequency will wander. My solution is to use a 90-degree pin header adapter to give the CPU some breathing room.

Hum, if you look at my installation script, what should i change or update? I could also try to get ads-b working since i have it running on a pi3 for several years.

daviddawe1982, 2023-03-07:
everything looks good except..
 
update projects/cariboulite/software/libcariboulite/src/caribou_smi/caribou_smi.c
as per https://github.com/cariboulabs/cariboulite/pull/75/commits/56d3047cf314ba8d8274747f57b2017dd8aee68e

other than that my installation is the same.

Gain is an issue also, I am not currently running an amplifier inline and as a result, I have to almost max out my software gain.

bcshort, 2023-03-07:
I've applied the phase reversal patch and this has seen an improvement with SoapySDRServer CPU usage dropping to 58%. This in part may be using CS8 as the remote format. I still however seem to be getting dropped audio at a fast frequency resulting in a warbling. .wav file attached (inside a zip) as sample.
[wavefile.zip](https://github.com/cariboulabs/cariboulite/files/10907782/wavefile.zip)

edegraaff, 2023-03-08:
Should we start with putting time and effort in getting a good working solution on pi3 and pi4 and from there improve to pi zero and or zero 2w since i have the feeling part of the issues is related to performance of the zero models. Maybe when the code is more stable and we are more in control we could take a look what should be minimal specs for a working solution? Beter to have a good approach than sdr/ham enthousiasts that hook off.

edegraaff, 2023-03-16:
I rebuild on a pi4 but it seems the tuning is an issue when you use soapy sdr server. Since i free-ed up a pi4 i could use one of the images with a full desktop and rebuild the whole and see if this could deliver a real working sdr. Until now, it is not working at all while the software works without errors.  One option could be my hat is not ok... ? , does something have a /boot/config.txt to compare?