cariboulabs/cariboulite #85: Sub-1GHz recognized, no mention of 2nd channel, dev not initialized

great9, 2023-03-16:
I got my full version of cariboulite yesterday and it seems that things are either broken in the software or the board.
I got a bunch of "caribou_fpga_set_smi_channel: dev not initialized".
What should be the output of a working 6GHz board?

`######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
03-16 18:50:03.661 10080 10080 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
03-16 18:50:03.661 10080 10080 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
03-16 18:50:03.661 10080 10080 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
03-16 18:50:03.661 10080 10080 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
03-16 18:50:03.661 10080 10080 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
03-16 18:50:03.774 10080 10080 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
03-16 18:50:03.775 10080 10080 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
03-16 18:50:03.775 10080 10080 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
03-16 18:50:03.775 10080 10080 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
03-16 18:50:03.775 10080 10080 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
03-16 18:50:03.775 10080 10080 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
03-16 18:50:03.775 10080 10080 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
03-16 18:50:03.775 10080 10080 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
        System Version: FF
        Manu. ID: FF
        Sys. Ctrl Version: FF
        IO Ctrl Version: FF
        SMI Ctrl Version: FF
03-16 18:50:03.775 10080 10080 I CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
03-16 18:50:03.777 10080 10080 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:259 Sending bitstream of size 32220
03-16 18:50:05.282 10080 10080 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:281 bitstream sent 32220 bytes
03-16 18:50:05.282 10080 10080 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
03-16 18:50:05.282 10080 10080 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:203 0 dummy clocks sent
03-16 18:50:05.282 10080 10080 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:291 FPGA programming - Success!

FPGA Versions:
        System Version: FF
        Manu. ID: FF
        Sys. Ctrl Version: FF
        IO Ctrl Version: FF
        SMI Ctrl Version: FF
03-16 18:50:05.382 10080 10080 E FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:182 Programming failed
03-16 18:50:05.382 10080 10080 E CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:599 FPGA programming failed
03-16 18:50:05.383 10080 10080 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
03-16 18:50:05.383 10080 10080 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
03-16 18:50:05.383 10080 10080 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
03-16 18:50:05.383 10080 10080 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
03-16 18:50:05.383 10080 10080 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
[ERROR] cariboulite_init_driver() failed
03-16 18:50:05.393 10080 10080 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
Printing 'findCariboulite' Request:
[INFO] Initializing DeviceID: 0, Label: CaribouLite S1G[9e02dae0], ChannelType: S1G
03-16 18:50:05.395 10080 10080 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:803 Activating channel 0, dir = RX, activate = 1
03-16 18:50:05.395 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.396 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.397 10080 10080 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:835 Setup Modem state tx_prep
03-16 18:50:05.397 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.397 10080 10080 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:849 Setup Modem state cmd_rx
03-16 18:50:05.427 10080 10080 E FPGA caribou_fpga_set_smi_channel@caribou_fpga.c:541 caribou_fpga_set_smi_channel: dev not initialized
03-16 18:50:05.427 10080 10080 E FPGA caribou_fpga_set_io_ctrl_dig@caribou_fpga.c:403 caribou_fpga_set_io_ctrl_dig: dev not initialized
03-16 18:50:05.427 10080 10080 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (1)
03-16 18:50:05.427 10080 10080 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:865 Failed to configure modem with cmd_rx
03-16 18:50:05.437 10080 10080 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:803 Activating channel 0, dir = RX, activate = 0
03-16 18:50:05.437 10080 10080 E FPGA caribou_fpga_set_smi_channel@caribou_fpga.c:541 caribou_fpga_set_smi_channel: dev not initialized
03-16 18:50:05.437 10080 10080 E FPGA caribou_fpga_set_io_ctrl_dig@caribou_fpga.c:403 caribou_fpga_set_io_ctrl_dig: dev not initialized
03-16 18:50:05.437 10080 10080 E CARIBOU_SMI caribou_smi_set_driver_streaming_state@caribou_smi.c:35 failed setting smi stream state (0)
03-16 18:50:05.468 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.468 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.468 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
[... 84 lines cut ...]
03-16 18:50:05.505 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.505 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.505 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.505 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.506 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.506 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.506 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.506 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.506 10080 10080 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:823 Setup Modem state trx_off
03-16 18:50:05.506 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.507 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.507 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.507 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.507 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.507 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.507 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.507 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.508 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.508 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.508 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
03-16 18:50:05.508 10080 10080 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
[INFO] Leaving Reader Thread

03-16 18:50:05.512 10080 10080 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
03-16 18:50:05.512 10080 10080 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released`

K7MDL2, 2023-03-17:
I just got my 6GHz unit today.  After some finagling I got SoapySDR built then the cariboulite install.sh script to run to end. 
 Got SDRPP recognizing the 2 transceivers but getting SMI timeouts when running.  Still investigating that.

In response to your question, here is what I get on a RPi 4B no errors.  note I had to run it with sudo.

pi@raspberrypi:~ $ sudo SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] SoapyCaribouliteSession, sessionCount: 0
03-16 18:33:33.746  9729  9729 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
03-16 18:33:33.746  9729  9729 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
03-16 18:33:33.746  9729  9729 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
03-16 18:33:33.747  9729  9729 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
03-16 18:33:33.747  9729  9729 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
03-16 18:33:33.841  9729  9729 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
03-16 18:33:33.841  9729  9729 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
03-16 18:33:33.841  9729  9729 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
03-16 18:33:33.841  9729  9729 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
03-16 18:33:33.841  9729  9729 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
03-16 18:33:33.841  9729  9729 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
03-16 18:33:33.841  9729  9729 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
03-16 18:33:33.841  9729  9729 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
	System Version: 01
	Manu. ID: 01
	Sys. Ctrl Version: 01
	IO Ctrl Version: 01
	SMI Ctrl Version: 01
03-16 18:33:33.841  9729  9729 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:188 FPGA already operational - not programming (use 'force_prog=true' to force update)
03-16 18:33:33.841  9729  9729 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
03-16 18:33:33.841  9729  9729 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
03-16 18:33:33.841  9729  9729 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
03-16 18:33:33.841  9729  9729 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
03-16 18:33:33.841  9729  9729 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
03-16 18:33:33.841  9729  9729 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
03-16 18:33:33.841  9729  9729 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: 4d7830e7-28bd-4bf8-9f95-d65f9081a6f2, Numeric serial: 0xFA504D6C
03-16 18:33:33.841  9729  9729 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
03-16 18:33:33.841  9729  9729 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
03-16 18:33:33.841  9729  9729 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
03-16 18:33:33.841  9729  9729 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
03-16 18:33:33.841  9729  9729 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:425 initializing caribou_smi
03-16 18:33:33.842  9729  9729 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:98 Unloading smi-stream module
03-16 18:33:33.874  9729  9729 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
SMI SETTINGS:
    width: 0
    pack: Y
    read setup: 0, strobe: 2, hold: 0, pace: 0
    write setup: 1, strobe: 4, hold: 1, pace: 0
    dma enable: Y, passthru enable: Y
    dma threshold read: 1, write: 63
    dma panic threshold read: 32, write: 32
03-16 18:33:33.882  9729  9729 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:286 INIT MODEM - AT86RF215
03-16 18:33:33.882  9729  9729 I AT86RF215_Main at86rf215_init@at86rf215.c:160 configuring reset and irq pins
03-16 18:33:33.882  9729  9729 I AT86RF215_Main at86rf215_init@at86rf215.c:168 Adding chip definition to io_utils_spi
03-16 18:33:33.882  9729  9729 I AT86RF215_Main at86rf215_init@at86rf215.c:197 Modem identity: Version: 03, Product: 35
03-16 18:33:33.884  9729  9730 D AT86RF215_Events at86rf215_radio_event_handler@at86rf215_events.c:61 INT @ RADIO09: Transceiver ready
03-16 18:33:33.894  9729  9729 D AT86RF215_Main at86rf215_calibrate_device@at86rf215.c:137 Calibration of modem: I=23, Q=29
03-16 18:33:33.906  9729  9729 D AT86RF215_Main at86rf215_calibrate_device@at86rf215.c:137 Calibration of modem: I=23, Q=27
03-16 18:33:33.917  9729  9729 D AT86RF215_Main at86rf215_calibrate_device@at86rf215.c:137 Calibration of modem: I=23, Q=27
[... cut ...]
03-16 18:33:35.076  9729  9729 I RFFC5072 rffc507x_release@rffc507x.c:208 Device release completed
03-16 18:33:35.076  9729  9729 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
03-16 18:33:35.076  9729  9729 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
03-16 18:33:35.082  9729  9729 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
pi@raspberrypi:~ $

great9, 2023-03-23:
Sucessfully working.
I made sure to remove all commented fields from /boot/config.txt and /boot/firmware/usercfg.txt on my rpi3b. and then added:
```
dtparam=audio=on
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=i2c_vc=on
dtoverlays=smi
dtoverlays=smi-dev
dtoverlay=dwc2,dr_mode=peripheral
```

closed 2023-03-23 by great9