cariboulabs/cariboulite #86: recompile raspian 64 bit full what sdr software does full ingegration wit carbulite?

edegraaff, 2023-03-17:
Since i am puzzeling for weeks to be able to receive a signal, from my porto or even a radiostation... i am able to compile without issues on a pi5, but it seems the cubicsdr is not recognising the sdr. What sdr program or best way to go forward?

sudo SoapySDRUtil --find does recognise the sdr

	MODEM Version: AT86RF215IQ (without basebands), version: 03Printing 'findCariboulite' Request:
Found device 0
  channel = S1G
  device_id = 0
  driver = Cariboulite
  label = CaribouLite S1G[ab0e17a0]
  name = CaribouLite RPI Hat
  serial = ab0e17a0
  uuid = 457dca56-0d60-41c2-9d9a-83b72f69af2d
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 1
  channel = HiF
  device_id = 1
  driver = Cariboulite
  label = CaribouLite HiF[ab0e17a1]
  name = CaribouLite RPI Hat
  serial = ab0e17a1
  uuid = 457dca56-0d60-41c2-9d9a-83b72f69af2d
  vendor = CaribouLabs LTD
  version = 0x0001

K7MDL2, 2023-03-17:
The S1G device you selected is limited to certain frequency ranges.  Your hardware interface says you are trying to tune at 400KHz which is out of range for S1G.  Can try 400Mhz which is just in range, better select the HiF 6GHz interface with a frequency above 1Mhz, preferable above 30MHz until things are working.

edegraaff, 2023-03-17:
Thanks for your comment, i tried it but seems still to fail, i am also not able to get the waterfall working, and or sound out of it...  Getting the whole working seems a process of hard working and trying:-(

K7MDL2, 2023-03-18:
I have been trying on a Pi4B a few SDR apps, still looking for success.  I can get RTLSDR direct and sometimes via SoapySDRServer working, or SDRpp Server, but when selecting CaribouLite radios I get no data transferred, usually get error messages on the server side saying SMI timed out.  Tried CubicSR, SDR++, now building SDRAngel.  The cariboulite test apps work fine as does SDRUtil.  Many library build variations tried.  So close, but not quite.  So you are not alone :-).  It will take some time to work out the finer points.