cariboulabs/cariboulite #87: SDR++ SPI

stevewa2066, 2023-03-18:
Hi, I get theses errors when loading SDR++. I run the SoapyUtil and it seem to read the CaribouLite just fine. 

running Pi 2 W

Any suggestions ?

[image]

K7MDL2, 2023-03-18:
Did you try running with sudo?

stevewa2066, 2023-03-18:
Wish it was that easy. Nope.. didn’t work.

Steve

edegraaff, 2023-03-19:
i have similar issues at this moment

stevewa2066, 2023-03-19:
I am going to run the Cariboulite Untility again and look at more closely to see if there is an issue. 

then again, it could be a SDR++ issue too.

konimaru, 2023-03-29:
Didn't try SDR++ yet, is it possible to convince it using the **CaribouLite** as remote device even on the same machine? Something along the lines of
```
remote=127.0.0.1, driver=remote, ...
```
This way I believe **SoapySDRServer** takes over which is - normally - running as root anyway.

K7MDL2, 2023-03-31:
Update, I now have OpenWebRX working on the CaribouLite.  It required some specific settings for SoapySDR I had missed.  This is covered in #90 .

So for my situation, SoapySDR seems to work fine for local apps, but I am having trouble via SoapySDR remote, at least from my Windows box.  SDR++ Server has been working great also on Pi4B 64bit.  I run both SoapySDRServer and sdrpp --server with sudo.

a-oxide, 2023-04-17:
@stevewa2066 is there any updates? I am struggling with the same issue. I also noticed when I run
`lsmod | grep smi`
I only get :
`bcm2835_smi         20480   1   bcm2835_smi_dev`
instead of : 
`bcm2835_smi_dev     16384   0
bcm2835_smi         20480   1   bcm2835_smi_dev`

I have tried it with both the @konimaru fork and the master branch.

a-oxide, 2023-04-17:
> 

I'm sure that the missing smi_dev has to do with this issue, but I don't know how to remedy it.

K7MDL2, 2023-04-17:
I believe you have the correct results, only 2. I am out of town now. I think i posted a working config showing only 2 lines.

K7MDL2, 2023-04-27:
I updated to the latest RPi.  header version changed.  uname -r gives 6.1.21-v8+.  I also switched to the main branch using fork https://github.com/konimaru/cariboulite.  Got the missing smi device issue.  Version headers were correct.  In install.sh, I added sudo in front of the cmake line and the make install line in the #4 section per below and that seemed to fix the issue for me as it has earlier.  Rebooted. working correctly now.

printf "${CYAN}4. Main software...${NC}\n"
cd $ROOT_DIR
mkdir -p build && cd build
sudo cmake $ROOT_DIR/software/libcariboulite/
make
sudo -u root make install

Was only getting 1 line before.
pi@raspberrypi:~/projects/cariboulite $ lsmod | grep smi
smi_stream_dev         20480  2
bcm2835_smi            20480  1 smi_stream_dev