cariboulabs/cariboulite #88: Cariboulite no hdmi

LeoBistmans, 2023-03-18:
after running install.sh, power off, install Cariboulite, no more hdmi.
Can reach the raspbian 10 pi4b via network.  
In dmesg:
[    7.033702] Console: switching to colour dummy device 80x30
[    7.042079] vc4_firmware_kms fe600000.firmwarekms: can't request region for resource [mem 0xfe600000-0xfe6000ff]
[    7.042213] [drm:vc4_ioremap_regs [vc4]] *ERROR* Failed to map registers: -16
[    7.042314] [drm:vc4_fkms_bind [vc4]] *ERROR* Oh dear, failed to map registers
[    7.042345] 8<--- cut here ---
[    7.042382] Unable to handle kernel paging request at virtual address fffffff0
[    7.042412] pgd = (ptrval)
[    7.042438] [fffffff0] *pgd=80000000007003, *pmd=2ff5e003, *pte=00000000
[    7.042514] Internal error: Oops: a07 [#1] SMP ARM
[    7.042537] Modules linked in: vc4(+) cec v3d(+) gpu_sched rfkill bcm2835_v4l2(C) drm_kms_helper bcm2835_smi_dev i2c_mux_pinctrl videobuf2_vmalloc i2c_mux bcm2835_codec(C) bcm2835_isp(C) i2c_bcm2835 bcm2835_smi bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common vc_sm_cma(C) drm drm_panel_orientation_quirks videodev snd_bcm2835(C) mc snd_soc_core snd_compress rpivid_mem snd_pcm_dmaengine snd_pcm snd_timer snd nvmem_rmem syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[    7.043021] CPU: 3 PID: 181 Comm: systemd-udevd Tainted: G         C        5.10.103-v7l+ #1529
[    7.043044] Hardware name: BCM2711
[    7.043149] PC is at vc4_fkms_bind+0x23c/0x320 [vc4]
[    7.043176] LR is at irq_work_queue+0x14/0x2c
[    7.043196] pc : [<bf396814>]    lr : [<c0350b24>]    psr: 60000013
[    7.043218] sp : c386dad8  ip : 00000000  fp : c386db3c
[    7.043237] r10: c3a80940  r9 : c3a808c0  r8 : 00040016
[    7.043257] r7 : c1a8d810  r6 : c28fe000  r5 : c3a80944  r4 : 00000000
[    7.043278] r3 : fffffff0  r2 : 6e4df213  r1 : 00000027  r0 : 00000042
[    7.043303] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    7.043326] Control: 30c5383d  Table: 037d24c0  DAC: 55555555
[    7.043347] Process systemd-udevd (pid: 181, stack limit = 0x(ptrval))
[    7.043369] Stack: (0xc386dad8 to 0xc386e000)
[    7.043390] dac0:                                                       c3a80940 c0330fd8
[    7.043423] dae0: a0000013 c1a8d9b0 c2926f40 c3a80940 c3a808c0 c1205048 00000000 c1a8d800
[    7.043456] db00: c0b978e4 00000001 00000002 6e4df213 c386db3c 00000000 c2927180 c2926f40
[    7.043488] db20: 00000018 bf3965d8 c28fe000 00000000 c386db7c c386db40 c0849528 bf3965e4
[    7.043519] db40: c386db64 c386db50 c0b978e4 c0e7957c 00000001 c28fe000 c1a8e210 eff4ec90
[    7.043551] db60: 00000000 c2927180 00000001 c12f0ddc c386dba4 c386db80 bf38f728 c0849424
[    7.043581] db80: c2926f40 c2927140 c33658c0 c2927140 c2927180 00000001 c386dbdc c386dba8
[    7.043613] dba0: c08499dc bf38f574 c0b92234 00000000 c12f0ddc c2927180 c12f0ddc c1a8e210
[    7.043643] dbc0: bf3b5974 bf38f348 c1a8e210 00000000 c386dc04 c386dbe0 c0849eb8 c0849878
[    7.043676] dbe0: c0c4d988 00000000 bf3c1734 c1205048 bf3b5ef8 00000000 c386dc44 c386dc08
[    7.043707] dc00: bf38f8d4 c0849dec c1a8d810 00000000 c33658c0 6e4df213 c0864574 c1a8e210
[    7.043738] dc20: 00000000 bf3c1150 00000000 00000000 bf3c1150 c1205048 c386dc64 c386dc48
[    7.043771] dc40: c0853964 bf38f82c c1a8e210 c1400544 c140054c 00000000 c386dc9c c386dc68
[    7.043802] dc60: c0851610 c0853918 c29267c0 00000024 c386dc9c c1a8e210 bf3c1150 bf3c1150
[    7.043833] dc80: c0851d00 c29267c0 bf3c18c0 c1205048 c386dcb4 c386dca0 c0851ab0 c0851514
[    7.043865] dca0: c1a8e210 00000000 c386dcd4 c386dcb8 c0851cf8 c0851a50 00000000 bf3c1150
[    7.043896] dcc0: c1a8e210 c0851d00 c386dcf4 c386dcd8 c0851d68 c0851c9c c1a5edb4 c1205048
[    7.043927] dce0: bf3c1150 c0851d00 c386dd24 c386dcf8 c084f648 c0851d0c c386dd30 c1974858
[    7.043959] dd00: c1a5edb4 6e4df213 bf3c1150 c12f13b8 c3365800 00000000 c386dd34 c386dd28
[    7.043990] dd20: c0850f74 c084f5d0 c386dd5c c386dd38 c08508ac c0850f54 bf3baac0 c386dd48
[    7.044021] dd40: bf3c1150 bf364000 c2926880 00000000 c386dd74 c386dd60 c0852440 c08506f0
[    7.044054] dd60: c1205048 bf364000 c386dd84 c386dd78 c08538b8 c08523c8 c386dd94 c386dd88
[    7.044085] dd80: bf364040 c0853874 c386de0c c386dd98 c02022a8 bf36400c 00000000 00000cc0
[    7.044116] dda0: c386ddc4 c386ddb0 c0b92234 c029bab4 c1801e40 c0425c14 c386de0c c386ddc8
[    7.044149] ddc0: c0425c14 c03e2324 c386de04 c386ddd8 c04076b0 00000008 c386de0c c29268c0
[    7.044180] dde0: f11e1000 6e4df213 00000002 bf3c18c0 00000002 c2926880 c2926780 c29267c0
[    7.044210] de00: c386de34 c386de10 c02cc8c4 c0202264 c386de34 c386de20 c0407818 c386df30
[    7.044243] de20: 00000002 00000002 c386df14 c386de38 c02cf178 c02cc87c bf3c18cc 00007fff
[    7.044274] de40: bf3c18c0 c02cb2d4 f1239094 c1205048 c0e2bc0c c0e2bc24 c0e2bbb4 bf373090
[    7.044305] de60: c0e2bc74 bf3c1ad4 bf3bf3a0 c28fd818 bf3c1908 c2926788 c0e2b760 c3860001
[    7.044336] de80: 00000000 c0ee89fc c0ed1098 00000000 00000000 00000000 00000000 00000000
[    7.044367] dea0: 00000000 00000000 6e72656b 00006c65 00000000 00000000 00000000 00000000
[    7.044397] dec0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    7.044428] dee0: 00000000 6e4df213 c386df2c c1205048 00000000 b6d5f8e0 0000001d c0200204
[    7.044459] df00: c386c000 0000017b c386dfa4 c386df18 c02cf764 c02ccc38 c386df2c 7fffffff
[    7.044491] df20: 00000000 00000002 00000000 f11e1000 f1213743 f121a0c0 f11e1000 00058094
[    7.044522] df40: f123889c f12386bc f1222dc0 00036000 0003c970 0000ea24 00040b75 00000000
[    7.044554] df60: 00000000 00000000 0000ea14 00000030 00000031 00000028 00000017 00000010
[    7.044584] df80: 00000000 6e4df213 6238a400 00000000 00000000 0000017b 00000000 c386dfa8
[    7.044615] dfa0: c02001e4 c02cf6a8 6238a400 00000000 0000001d b6d5f8e0 00000000 b6d603f4
[    7.044648] dfc0: 6238a400 00000000 00000000 0000017b 01916e50 0049a1dc 019132b0 00000000
[    7.044679] dfe0: bef76160 bef76150 b6d569d8 b6e46350 60000010 0000001d 00000000 00000000
[    7.044706] Backtrace: 
[    7.044833] [<bf3965d8>] (vc4_fkms_bind [vc4]) from [<c0849528>] (component_bind_all+0x110/0x23c)
[... 46 lines cut ...]
[    7.046707] dfe0: bef76160 bef76150 b6d569d8 b6e46350
[    7.046731]  r7:0000017b r6:00000000 r5:00000000 r4:6238a400
[    7.046759] Code: f57ff04e e59a3000 e3a04000 e59332f8 (e5834000) 
[    7.046883] ---[ end trace 3ba1ad5a7d0f65e3 ]---
[    7.056443] [drm] Initialized v3d 1.0.0 20180419 for fec00000.v3d on minor 0
[    7.193332] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    7.252152] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    7.304754] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    7.418592] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.477571] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.478438] usbcore: registered new interface driver brcmfmac
[    7.754683] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.754834] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.768118] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    9.118038] uart-pl011 fe201000.serial: no DMA platform data
[    9.125627] 8021q: 802.1Q VLAN Support v1.8
[    9.325586] random: crng init done
[    9.325633] random: 7 urandom warning(s) missed due to ratelimiting
[    9.535468] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    9.729357] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   10.143487] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)

Power down, remove Cariboulite, pi4 had hdmi again.
Removing the /boot/config.txt entry "dtparam=i2c_vc=on"  does not make a difference.
Any hints welcome!

kcman100, 2023-03-26:
Maybe a similar/same problem for me with RPi-400?
No HDMI, but sill have networking with CaribouLite connected.

[    5.392579] Console: switching to colour dummy device 80x30
[    5.415452] vc4_firmware_kms fe600000.firmwarekms: can't request region for resource [mem 0xfe600000-0xfe6000ff]
[    5.415576] [drm:vc4_ioremap_regs [vc4]] *ERROR* Failed to map registers: -16
[    5.415681] [drm:vc4_fkms_bind [vc4]] *ERROR* Oh dear, failed to map registers
[    5.415708] 8<--- cut here ---
[    5.415735] Unable to handle kernel paging request at virtual address fffffff0
[    5.415758] pgd = (ptrval)
[    5.415784] [fffffff0] *pgd=80000000007003, *pmd=2ff5e003, *pte=00000000
[    5.415852] Internal error: Oops: a07 [#1] SMP ARM
[    5.415867] Modules linked in: cfg80211(+) rfkill vc4(+) bcm2835_smi_dev v3d(+) cec gpu_sched bcm2835_smi raspberr$
[    5.416315] CPU: 1 PID: 174 Comm: systemd-udevd Tainted: G         C        5.10.103-v7l+ #1529
[    5.416330] Hardware name: BCM2711
[    5.416437] PC is at vc4_fkms_bind+0x23c/0x320 [vc4]
[    5.416462] LR is at irq_work_queue+0x14/0x2c
[    5.416477] pc : [<bf3b6814>]    lr : [<c0350b24>]    psr: 60000013
[    5.416491] sp : c36bfad8  ip : 00000000  fp : c36bfb3c
[    5.416504] r10: c395bd40  r9 : c395bcc0  r8 : 00040016
[    5.416518] r7 : c1a8d810  r6 : c3241800  r5 : c395bd44  r4 : 00000000
[    5.416532] r3 : fffffff0  r2 : 6e4df213  r1 : 00000027  r0 : 00000042
[    5.416548] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    5.416567] Control: 30c5383d  Table: 035d1940  DAC: 55555555
[    5.416582] Process systemd-udevd (pid: 174, stack limit = 0x(ptrval))
[    5.416597] Stack: (0xc36bfad8 to 0xc36c0000)

kcman100, 2023-03-26:
I did a clean OS for my RPi-400.  Latest desktop 64-bit load from RPi-Website.
And now seems to be working.  cariboulite_test_app runs.
This was an upgrade from 10-Buster to 11-Bullseye (both were 64-bit).
And I suppose a chance something I had previously loaded on 10-Buster was causing a problem?

LeoBistmans, 2023-04-09:
Thanks for information,
followed the lead, upgrade to 11 Bullseye.   Made things worse, does not detect the 2 radios of the Caribou any more like it did in 10 Buster.
Seems to lack /usr/bin/c++ for compilation?