cariboulabs/cariboulite #89: writeStream method not implemented in soapy_api

hudouseko, 2023-03-19:
When I try to call writeStream method using Soapy API (see code snippet below), I get always back (instead of number of samples written) error code -5 in the returned status variable. This seems to correspond to `#define SOAPY_SDR_NOT_SUPPORTED (-5)` as defined in [https://github.com/pothosware/SoapySDR/blob/master/include/SoapySDR/Errors.h](url) . Equal code works for me fine with other SDRs (e.g. Lime Mini). Looking into implementation of soapi_api in cariboulite project, I'd expect int Cariboulite::writeStream to be implemented here: [https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/soapy_api/CaribouliteStreamFunctions.cpp](url) , but I can't find it there nor elsewhere in the project, therefore I'm assuming it is not implemented (yet)

```
synthStream = sdr.setupStream(SOAPY_SDR_TX, SOAPY_SDR_CF32, [0], sargs)
sdr.activateStream(synthStream)
mtu = sdr.getStreamMTU(synthStream)
cw = np.ones(mtu,dtype=np.complex64)
status = sdr.writeStream(synthStream, [cw], cw.size, timeoutUs=1000000)
```

hudouseko, 2023-03-30:
I've tried to implement the missing soapy_api part here: [https://github.com/hudouseko/cariboulite/tree/main/software/libcariboulite/src/soapy_api](url), nevertheless it seems there is something more to be done in other parts of the code (libcariboulite or others). I always end up with timeout in `int ret = cariboulite_radio_write_samples(radio, buffer, num_samples);`

Overall it doesn't seem that there would be much testing done on transmit direction (streaming IQ samples out). All Tx direction testing seems to work with CW only that is generated not by sending out samples, but by either putting fixed value to DACs using control registers or by using LO of RF mixer.

jessupp, 2023-04-07:
Thank you for trying to get this one sorted.  So far I have seen no evidence of being able to use this for transmission other than using the App exactly as you have mentioned above.

I saw your fork and tried it. I used ODR Dabmod to see if I could get any DAB transmission.

I ended up with the following error:

04-03 11:15:03.274  1575  1587 D CARIBOULITE Radio cariboulite_radio_activate_channel@cariboulite_radio.c:849 Setup Modem state cmd_rx
[ERROR] Cariboulite::writeStream - not TX stream!!
ERROR Unexpected stream error NOT_SUPPORTED
ERROR SDR output thread caught runtime error: Fault in Soapy
ERROR Exception caught: SDR thread failed
ERROR Modulator failure.
      40 DAB frames encoded

I believe DabMod outputs CF32 and could not understand how that ended up not being a TX stream!! which I can see on on line 273 of CaribouliteStreamFunctions.

The above may be useful to you and if you have any ideas how I might go about correcting this I would be grateful.  Thanks.

meexmachina, 2023-05-05:
the transmission side is not yet fully implemented and I working on it right now. 
It is part of the local repo I working on currently:
1. firmware - adding a TX I/Q queue.
2. integrating this queue in the SMI controller in FPGA
3. supporting in Soapy (as it is currently already supported in the c -api)