cariboulabs/cariboulite #91: Compiling issues

zetaerre, 2023-03-20:
Hi,

fresh install with device attached to an RasperryPi 400 give this result below, I would like to report the 
KERNELHEADERS_DIR-NOTFOUND when compiling  module smi_stream_dev.ko and the missing ARCH=arm64 that create issue if  KERNER HEADER is sovled.

zetaerre@raspberrypi400:~/ $ uname -a
Linux raspberrypi400 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

zetaerre@raspberrypi400:~ $ mkdir ~/projects
cd ~/projects
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite
Cloning into 'cariboulite'...
remote: Enumerating objects: 7358, done.
remote: Counting objects: 100% (1687/1687), done.
remote: Compressing objects: 100% (593/593), done.
remote: Total 7358 (delta 1051), reused 1643 (delta 1036), pack-reused 5671
Receiving objects: 100% (7358/7358), 235.04 MiB | 7.50 MiB/s, done.
Resolving deltas: 100% (4724/4724), done.
Updating files: 100% (498/498), done.
zetaerre@raspberrypi400:~/projects/cariboulite $ ./install.sh 

[  1  ] CaribouLite Git Repo
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
Submodule 'software/libcariboulite/src/iir' (https://github.com/berndporr/iir1) registered for path 'software/libcariboulite/src/iir'
Cloning into '/home/zetaerre/projects/cariboulite/software/libcariboulite/src/iir'...
Submodule path 'software/libcariboulite/src/iir': checked out 'fc8634d15b470a00ba4859915fec0047bb486dd4'

[  2  ] Updating system and installing dependencies...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:3 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]        
Get:5 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [232 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main armhf Packages [232 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main Translation-en [154 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [315 kB]
Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
Fetched 1,263 kB in 2s (833 kB/s)                    
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
avahi-daemon is already the newest version (0.8-5+deb11u1).
pkg-config is already the newest version (0.29.2-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  cmake-data comerr-dev icu-devtools krb5-multidev libavahi-common-dev libbsd-dev libdbus-1-dev libgssrpc4 libicu-dev libjsoncpp24 libkadm5clnt-mit12 libkadm5srv-mit12 libkdb5-10
  libkrb5-dev libmd-dev libncurses-dev libnorm-dev libpgm-dev librhash0 libsodium-dev libxml2-dev
Suggested packages:
  cmake-doc ninja-build doc-base git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn krb5-doc krb5-user icu-doc ncurses-doc
  libnorm-doc
The following NEW packages will be installed:
  cmake cmake-data comerr-dev icu-devtools krb5-multidev libavahi-client-dev libavahi-common-dev libbsd-dev libdbus-1-dev libgssrpc4 libicu-dev libjsoncpp24 libkadm5clnt-mit12
  libkadm5srv-mit12 libkdb5-10 libkrb5-dev libmd-dev libncurses-dev libncurses5-dev libnorm-dev libpgm-dev librhash0 libsodium-dev libxml2-dev libzmq3-dev module-assistant
  raspberrypi-kernel-headers
[... 685 lines cut ...]
[ 59%] Built target hat
[ 66%] Built target production_utils
[ 70%] Built target zf_log
[ 75%] Built target cariboulite
[ 79%] Built target cariboulite_util
[ 83%] Built target cariboulite_test_app
[ 87%] Built target fpgacomm
[ 90%] Built target caribou_programmer
[100%] Built target SoapyCariboulite
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8-2/libSoapyCariboulite.so
-- Set runtime path of "/usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8-2/libSoapyCariboulite.so" to ""
-- Installing: /usr/local/lib/libcariboulite.a
-- Installing: /usr/local/bin/cariboulite_util
-- Set runtime path of "/usr/local/bin/cariboulite_util" to ""

[  6  ] Environmental Settings...
1. SPI configuration...  OK :)
2. ARM I2C Configuration...  OK :)
3. I2C-VC Configuration...  Warning
To communicate with CaribouLite EEPROM, the i2c_vc device needs to be enabled
Please add the following to the '/boot/config.txt' file: 'dtparam=i2c_vc=on'

[  7  ] Installation errors occured.

ImDroided, 2023-03-20:
you need to install your kernel headers and make sure they are linked correctly. Just go download Dragon OS or the default OS and you shouldn't have any issues.

Alle1307, 2023-06-28:
Same sort of issue here, fresh install on Raspberry Pi 2 Zero with latest official 64bit lite kernel on new 32GB SD card. Only addition is GIT installed. Looks like some dependancies are missing. Anyway it crashes somehow. 
What is missing? Are the instructions complete?

Here is my log:

[  1  ] CaribouLite Git Repo
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
Submodule 'software/libcariboulite/src/iir' (https://github.com/berndporr/iir1) registered for path 'software/libcariboulite/src/iir'
Cloning into '/home/alle/projects/cariboulite/software/libcariboulite/src/iir'...
Submodule path 'software/libcariboulite/src/iir': checked out 'fc8634d15b470a00ba4859915fec0047bb486dd4'

[  2  ] Updating system and installing dependencies...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Fetched 44.1 kB in 3s (14.4 kB/s)             
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.30.2-1+deb11u2).
pkg-config is already the newest version (0.29.2-1).
The following additional packages will be installed:
  cmake-data comerr-dev icu-devtools krb5-multidev libarchive13 libbsd-dev
  libgssrpc4 libicu-dev libjsoncpp24 libkadm5clnt-mit12 libkadm5srv-mit12
  libkdb5-10 libkrb5-dev libmd-dev libncurses-dev libnorm-dev libnorm1
  libpgm-5.3-0 libpgm-dev librhash0 libsodium-dev libsodium23 libxml2-dev
  libzmq5
Suggested packages:
  cmake-doc ninja-build doc-base krb5-doc lrzip krb5-user icu-doc ncurses-doc
  libnorm-doc
The following NEW packages will be installed:
  cmake cmake-data comerr-dev icu-devtools krb5-multidev libarchive13
  libbsd-dev libgssrpc4 libicu-dev libjsoncpp24 libkadm5clnt-mit12
  libkadm5srv-mit12 libkdb5-10 libkrb5-dev libmd-dev libncurses-dev
  libncurses5-dev libnorm-dev libnorm1 libpgm-5.3-0 libpgm-dev librhash0
  libsodium-dev libsodium23 libxml2-dev libzmq3-dev libzmq5 module-assistant
  raspberrypi-kernel-headers
0 upgraded, 29 newly installed, 0 to remove and 0 not upgraded.
Need to get 29.8 MB of archives.
After this operation, 159 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bullseye/main arm64 libarchive13 arm64 3.4.3-2+deb11u1 [320 kB]
Get:2 http://archive.raspberrypi.org/debian bullseye/main arm64 cmake-data all 3.18.4-2+rpt1+rpi1+deb11u1 [1726 kB]
Get:3 http://deb.debian.org/debian bullseye/main arm64 libjsoncpp24 arm64 1.9.4-4 [72.5 kB]
Get:4 http://deb.debian.org/debian bullseye/main arm64 librhash0 arm64 1.4.1-2 [127 kB]
Get:5 http://deb.debian.org/debian bullseye/main arm64 comerr-dev arm64 2.1-1.46.2-2 [108 kB]
Get:6 http://deb.debian.org/debian bullseye/main arm64 icu-devtools arm64 67.1-7 [189 kB]
Get:7 http://deb.debian.org/debian bullseye/main arm64 libgssrpc4 arm64 1.18.3-6+deb11u3 [89.0 kB]
Get:8 http://deb.debian.org/debian bullseye/main arm64 libkdb5-10 arm64 1.18.3-6+deb11u3 [72.2 kB]
Get:9 http://deb.debian.org/debian bullseye/main arm64 libkadm5srv-mit12 arm64 1.18.3-6+deb11u3 [84.1 kB]
Get:10 http://deb.debian.org/debian bullseye/main arm64 libkadm5clnt-mit12 arm64 1.18.3-6+deb11u3 [72.3 kB]
Get:11 http://deb.debian.org/debian bullseye/main arm64 krb5-multidev arm64 1.18.3-6+deb11u3 [158 kB]
Get:12 http://deb.debian.org/debian bullseye/main arm64 libmd-dev arm64 1.0.3-3 [45.9 kB]
Get:13 http://deb.debian.org/debian bullseye/main arm64 libbsd-dev arm64 0.11.3-1 [233 kB]
Get:14 http://deb.debian.org/debian bullseye/main arm64 libicu-dev arm64 67.1-7 [9468 kB]
Get:15 http://archive.raspberrypi.org/debian bullseye/main arm64 cmake arm64 3.18.4-2+rpt1+rpi1+deb11u1 [3673 kB]
Get:16 http://deb.debian.org/debian bullseye/main arm64 libkrb5-dev arm64 1.18.3-6+deb11u3 [47.5 kB]
[... 403 lines cut ...]
/home/alle/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:255: Warning 560: Unknown Doxygen command: parblock.
/home/alle/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:259: Warning 560: Unknown Doxygen command: endparblock.
/home/alle/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:261: Warning 560: Unknown Doxygen command: parblock.
/home/alle/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:265: Warning 560: Unknown Doxygen command: endparblock.
/home/alle/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:234: Warning 560: Unknown Doxygen command: parblock.
/home/alle/projects/cariboulite/installations/SoapySDR/include/SoapySDR/Device.hpp:252: Warning 560: Unknown Doxygen command: endparblock.
[... cut ...]
/home/alle/projects/cariboulite/installations/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:56409:3: note: in expansion of macro ‘SWIG_PYTHON_INITIALIZE_THREADS’
56409 |   SWIG_PYTHON_INITIALIZE_THREADS;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.9/Python.h:145,
                 from /home/alle/projects/cariboulite/installations/SoapySDR/build/swig/python/python3/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx:180:
/usr/include/python3.9/ceval.h:130:37: note: declared here
  130 | Py_DEPRECATED(3.9) PyAPI_FUNC(void) PyEval_InitThreads(void);
      |                                     ^~~~~~~~~~~~~~~~~~

meexmachina, 2023-07-03:
@Alle1307 hi,
can you reinstall python dev? specifically :
`sudo apt-get install libpython3-dev`

seems like the SoapySDR has a issue with binding with python.

@ImDroided  thanks for the instruction for @zetaerre

Alle1307, 2023-07-03:
Yes all the latest updates. Does the software run on Python2 perhaps. That is not valid anymore.

meexmachina, 2023-07-03:
Actually SoapySDR is a third party. Lets try and install it with apt-get. Then you will not need to recompile it. the install script checks whether Soapy-API is installed and if not, it clones the PothosSDR repo and compiles it. There is a direct installation option as follows:
`sudo apt-get install libsoapysdr-dev`
`sudo apt-get install soapysdr-module-remote`

after installing those, run the install.sh script and it should skip that compilation step.

evandonohoe, 2023-09-29:
@meexmachina Has there been any solution to this? I am installing cariboulite on an RPi zero and had the same output as @Alle1307. I discovered that the default version of swig that was installed (4.0.2) doesn't handle the deprecated PyEval_InitThreads() function well so I manually updated the swig to a new version and that error went away but the install still crashes when making SoapySDR and getting to this point:

[ 97%] Building CXX object swig/python/python3/CMakeFiles/_SoapySDR3.dir/CMakeFiles/_SoapySDR3.dir/SoapySDRPYTHON_wrap.cxx.o

I've tried suppressing the warnings that are produced but it still crashes.

meexmachina, 2023-09-30:
Can you try suppressing the soapy install within the install shell script and use
sudo apt install SoapySDR libSoapySDR-dev