cariboulabs/cariboulite #92: SMI Driver KERNELHEADERS_DIR-NOTFOUND

zetaerre, 2023-03-20:
Hi,

after a fresh install on Pi 400 an error KERNELHEADERS_DIR-NOTFOUND appear when compiling smi 

zetaerre@raspberrypi400:~ $ uname -a
Linux raspberrypi400 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

zetaerre@raspberrypi400:~ $ mkdir ~/projects
cd ~/projects
git clone https://github.com/cariboulabs/cariboulite
cd cariboulite
Cloning into 'cariboulite'...
remote: Enumerating objects: 7358, done.
remote: Counting objects: 100% (1687/1687), done.
remote: Compressing objects: 100% (593/593), done.
remote: Total 7358 (delta 1051), reused 1643 (delta 1036), pack-reused 5671
Receiving objects: 100% (7358/7358), 235.04 MiB | 7.50 MiB/s, done.
Resolving deltas: 100% (4724/4724), done.
Updating files: 100% (498/498), done.
zetaerre@raspberrypi400:~/projects/cariboulite $ ./install.sh 

[  1  ] CaribouLite Git Repo
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
Submodule 'software/libcariboulite/src/iir' (https://github.com/berndporr/iir1) registered for path 'software/libcariboulite/src/iir'
Cloning into '/home/zetaerre/projects/cariboulite/software/libcariboulite/src/iir'...
Submodule path 'software/libcariboulite/src/iir': checked out 'fc8634d15b470a00ba4859915fec0047bb486dd4'

[  2  ] Updating system and installing dependencies...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:3 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]        
Get:5 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [232 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main armhf Packages [232 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main Translation-en [154 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [315 kB]
Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
Fetched 1,263 kB in 2s (833 kB/s)                    
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
avahi-daemon is already the newest version (0.8-5+deb11u1).
pkg-config is already the newest version (0.29.2-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  cmake-data comerr-dev icu-devtools krb5-multidev libavahi-common-dev libbsd-dev libdbus-1-dev libgssrpc4 libicu-dev libjsoncpp24 libkadm5clnt-mit12 libkadm5srv-mit12 libkdb5-10
  libkrb5-dev libmd-dev libncurses-dev libnorm-dev libpgm-dev librhash0 libsodium-dev libxml2-dev
Suggested packages:
  cmake-doc ninja-build doc-base git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn krb5-doc krb5-user icu-doc ncurses-doc
  libnorm-doc
The following NEW packages will be installed:
  cmake cmake-data comerr-dev icu-devtools krb5-multidev libavahi-client-dev libavahi-common-dev libbsd-dev libdbus-1-dev libgssrpc4 libicu-dev libjsoncpp24 libkadm5clnt-mit12
  libkadm5srv-mit12 libkdb5-10 libkrb5-dev libmd-dev libncurses-dev libncurses5-dev libnorm-dev libpgm-dev librhash0 libsodium-dev libxml2-dev libzmq3-dev module-assistant
  raspberrypi-kernel-headers
The following packages will be upgraded:
[... 684 lines cut ...]
[ 59%] Built target hat
[ 66%] Built target production_utils
[ 70%] Built target zf_log
[ 75%] Built target cariboulite
[ 79%] Built target cariboulite_util
[ 83%] Built target cariboulite_test_app
[ 87%] Built target fpgacomm
[ 90%] Built target caribou_programmer
[100%] Built target SoapyCariboulite
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8-2/libSoapyCariboulite.so
-- Set runtime path of "/usr/local/lib/aarch64-linux-gnu/SoapySDR/modules0.8-2/libSoapyCariboulite.so" to ""
-- Installing: /usr/local/lib/libcariboulite.a
-- Installing: /usr/local/bin/cariboulite_util
-- Set runtime path of "/usr/local/bin/cariboulite_util" to ""

[  6  ] Environmental Settings...
1. SPI configuration...  OK :)
2. ARM I2C Configuration...  OK :)
3. I2C-VC Configuration...  Warning
To communicate with CaribouLite EEPROM, the i2c_vc device needs to be enabled
Please add the following to the '/boot/config.txt' file: 'dtparam=i2c_vc=on'

[  7  ] Installation errors occured.

hudouseko, 2023-03-20:
Hi, I have moved my Cariboulite from previously working on RPi 0 to a fresh installation on RPi4 (64bit raspbian bullseye) and hit exactly the same issue. Kernel module smi_stream_dev is not compiled due to to the cmake failing to find kernel headers. In fact looking into [https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/caribou_smi/kernel/CMakeLists.txt](url) I see that it uses `uname -r` to get kernel version and assumes headers to be present at `/usr/src/linux-headers-${KERNEL_RELEASE}`. The issue, at least in my case, is that `uname -r` returns 6.1.19-v8+, while kernel headers present are `linux-headers-6.1.19-v7+`. I've got latest headers package installed: `raspberrypi-kernel-headers is already the newest version (1:1.20230317-1)`, but that only seems to contain 
```
linux-headers-6.1.19+
linux-headers-6.1.19-v7+
linux-headers-6.1.19-v7l+
```
I assume this is some temporary misalignment between versions of kernel and kernel headers in their latest packages. Instead of trying to resolve it, I just tried to use as a quick hack headers 6.1.19-v7+ with kernel 6.1.19-v8+. In particular I created the following symlink: `/usr/src $ sudo ln -s linux-headers-6.1.19-v7+ linux-headers-6.1.19-v8+`, then re-run `./install.sh` in cariboulite. Nevertheless, while headers were found, there are other errors popping up now:
```
compiling module smi_stream_dev.ko...
In file included from ./include/linux/types.h:6,
                 from ./include/linux/limits.h:6,
                 from ./include/linux/kernel.h:16,
                 from /home/pi/src/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build/smi_stream_dev.c:41:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
```
So there seems to be more issues, perhaps not just due to slightly different version of kernel headers.

K7MDL2, 2023-03-30:
Found similar issue among others.  Resolved it with details in #66.