cariboulabs/cariboulite #94: Pi 3 DragonOS successful install and testing - hope this helps

ImDroided, 2023-03-20:
Ok so since so many have been having issues I figured I would post what I did to get it working. I am using a Pi 3 in headless mode. I chose DragonOS as it has pretty much everything SDR related already installed and seems to have the highest success rate of getting the Caribou working.

First I downloaded DragonOS from https://sourceforge.net/projects/dragonos-pi64/ I then flashed that to a 64gb sd card and booted up the Pi.

Once in a desktop I opened a terminal and ran 

sudo raspi-config

Once in raspi-config I changed its auto login to console only. After that I enabled SSH and rebooted.

If you are planing on using the pi with a desktop you can skip the raspi-config steps. I am running headless over network. 

Next I chose a fork of this repository because for some reason the owner of this one still has not taken in the fix for the mirror signal. Konimaru fixed the issue and is still waiting to be merged into this repo. 

I then ran the following commands

mkdir ~/projects
cd ~/projects
git clone https://github.com/konimaru/cariboulite.git
cd cariboulite
./install.sh

Note that the script will complain about the raspberry pi kernel headers not being found on apt-get but dont worry they are pre-installed in DragonOS

Let that chug along and then reboot when it finishes. 

From here I used my laptop and SSH into the Caribou. If you are still on the Caribou itself in headless mode just login or if in a desktop open a terminal. 

ssh ubuntu@(your caribou's ip)

sudo SoapySDRUtil --find and you should see the Caribou

Congratulations its all set up. 

Now for the receiving part. I cant for the life of me get anything to connect to SoapySDR but I found a way to get it going. SDR++ or SDRPP as the command goes is already installed and has a server mode that automatically sees the SoapySDR server so I ran
 
sudo sdrpp --server 

Then on my laptop I installed sdrpp from https://github.com/AlexandreRouma/SDRPlusPlus/releases and connected to the sdrpp server 

In SDRPP select SDRPP Server as source, input your Caribou's  IP and hit connect. It will then show remote source, select soapyremote and the caribou will show up 

Auto gain control is broken and spikes the signal so I was getting good results between 22-30 and set decimation at 4-8. I had it at 4 but there is a post somewhere that said 4-8 seems best

tune to a receivable freq for the caribou tuner you selected and hit the play button.

good luck

ImDroided, 2023-03-24:
> One addition note to the great write up, in the script look at line 23 and take out raspberrypi-kernel-headers. This will keep the Ubuntu kernel in place, I seem to recall it getting changed otherwise.
> 
> I’ve been wanting to build a new updated DragonOS Pi64 build once the mirror effect and all other fixes where in, but this is excellent for those wanting to get up and going quickly in the meantime.

If I remember right it just complained that the raspberry pi headers couldnt be found in apt but the rest went on ok

wrju333, 2023-03-30:
Thanks for this @ImDroided  , Best results I've had yet.

alphafox02, 2023-05-19:
Flashed a new install of DragonOS Pi64 and decided to give the main repo another try.

As of this moment you have to open and edit CaribouliteStreamFunctions.cpp and cariboulite.c files and change
cariboulite_get_native_mtu_size_samples
to
cariboulite_radio_get_native_mtu_size_samples
mentioned here https://github.com/cariboulabs/cariboulite/issues/119

After that run the install.sh script and make sure to correctly set the /boot/firmware/config.txt file regardless of what the install script things is correct.

One thing I noticed with sudo sdrpp --server is that it fails to connect to audio and segfaults. Instead I run sudo SoapySDRServer --bind. On another PC on the network I can open and use GQRX to connect to the hat. It works pretty well. 

SDRPP and CubicSDR does not like the connection. SigDigger on the other hand works well. All of course running on the remote laptop connecting to the soapy server.

meexmachina, 2023-06-08:
Updated the install script to correctly address DragonOS issues. 
Still have one udev issue that is different between the Raspbian and the DragonOS. Ubuntu doesn't define devices within the "spi", "i2c" "gpio" groups. they are all inside "dialout". Changing the "udev" rules accordingly.

banshee1971, 2023-10-14:
I try DragonOS (with the UI), and can't see the CaribouLite here when try
sudo SoapySDRUtil --find
or simply open SDR++  .. but can't find the Soapy here...

The only way i was able to see it, was with an old img : 2022-01-07-PiSDR-vanilla.img
But, with lot of error...

banshee1971, 2023-10-19:
same for me.. 
sudo SoapySDRUtil --find
``
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
No devices found!

alphafox02, 2023-10-19:
Check the getting started guide on the desktop if you’re using the DragonOS Pi64 image. Last few steps explain what’s required.

banshee1971, 2023-10-20:
damn... should i seen it .... now i can communicate with the Caribou (i can see it) by doing what it said on the guide.

Now when i try SoapySDRUtil --find .. i now have the cariboulite, with lot of error appear... So i re-install the drivers, just to be sure, reboot and test again.. but during installation, it complain about a missing element in config.txt
But i reboot anyway and test it in SDR++ (add soapy in the source manager.. wasn't there by default).. no luck
So i added the "complained" missing value (it complain also about the value with "off" but i ignore them
**dtoverlay=spi1-3cs**

Then after reboot, try again SoapySDRUtil --find

Still have 2 error
`

10-19 21:17:36.321  3825  3825 E IO_UTILS_Main io_utils_setup@io_utils.c:36 initializing pigpio failed
10-19 21:17:36.321  3825  3825 E CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:205 Error setting up io_utils
[ERROR] cariboulite_init_driver() failed
10-19 21:17:36.321  3825  3825 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
Printing 'findCariboulite' Request:
shm_open: No such file or directory
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
Found device 0
  channel = S1G
  device_id = 0
  driver = Cariboulite
  label = CaribouLite S1G[ba9c3dec]
  name = CaribouLite RPI Hat
  serial = ba9c3dec
  uuid = 2cd749a3-5f08-48c6-ae91-c66c29e23671
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 1
  channel = HiF
  device_id = 1
  driver = Cariboulite
  label = CaribouLite HiF[ba9c3ded]
  name = CaribouLite RPI Hat
  serial = ba9c3ded
  uuid = 2cd749a3-5f08-48c6-ae91-c66c29e23671
  vendor = CaribouLabs LTD
  version = 0x0001

10-19 21:17:38.367  3825  3825 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
10-19 21:17:38.368  3825  3825 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released

`
Still try SDR++ in soapy .. no luck... nothing.
So i try then (to see more detail or error info from a putty session, while looking the Dragon OS UI screen
sudo sdrpp --server     (on the status ... BW was always 0)
`
10-19 21:23:58.441 21628 21628 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:188 FPGA already operational - not programming (use 'force_prog=true' to force update)
10-19 21:23:58.441 21628 21628 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [0,1,1,1]
10-19 21:23:58.441 21628 21628 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
10-19 21:23:58.441 21628 21628 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
10-19 21:23:58.442 21628 21628 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
10-19 21:23:58.442 21628 21628 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0002, Numeric: 2
10-19 21:23:58.442 21628 21628 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
10-19 21:23:58.442 21628 21628 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: 2cd749a3-5f08-48c6-ae91-c66c29e23671, Numeric serial: 0xDD4E1EF6
10-19 21:23:58.442 21628 21628 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
10-19 21:23:58.442 21628 21628 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:754 # Board Info - Product Type: CaribouLite ISM
10-19 21:23:58.442 21628 21628 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
10-19 21:23:58.442 21628 21628 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
10-19 21:23:58.442 21628 21628 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:447 initializing caribou_smi
10-19 21:23:58.443 21628 21628 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
10-19 21:23:58.445 21628 21628 E CARIBOU_SMI_MODULES caribou_smi_insert_smi_modules@caribou_smi_modules.c:71 Module insertion 'smi_stream_dev' failed
10-19 21:23:58.445 21628 21628 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:456 Problem reloading SMI kernel modules
10-19 21:23:58.445 21628 21628 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
10-19 21:23:58.445 21628 21628 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
[... 43 lines cut ...]
10-19 21:23:59.711 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.711 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.711 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.711 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.712 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.712 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
[... cut ...]
10-19 21:23:59.713 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.713 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.713 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.714 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
10-19 21:23:59.714 21628 21628 E IO_UTILS_SPI io_utils_spi_transmit@io_utils_spi.c:517 uninitialized device
`

alphafox02, 2023-10-20:
Try this 

https://sourceforge.net/projects/dragonos-pi64/files/DragonOS_Pi64_Beta31.img.gz/download

Download the older build and do not update/upgrade anything. Make the changes it says in the getting started and see if that does it for you. I had it working at one point on that specific build and since then I’ve just not had much luck.

banshee1971, 2023-10-20:
I try the older build, and follow the "Getting Started" on the destop (only do the Cariboulite part, as suggest)
no luck either (but SoapySDRUtil --find don't show any error now... only those for device not there, like SDRPlay)

So.. i also try the SDRPP --server, and try to remote on it (so i can see the error from the Pi).. This is what i got : 

[image]

Here the only
[image]
 error i can see (that only show SDRPlay error.. suppose it'S because it try to connect device not there)

banshee1971, 2023-11-01:
It seem it will took a while to fix this...