cariboulabs/cariboulite #95: Need Help Please

RSTChile, 2023-03-20:
Estimados Amigos:

Hace 2 días finalmente llegó mi placa Cariboulite, y estoy muy contento de haberla recibido, pero no logro hacerla funcionar.

La tengo instalada en una Raspberry PI Zero, y he realizado la instalación desde cero cuatro veces.

Sin embargo, al hacer las pruebas me aparece siempre varios errores:

CARIBOU_SMI caribou_smi_init@caribou_smi.c:460 couldn't open smi driver file '/dev/smi'
FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized

Y cuando intento usar GQRX

[image]

Por favor, necesito ayuda.

Saludos desde Chile

Alexis

alexis@raspberrypi:~/projects/cariboulite $ sudo SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] SoapyCaribouliteSession, sessionCount: 0
03-20 17:02:28.109  2482  2482 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
03-20 17:02:28.113  2482  2482 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
03-20 17:02:28.113  2482  2482 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
03-20 17:02:28.113  2482  2482 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
03-20 17:02:28.118  2482  2482 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
03-20 17:02:28.359  2482  2482 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
03-20 17:02:28.359  2482  2482 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
03-20 17:02:28.360  2482  2482 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
03-20 17:02:28.360  2482  2482 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
03-20 17:02:28.360  2482  2482 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
03-20 17:02:28.360  2482  2482 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
03-20 17:02:28.360  2482  2482 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
03-20 17:02:28.360  2482  2482 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
FPGA Versions:
	System Version: 01
	Manu. ID: 01
	Sys. Ctrl Version: 01
	IO Ctrl Version: 01
	SMI Ctrl Version: 01
03-20 17:02:28.361  2482  2482 I FPGA caribou_fpga_program_to_fpga@caribou_fpga.c:188 FPGA already operational - not programming (use 'force_prog=true' to force update)
03-20 17:02:28.361  2482  2482 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
03-20 17:02:28.361  2482  2482 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
03-20 17:02:28.363  2482  2482 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
03-20 17:02:28.371  2482  2482 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
03-20 17:02:28.371  2482  2482 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
03-20 17:02:28.371  2482  2482 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
03-20 17:02:28.371  2482  2482 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: e17c4c0a-27f5-4d4b-8fdd-0573552f982d, Numeric serial: 0x4954CC43
03-20 17:02:28.372  2482  2482 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
03-20 17:02:28.372  2482  2482 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
03-20 17:02:28.372  2482  2482 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
03-20 17:02:28.372  2482  2482 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
03-20 17:02:28.372  2482  2482 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
03-20 17:02:28.382  2482  2482 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:98 Unloading smi-stream module
03-20 17:02:28.395  2482  2482 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
03-20 17:02:28.487  2482  2482 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:460 couldn't open smi driver file '/dev/smi'
03-20 17:02:28.505  2482  2482 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
03-20 17:02:28.508  2482  2482 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
03-20 17:02:28.518  2482  2482 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
03-20 17:02:28.525  2482  2482 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
03-20 17:02:28.526  2482  2482 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
03-20 17:02:28.526  2482  2482 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
03-20 17:02:28.526  2482  2482 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
03-20 17:02:28.527  2482  2482 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
[ERROR] cariboulite_init_driver() failed
03-20 17:02:28.859  2482  2482 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
Printing 'findCariboulite' Request:
Found device 0
  channel = S1G
  device_id = 0
  driver = Cariboulite
  label = CaribouLite S1G[92a99886]
  name = CaribouLite RPI Hat
  serial = 92a99886
  uuid = e17c4c0a-27f5-4d4b-8fdd-0573552f982d
  vendor = CaribouLabs LTD
  version = 0x0001

Found device 1
  channel = HiF
  device_id = 1
  driver = Cariboulite
  label = CaribouLite HiF[92a99887]
  name = CaribouLite RPI Hat
  serial = 92a99887
  uuid = e17c4c0a-27f5-4d4b-8fdd-0573552f982d
  vendor = CaribouLabs LTD
  version = 0x0001

03-20 17:02:30.874  2482  2482 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
03-20 17:02:30.877  2482  2482 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
03-20 17:02:30.877  2482  2482 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
03-20 17:02:30.877  2482  2482 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:440 FPGA communication release failed (-1)
03-20 17:02:30.877  2482  2482 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
03-20 17:02:30.877  2482  2482 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:338 closing uninitialized device
03-20 17:02:30.877  2482  2482 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
03-20 17:02:30.877  2482  2482 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
alexis@raspberrypi:~/projects/cariboulite $

ImDroided, 2023-03-22:
I can't get this to translate. You're going to have to post this in English please.

RSTChile, 2023-03-23:
Dear friends:

2 days ago I finally received my Cariboulite plate, and I'm very happy to have received it, but I can't get it to work.

I have it installed on a Raspberry PI Zero, and it did the installation from zero four times.

However, when doing the tests I always get several errors:

CARIBOU_SMI caribou_smi_init@caribou_smi.c :460 could not open smi driver file '/dev/smi'
FPGA caribou_fpga_close@caribou_fpga.c :225 caribou_fpga_close: dev not initialized

Please Help

RSTChile, 2023-03-23:
Dear friends:

2 days ago I finally received my Cariboulite plate, and I'm very happy to have received it, but I can't get it to work.

I have it installed on a Raspberry PI Zero, and it did the installation from zero four times.

However, when doing the tests I always get several errors:

CARIBOU_SMI caribou_smi_init@caribou_smi.c :460 could not open smi driver file '/dev/smi'
FPGA caribou_fpga_close@caribou_fpga.c :225 caribou_fpga_close: dev not initialized

And when I try to use GQRX (See register before)

Please Help

Alexis

ImDroided, 2023-03-24:
Did you edit your config.txt file with the proper dtprams? also everything must be run as root or it will fail.