cariboulabs/cariboulite #96: Sampling rate limited to 4MS/s only in soapy_api

hudouseko, 2023-03-22:
Method `Cariboulite::setSampleRate` as defined in [https://github.com/cariboulabs/cariboulite/blob/main/software/libcariboulite/src/soapy_api/Cariboulite.cpp](url) ignores requested sample rate and always sets it to 4MS/s. Correspondingly method `Cariboulite::getSampleRate` always returns 4000000 (as 4MS/s). Nevertheless device itself supports multiple sample rates from 400kS/s up to 4MS/s, see clause 6.1.5.2 RFn_TXDFE - Transmitter TX Digital Frontend of the AT86RF215 datasheet here: [https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42415-WIRELESS-AT86RF215_Datasheet.pdf](url). It seems that rest of the code is ready to support all sampling rates, while soapi_api is hardcoded for 4MS/s only.

edegraaff, 2023-04-03:
BTW if i change sample freq in openwebrx and change the sample freq to above or below 4 MS/s it is as if i tune up or down the speed of the sound like the pitch of a pick-up

meexmachina, 2023-05-05:
The sampling frequency is limited only by SoapyAPI.
firmware and c API supports different sampling rates.
but in the lower sampling rates (under 800 KSPS) an issue exists in the kernel module. It gets a read sample timeout while waiting for new batches of samples.

meexmachina, 2023-06-08:
The different sampling rates are available in Soapy as well now.