cariboulabs/cariboulite #100: meexmachina: no-shows!

hb9dug, 2023-03-27:
For more than a month, meexmachina, which seems to be the leader of the project, has not made any more contributions. 

A status report on the project, especially on the firmware and the promised basic software, seems to me a reasonable request from the people who invested in CaribouLite!

A prompt answer will be welcome

Cordialement,
Michel, HB9DUG

gerd0355, 2023-03-27:
I agree with you. I am also waiting for more informations. 

Regards
Gerhard

great9, 2023-03-27:
@konimaru seems to be the only guy doing updates on this codebase

hb9dug, 2023-03-27:
I confirm, the package provided by Marco works on PiSDR. 
SDR++ must be run from sudo and  I have to open before a console and execute on it : sudo pulseaudio --system=true to have audio.
GNUradio Soapy Source block works but Sink block craches GNUradio.

Regards,
Michel

hudouseko, 2023-03-28:
>  GNUradio Soapy Source block works but Sink block craches GNUradio.
> 

If not for other reasons as well, I'd assume Soapy sink wouldn't work because writeStream method in soapy_api doesn't seem to be implemented at all (see [https://github.com/cariboulabs/cariboulite/issues/89](url) ).

G0MJW, 2023-03-28:
I think the project has been abandoned. Such a pity.

G0MJW, 2023-03-30:
> Below is Crowdsupply's response to my concerns about the viability of the project:
>
> New reply for the ticket #OTH00021757
>
> Hi Michel,
> Thanks for reaching out, we appreciate your honest feedback.
> This is still an active project, the majority of backer fulfillment has recently been completed. The projects team is communicating with David about the status of a new project update.
> In the meantime, please use the resources already available:
> https://www.crowdsupply.com/cariboulabs/cariboulite-rpi-hat#support- <https://www.crowdsupply.com/cariboulabs/cariboulite-rpi-hat#support-&-documentation> &-documentation
> And check the most recent software update:
> https://www.crowdsupply.com/cariboulabs/cariboulite-rpi-hat/updates/assembly-is-finished-testing-in-progress#software-update
>
> Thank you for having supported this project. Take care!
> Regards,
> Crowd Supply Support
>
> New reply for the ticket #OTH00021757
>
> Hi Michel,
> Additionally, you can share your feedback directly with David here: https://www.crowdsupply.com/cariboulabs/cariboulite-rpi-hat/ask-question
> Also to follow up, the manager assigned to this project has requested the creator write a new update for backers, with a focus on resources for getting started with the board, as well as reporting on further software and firmware status updates.
>
> Thanks again for your message.

Interesting. Thanks.

 

Mike

brianrusso, 2023-04-02:
I'm sure I'll get voted off the island, but let's consider that a 1-person operation designing and constructing an open source, small-scale SDR during a global chip supply chain fiasco is pretty tough.

I agree an update is in order, but let's not get our pitchforks out just yet.

I have received one, and while it has some stutter issues (presumably due to a bandwidth bottleneck somewhere) - it does technically work. Hopefully the issues are just software related and can be resolved.

$0.02

great9, 2023-04-04:
> I'm sure I'll get voted off the island, but let's consider that a 1-person operation designing and constructing an open source, small-scale SDR during a global chip supply chain fiasco is pretty [...]

for almost 2 months now

ImDroided, 2023-04-07:
> I'm sure I'll get voted off the island, but let's consider that a 1-person operation designing and constructing an open source, small-scale SDR during a global chip supply chain fiasco is pretty [...]

While I agree to a point unless the guy is dead there is no reason not to do the 30 second merge of the fixes that have been waiting 2 months in que. Like the code is done other people fixed it for you all you need to do is click merge. The sad part is is if people don't come and read the issues, they'll never know that a fix exists for the mirror issue and other stuff but if this guy would just merge the damn things, they wouldn't have to worry about it from the get-go.

great9, 2023-04-08:
> While I agree to a point unless the guy is dead there is no reason not to do the 30 second merge of the fixes that have been waiting 2 months in que. Like the code is done other people fixed it for [...]

The owner of this repo and founder of cariboulite has interacted in the issues section when these fixes were posted.

meexmachina, 2023-05-05:
Hi contributors, 
Indeed I was extremely unavailable in the last few months and even the easiest pull request has been neglected by me for too long. The firmware and software development is continuing, very much thanks to you.
Currently working on the TX side - I/Q transmission - firmware and software. takes time....

brianrusso, 2023-05-05:
That's great news, glad to hear this project still has life. If you could commit (pun intended) to a regular cadence of updates, I believe that would add a lot of value for this community.

closed 2023-05-05 by meexmachina