cariboulabs/cariboulite #104: smi_stream_dev modules fail to load

McGr, 2023-03-29:
my environment is a Pi4, 8G with a fresh install of Bullsye 64bit.
I've been through the ./install.sh script but so far I've not been able to run cariboulite_test_app due to a `module insertion 'smi_stream_dev' failed`  error.

Front page of the repo says the kernel headers should be in `/software/libcariboulite/caribou_smi/kernel` but the install.sh script did not create this directory. I do find that directory in the cariboulite repo download, but no .ko modules.

Anyone know how to compile these modules and get them inserted into the kernel?

Thanks

Below are the logs from `sudo ./cariboulite_test_app` 

`03-29 10:47:10.293  8937  8937 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
03-29 10:47:10.295  8937  8937 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
03-29 10:47:10.295  8937  8937 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
03-29 10:47:10.295  8937  8937 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
03-29 10:47:10.296  8937  8937 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
03-29 10:47:10.437  8937  8937 I IO_UTILS_Main io_utils_setup@io_utils.c:39 pigpio version 79
03-29 10:47:10.437  8937  8937 I IO_UTILS_SPI io_utils_spi_init@io_utils_spi.c:321 configuring gpio setups
03-29 10:47:10.437  8937  8937 D mCARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:583 Initializing FPGA
03-29 10:47:10.437  8937  8937 I FPGA caribou_fpga_init@caribou_fpga.c:116 configuring reset and irq pins
03-29 10:47:10.437  8937  8937 I FPGA caribou_fpga_init@caribou_fpga.c:124 Initializing io_utils_spi
03-29 10:47:10.437  8937  8937 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:84 FPGA is already configured and running
03-29 10:47:10.437  8937  8937 I CARIBOU_PROG caribou_prog_init@caribou_prog.c:87 device init completed
03-29 10:47:10.437  8937  8937 D 0;32mCARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:596 Programming FPGA
03-29 10:47:10.438  8937  8937 I CARIBOU_PROG caribou_prog_configure_prepare@caribou_prog.c:148 RESET low, Waiting for CDONE low
03-29 10:47:10.439  8937  8937 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:259 Sending bitstream of size 32220
FPGA Versions:
	System Version: 00
	Manu. ID: 00
	Sys. Ctrl Version: 00
	IO Ctrl Version: 00
	SMI Ctrl Version: 00
03-29 10:47:11.580  8937  8937 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:281 bitstream sent 32220 bytes
03-29 10:47:11.580  8937  8937 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:192 sending dummy clocks, waiting for CDONE to rise (or fail)
03-29 10:47:11.580  8937  8937 I CARIBOU_PROG caribou_prog_configure_finish@caribou_prog.c:203 0 dummy clocks sent
03-29 10:47:11.580  8937  8937 I CARIBOU_PROG caribou_prog_configure_from_buffer@caribou_prog.c:291 FPGA programming - Success!

03-29 10:47:11.680  8937  8937 D CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:609 FPGA Digital Values: led0: 0, led1: 0, btn: 1, CFG[0..3]: [1,1,1,1]
03-29 10:47:11.680  8937  8937 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:622 Detecting Board Information
03-29 10:47:11.681  8937  8937 I HAT hat_print_board_info@hat.c:703 # Board Info - Category name: hat
03-29 10:47:11.681  8937  8937 I HAT hat_print_board_info@hat.c:704 # Board Info - Product name: CaribouLite RPI Hat
03-29 10:47:11.681  8937  8937 I HAT hat_print_board_info@hat.c:705 # Board Info - Product ID: 0x0001, Numeric: 1
03-29 10:47:11.681  8937  8937 I HAT hat_print_board_info@hat.c:706 # Board Info - Product Version: 0x0001, Numeric: 1
03-29 10:47:11.681  8937  8937 I HAT hat_print_board_info@hat.c:707 # Board Info - Product UUID: 1e8410d3-fe98-4019-a17e-c9f83b79209f, Numeric serial: 0x41624B2C
03-29 10:47:11.681  8937  8937 I HAT hat_print_board_info@hat.c:708 # Board Info - Vendor: CaribouLabs LTD
03-29 10:47:11.681  8937  8937 I CARIBOULITE Setup cariboulite_print_board_info@cariboulite_setup.c:753 # Board Info - Product Type: CaribouLite FULL
**03-29 10:47:11.681  8937  8937 I CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:272 initializing submodules
03-29 10:47:11.681  8937  8937 D CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:276 INIT FPGA SMI communication
03-29 10:47:11.681  8937  8937 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
03-29 10:47:11.681  8937  8937 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
03-29 10:47:11.682  8937  8937 E CARIBOU_SMI_MODULES caribou_smi_insert_smi_modules@caribou_smi_modules.c:71 Module insertion 'smi_stream_dev' failed
03-29 10:47:11.682  8937  8937 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:451 Problem reloading SMI kernel modules
03-29 10:47:11.682  8937  8937 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule**
03-29 10:47:11.682  8937  8937 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
03-29 10:47:11.682  8937  8937 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
03-29 10:47:11.682  8937  8937 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
03-29 10:47:11.682  8937  8937 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
03-29 10:47:11.682  8937  8937 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
03-29 10:47:11.682  8937  8937 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
03-29 10:47:11.682  8937  8937 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
FPGA Versions:
	System Version: 01
	Manu. ID: 01
	Sys. Ctrl Version: 01
	IO Ctrl Version: 01
	SMI Ctrl Version: 01
03-29 10:47:11.698  8937  8937 E 0;31mCARIBOULITE Test main@cariboulite_test_app.c:68 driver init failed, terminating...

`

[... 76 lines cut ...]
03-29 13:10:23.292 10839 10839 I CARIBOU_SMI caribou_smi_init@caribou_smi.c:442 initializing caribou_smi
03-29 13:10:23.292 10839 10839 D CARIBOU_SMI_MODULES caribou_smi_check_modules@caribou_smi_modules.c:111 Loading smi-stream module
03-29 13:10:23.293 10839 10839 E CARIBOU_SMI_MODULES caribou_smi_insert_smi_modules@caribou_smi_modules.c:71 Module insertion 'smi_stream_dev' failed
03-29 13:10:23.293 10839 10839 E CARIBOU_SMI caribou_smi_init@caribou_smi.c:451 Problem reloading SMI kernel modules
03-29 13:10:23.293 10839 10839 E CARIBOULITE Setup cariboulite_init_submodules@cariboulite_setup.c:280 Error setting up smi submodule
03-29 13:10:23.293 10839 10839 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
03-29 13:10:23.293 10839 10839 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 0
03-29 13:10:23.293 10839 10839 I IO_UTILS_SPI io_utils_spi_remove_chip@io_utils_spi.c:475 removing an spi device with handle 1
03-29 13:10:23.293 10839 10839 I CARIBOU_PROG caribou_prog_release@caribou_prog.c:122 device release completed
03-29 13:10:23.293 10839 10839 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
03-29 13:10:23.294 10839 10839 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:240 Releasing board I/Os - closing SPI
03-29 13:10:23.294 10839 10839 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
[ERROR] cariboulite_init_driver() failed
03-29 13:10:23.324 10839 10839 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
03-29 13:10:23.325 10839 10839 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:699 driver being released
03-29 13:10:23.325 10839 10839 D CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:436 CLOSE FPGA communication
03-29 13:10:23.325 10839 10839 E FPGA caribou_fpga_close@caribou_fpga.c:225 caribou_fpga_close: dev not initialized
03-29 13:10:23.325 10839 10839 E CARIBOULITE Setup cariboulite_release_submodules@cariboulite_setup.c:440 FPGA communication release failed (-1)
[... cut ...] PI
03-29 13:10:23.325 10839 10839 E IO_UTILS_SPI io_utils_spi_close@io_utils_spi.c:338 closing uninitialized device
03-29 13:10:23.325 10839 10839 I CARIBOULITE Setup cariboulite_release_io@cariboulite_setup.c:243 Releasing board I/Os - io_utils_cleanup
03-29 13:10:23.325 10839 10839 I CARIBOULITE Setup cariboulite_release_driver@cariboulite_setup.c:708 driver released
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/cariboulite/software/libcariboulite/build

konimaru, 2023-03-30:
I suspect it's a version mismatch (we had a kernel update recently). Can you find your .ko file and check its version against your current kernel version?
```
$ strings smi_stream_dev.ko | grep vermagic=
```
vs
```
$ uname -a
```
If this matches try loading the module manually via `modprobe` in order to get some error message.

McGr, 2023-03-30:
Its a good thought but I think in my case the smi_stream_dev.ko wasn't built.
I'm unable to find the module anywhere in the cariboulite dir structure or anywhere under /
When I run 
`lsmod | grep smi`
I get
`bcm2835_smi  20480 0`
but the bcm2835_smi_dev is not listed.

I did try building the libcariboutlite after the install.sh (log above) but looks like that didn't build the module either
 The readme suggests that the FPGA might not be properly flashed but doesn't yet give any info on how to check

konimaru, 2023-03-30:
In the build log there should be something like this:
```
[ 5 ] Compiling main source...

...

3. SMI kernel module...
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Kernel release: 5.15.84-v8+

...
```
Can you find this for **your** build and post it here?

McGr, 2023-03-30:
I didn't find a build log so I just reran install.sh and captured stderr and stdout.

Looks like there are two problems.

KERNELHEADERS_DIR not found
and 
needing sudo to run part of the script

```
[  5  ] Compiling main source...
External Tools...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/cariboulite/software/utils/build
[ 50%] Linking C executable generate_bin_blob
[100%] Built target generate_bin_blob
 libIIR 
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/cariboulite/software/libcariboulite/src/iir/build

...

Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /usr/local/lib/arm-linux-gnueabihf/libiir.so.1.9.1
-- Up-to-date: /usr/local/lib/arm-linux-gnueabihf/libiir.so.1
-- Up-to-date: /usr/local/lib/arm-linux-gnueabihf/libiir.so
-- Up-to-date: /usr/local/include/iir/Biquad.h
-- Up-to-date: /usr/local/include/iir/Butterworth.h
-- Up-to-date: /usr/local/include/iir/Cascade.h
-- Up-to-date: /usr/local/include/iir/ChebyshevI.h
-- Up-to-date: /usr/local/include/iir/ChebyshevII.h
-- Up-to-date: /usr/local/include/iir/Common.h
-- Up-to-date: /usr/local/include/iir/Custom.h
-- Up-to-date: /usr/local/include/iir/Layout.h
-- Up-to-date: /usr/local/include/iir/MathSupplement.h
-- Up-to-date: /usr/local/include/iir/PoleFilter.h
-- Up-to-date: /usr/local/include/iir/RBJ.h
-- Up-to-date: /usr/local/include/iir/State.h
-- Up-to-date: /usr/local/include/iir/Types.h
-- Up-to-date: /usr/local/include/Iir.h
-- Up-to-date: /usr/local/lib/arm-linux-gnueabihf/libiir_static.a
-- Up-to-date: /usr/local/include/iir/Biquad.h
-- Up-to-date: /usr/local/include/iir/Butterworth.h
-- Up-to-date: /usr/local/include/iir/Cascade.h
-- Up-to-date: /usr/local/include/iir/ChebyshevI.h
-- Up-to-date: /usr/local/include/iir/ChebyshevII.h
-- Up-to-date: /usr/local/include/iir/Common.h
-- Up-to-date: /usr/local/include/iir/Custom.h
-- Up-to-date: /usr/local/include/iir/Layout.h
-- Up-to-date: /usr/local/include/iir/MathSupplement.h
-- Up-to-date: /usr/local/include/iir/PoleFilter.h
-- Up-to-date: /usr/local/include/iir/RBJ.h
-- Up-to-date: /usr/local/include/iir/State.h
-- Up-to-date: /usr/local/include/iir/Types.h
-- Up-to-date: /usr/local/include/Iir.h
-- Up-to-date: /usr/local/lib/cmake/iir/iir-config.cmake
-- Installing: /usr/local/lib/cmake/iir/iir-config-relwithdebinfo.cmake
 SMI kernel module...
mkdir: cannot create directory ‘build’: File exists
-- Kernel release: 6.1.19-v8+
-- Kernel headers: KERNELHEADERS_DIR-NOTFOUND
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mike/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build
compiling module smi_stream_dev.ko...
make[3]: *** KERNELHEADERS_DIR-NOTFOUND: No such file or directory.  Stop.
make[2]: *** [CMakeFiles/smi_stream_dev.dir/build.make:87: smi_stream_dev] Error 2
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/smi_stream_dev.dir/all] Error 2
make: *** [Makefile:103: all] Error 2
 Main software...
mkdir: cannot create directory ‘build’: File exists
[INFO] SoapyCaribouliteSession, sessionCount: 0
03-30 10:26:06.798 21511 21511 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:560 driver initializing
03-30 10:26:06.798 21511 21511 I CARIBOULITE Setup cariboulite_init_driver_minimal@cariboulite_setup.c:569 Initializing signals
03-30 10:26:06.798 21511 21511 I CARIBOULITE Setup cariboulite_setup_signal_handler@cariboulite_setup.c:687 setting up signal handler
03-30 10:26:06.799 21511 21511 I CARIBOULITE Setup cariboulite_setup_io@cariboulite_setup.c:202 Setting up board I/Os
03-30 10:26:06.799 21511 21511 I IO_UTILS_Main io_utils_setup@io_utils.c:26 initializing pigpio
03-30 10:26:06.800 21511 21511 V initCheckPermitted@pigpio.c:7396 
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

```

konimaru, 2023-03-30:
What kernel have you got and what headers are available under `/usr/src`?

McGr, 2023-03-30:
This build was just a basic fresh install of the current 64Bit bullseye on a Pi 4, 8G.
The /usr/src dir contains
`linux-headers-6.1.19+
linux-headers-6.1.19-v7+
linux-headers-6.1.19-v7l+
sense-hat
`

konimaru, 2023-03-30:
And your kernel is at? (`uname -a`)

McGr, 2023-03-30:
Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux

McGr, 2023-03-30:
Ok - I'll create the softlink.  What should they point too (I'm new at this)

konimaru, 2023-03-30:
In `/usr/src` run the following:
```
$ sudo -s linux-headers-6.1.19-v7+ linux-headers-6.1.19-v8+
```
That said, it may well be that you need the real headers in case there is a significant change. But try this first and the remove caribou_smi/kernel/build just in case.

konimaru, 2023-03-30:
Ouch, one last try, remove the link you created and update like this:
```
$ sudo apt install raspberrypi-kernel-headers
```

hudouseko, 2023-03-30:
> Yeah, I think the 64bit version is a little bit behind the 32bit version. No big deal to just reflash the image and see what happens. I did try updating the headers. Looks like I have the latest version
> 
I'm in the same state - stuck on different kernel version and different kernel headers - RPi4 with Raspbian. I have tried one more thing - installing clean 32bit version, but that has not helped. Actually first `apt update ` and `apt upgrade` performs upgrade to 64bit version with the same kernel/kernel header mismatch described above. I.have tried the symlink as well, bit that didn't work.
So I hope.that some upcoming update of Raspbian will fix the versions mismatch

konimaru, 2023-03-30:
Beats me why only the RPi4 is affected. Should check on my 400. RPi3B is fine.

EelkeVisser, 2023-03-30:
The Raspberry project has changed the kernel from 32 to 64 bit default for the Pi 4B. Even if you installed a 32 bit image! But there are no updated kernel headers available yet. The `./install.sh` will always update everything to the latest version, and will thus change your kernel from 32 to 64 bit. No questions asked! You can change back to the 32 bit kernel by adding `arm_64bit=0` to `/boot/config.txt`. Please check if you have a 32 bit `kernel7.img` kernel in `/boot`, before changing. See:

https://github.com/raspberrypi/linux/issues/5402#issuecomment-1484676216

hudouseko, 2023-03-31:
> `arm_64bit=0` to `/boot/config`

Thanks a lot, this actually resolves the issue of kernel module compilation on RPi4B with Raspbian bullseye. Now after switching to 32bit kernel, actual (most recent) kernel version used is 6.1.19-v71+, which has corresponding kernel headers available in `/usr/src/linux-headers-6.1.19-v71+`. Following that compilation of the `smi_stream_dev` kernel modules was successful and I've got cariboulite running on RPi4B now :)

McGr, 2023-04-01:
@hudouseko I'm trying to do the same. I see the src code for the headers but not sure of the steps to compile them.
Could you be specific in the steps you took?
Thanks MIke.

ImDroided, 2023-04-01:
You guys should try DragonOS download the image and on first run do a sudo apt-get update then upgrade then dist-upgrade then a update and upgrade again you will be on the latest everything with 64 bit and the kernel headers are already preinstalled.

hudouseko, 2023-04-01:
> @hudouseko I'm trying to do the same. I see the src code for the headers but not sure of the steps to compile them. Could you be specific in the steps you took? Thanks MIke.

Hi Mike, not sure what state you are in, let's assume you've got RPi4B, switched it to 32bit kernel as described by @EelkeVisser (`arm_64bit=0` to `/boot/config`) and did `sudo apt update` & `sudo apt upgrade`. By this time you should be running kernel 6.1.19-v7l+ (verify using `uname -r`) and have corresponding kernel headers (verify using `ls /usr/src`, directory `linux-headers-6.1.19-v7l+` should be present).

If you have the above steps completed, then it is fairly straightforward:
Clone cariboulite repository
>pi@pi4ant2:~ $ git clone https://github.com/cariboulabs/cariboulite.git
Cloning into 'cariboulite'...
remote: Enumerating objects: 7358, done.
remote: Counting objects: 100% (1687/1687), done.
remote: Compressing objects: 100% (597/597), done.
remote: Total 7358 (delta 1048), reused 1641 (delta 1032), pack-reused 5671
Receiving objects: 100% (7358/7358), 235.06 MiB | 1.02 MiB/s, done.
Resolving deltas: 100% (4717/4717), done.
Updating files: 100% (498/498), done.

Run installation script, which as one of the steps compiles and installs corresponding kernel modules (long output, I have abbreviated it and left part in the middle that corresponds to kernel module compilation)
>pi@pi4ant2:~/cariboulite $ ./install.sh 
[  1  ] CaribouLite Git Repo
Already up to date.
Submodule 'software/libcariboulite/src/iir' (https://github.com/berndporr/iir1) registered for path 'software/libcariboulite/src/iir'
Cloning into '/home/pi/cariboulite/software/libcariboulite/src/iir'...
Submodule path 'software/libcariboulite/src/iir': checked out 'fc8634d15b470a00ba4859915fec0047bb486dd4'
[  2  ] Updating system and installing dependencies...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
....
>3. SMI kernel module...
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Kernel release: 6.1.19-v7l+
-- Kernel headers: /usr/src/linux-headers-6.1.19-v7l+
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/cariboulite/software/libcariboulite/src/caribou_smi/kernel/build
Scanning dependencies of target smi_stream_dev
compiling module smi_stream_dev.ko...
The code filename is path: '..', name: 'smi_stream_dev_gen', ext: 'h'
Built target smi_stream_dev
...
>[  6  ] Environmental Settings...
>1. SPI configuration...  OK :)
>2. ARM I2C Configuration...  OK :)
>3. I2C-VC Configuration...  OK :)
>[  7  ] All went well. Please reboot the system to finalize installation...

Reboot
`sudo reboot`

Check that kernel module is loaded
>pi@pi4ant2:~/cariboulite $ lsmod | grep smi
smi_stream_dev         20480  0
bcm2835_smi            20480  1 smi_stream_dev

@ImDroided Thanks for the tip, I'll give it a try

konimaru, 2023-04-02:
For the record, on a RPi400 starting with Pi OS Lite 64bit (from the imager) I never had any issues with upgrading. Headers always matched. I have a feeling that going from 32 to 64 will mess things up a little.

meexmachina, 2023-06-08:
an installation script, and moving the driver "update" part to `/driver/` makes it easier to update the driver when disrto/linux system updates. The module that was compiled with a specific header set of a certain kernel is not anymore valid after kernel version is updating. there were many updates lately in the Raspbian side. Thats why we moved all the driver outside of the code as it should be and added an specific installation script. 
It also solves the `sudo`ing problem for the driver. hoping to push the kernel module to the mainline linux code (need help with that) and then the problem will be removed completely.